dresden-elektronik/deconz-rest-plugin #2817: device request smoke sensor klik aan klik uit ZSDR-850

goudhaan, 2020-05-25:
[image]

[image]

[image]

[image]

SwoopX, 2020-05-25:
This device is already supported. It's a rebranded Heiman smoke detector, as you can see from the basic cluster output.

goudhaan, 2020-05-26:
Well I believe that’s not completely true. Because it is not showing up in the app. 
Maybe slightly different?

Met vriendelijke groeten,
With kind regards,

Jan Goudriaan

SwoopX, 2020-05-27:
It is supported. Phoscon is not deconz and the device then has just not been added to Phoscon, which doesn't impact functionality.

niceguy0815, 2020-06-04:
Hi I have buyed zhis smoke detector Zigbee Smarter Rauchmelder Trust ZSDR-850:
https://www.hornbach.de/shop/Zigbee-Smarter-Rauchmelder-Trust-ZSDR-850-Kompatibel-mit-all-SMART-HOME-by-hornbach/10211208/artikel.html?wt_cc2=41053550860&wt_cc3=183363166959&gclid=CjwKCAjwt-L2BRA_EiwAacX32erXysLoQEq5sKlST5iIOUNaTO5gPorx-a4Xqv-4YwPbaIYUdRJNYhoCQsMQAvD_BwE&wt_cc6=10211208&wt_mc=de.paid.sea.google.alwayson_assortment.leu.pla.766470194.41053550860.&wt_cc1=766470194

Now i try to conect it to deconze.
I have paired it over Poscon app and see it in Deconz.
[image]

But there is no "Alarms"Entry on the device.

I think i do something wrong?
Is there a secial Hint to connect this device right?
Ore is it not supported?

niceguy0815, 2020-06-04:
Hi and thanks for your answer.

But when i check sensor with res api iget:
`{"config":{"battery":null,"on":true,"pending":[],"reachable":true},"ep":1,"etag":"6ee283b50877bdebd7f288addc8102db","lastseen":"2020-06-04T16:13:56.626","manufacturername":"Heiman","modelid":"SmokeSensor-EM","name":"Rauchmelder","state":{"fire":null,"lastupdated":"none","lowbattery":null,"tampered":null},"swversion":"1.1.1","type":"ZHAFire","uniqueid":"00:0d:6f:00:14:df:cb:14-01-0500"}`

It loks for me it get no readings from the device?
And I also have definied the sensor in fhem and ther are no readings from the device?

niceguy0815, 2020-06-04:
[image]
[image]

niceguy0815, 2020-06-05:
So it is not a Problem on my side?

SwoopX, 2020-06-05:
Try to reset and re-pair the device. I pretty much got the same device and it not always hits the first time.

niceguy0815, 2020-06-05:
i have re-paired the device now 5 times but no luck.
In Res API it seams to be conectet but there no readings:
`{"config":{"battery":null,"on":true,"pending":[],"reachable":true},"ep":1,"etag":"66b41140c7c06afc55b15ae963b510a6","lastseen":"2020-06-05T16:51:48.312","manufacturername":"Heiman","modelid":"SmokeSensor-EM","name":"Rauchmelder","state":{"fire":null,"lastupdated":"none","lowbattery":null,"tampered":null},"swversion":"1.1.1","type":"ZHAFire","uniqueid":"00:0d:6f:00:14:df:cb:14-01-0500"}`

SwoopX, 2020-06-05:
Let's leave it for 24h and see if it changes. Meanwhile, let me check what my sensors exact model ID is.

niceguy0815, 2020-06-08:
Hi, after leving the device now over 24h, i get now some mor Date from REs API:
`{"config":{"battery":100,"on":true,"pending":[],"reachable":true},"ep":1,"etag":"f1ad53301918f82c948f18cbd434baac","lastseen":"2020-06-08T09:02:21.667","manufacturername":"Heiman","modelid":"SmokeSensor-EM","name":"Rauchmelder","state":{"fire":null,"lastupdated":"none","lowbattery":null,"tampered":null},"swversion":"1.1.1","type":"ZHAFire","uniqueid":"00:0d:6f:00:14:df:cb:14-01-0500"}`

And In FHEM now shows also some readings:
[image]

But the Reading "Fire" is not in the List?

```
Internals:
   .FhemMetaInternals 1
   .triggerUsed 1
   DEF        sensor 4  IODev=deCONZ
   FUUID      5ed25c89-f33f-5d72-acc1-53dfc3466ff76bd6
   FVERSION   31_HUEDevice.pm:0.218370/2020-05-02
   ID         S4
   INTERVAL   
   IODev      deCONZ
   NAME       FIRE_Schlafzimmer
   NR         532
   STATE      Initialized
   TYPE       HUEDevice
   lastupdated 
   lastupdated_local 
   manufacturername Heiman
   modelid    SmokeSensor-EM
   name       Rauchmelder
   on         1
   reachable  1
   swversion  1.1.1
   type       ZHAFire
   uniqueid   00:0d:6f:00:14:df:cb:14-01-0500
   .attraggr:
   .attrminint:
   READINGS:
     2020-06-08 10:17:20   battery         100
     2020-06-08 10:17:20   batteryPercent  100
     2020-06-08 10:17:20   reachable       1
   helper:
     devtype    S
     reachable  0
     update_timeout 1
     configList:
     json:
       ep         1
       etag       f6caad8ebe8f3c5a75a38f7c51a51098
       lastseen   2020-06-08T08:57:21.982
       manufacturername Heiman
       modelid    SmokeSensor-EM
       name       Rauchmelder
       swversion  1.1.1
       type       ZHAFire
       uniqueid   00:0d:6f:00:14:df:cb:14-01-0500
       config:
         battery    100
         pending:
       state:
         fire       
         lastupdated none
         lowbattery 
         tampered   
     setList:
Attributes:
   IODev      deCONZ
   alias      FIRE_Schlafzimmer
   model      SmokeSensor-EM
```

I have tested to temper the smoke detectore with e Zigarret to get a Fire or tempered Reading but no State Update in Res API after triggering the alarm.

I think i miss a configuration step?
do i nead to binde the smoke sensor in deconz manuel to another device / IAS Zone?

SwoopX, 2020-06-08:
Ok, so the battery is solved. Probably just a long update interval (which makes kinda sense).

Now, for the fire sensor, can you please read the IAS zone clusters attributes? You must wake the sensor first by pressing the physical button with a needle or alike. Would like to see if the IAS CIE address has been written and if the zone status is "enrolled".

niceguy0815, 2020-06-08:
[image]
now it is enrolled but at first it was not enrolled and i pushed the exec Button here:
[image]

SwoopX, 2020-06-08:
Please also enter the coordinators MAC address in the CIE field following the format presented.

I would have thought that this behaviour would be gone by now after the updates.

niceguy0815, 2020-06-08:
Like this?
[image]
[image]

SwoopX, 2020-06-08:
Looks right

niceguy0815, 2020-06-08:
Yes, now it workes Thank YOU :D 
after entering the Mac Address in the CIE i waite a wile for a device update but nothing happends.
Than i pressed the Test Button (yes its loud :) ) and after that i get now readings in res API and also in FHEM:

RES:
`

config |  
-- | --
battery | 100
on | true
pending | []
reachable | true
ep | 1
etag | "5cc8bf7520af2f2a5c977158e0f30268"
lastseen | "2020-06-08T23:10:43.684"
manufacturername | "Heiman"
modelid | "SmokeSensor-EM"
name | "Rauchmelder"
state |  
fire | false
lastupdated | "2020-06-08T23:10:39.723"
lowbattery | false
tampered | false
swversion | "1.1.1"
type | "ZHAFire"
uniqueid | "00:0d:6f:00:14:df:cb:14-01-0500"
`
FHEM:

`Internals:
   .FhemMetaInternals 1
   .triggerUsed 1
   DEF        sensor 4  IODev=deCONZ
   FUUID      5ed25c89-f33f-5d72-acc1-53dfc3466ff76bd6
   FVERSION   31_HUEDevice.pm:0.218370/2020-05-02
   ID         S4
   INTERVAL   
   IODev      deCONZ
   NAME       FIRE_Schlafzimmer
   NR         532
   STATE      nofire
   TYPE       HUEDevice
   lastupdated 
   lastupdated_local 
   manufacturername Heiman
   modelid    SmokeSensor-EM
   name       Rauchmelder
   on         1
   reachable  1
   swversion  1.1.1
   type       ZHAFire
   uniqueid   00:0d:6f:00:14:df:cb:14-01-0500
   .attraggr:
   .attrminint:
   READINGS:
     2020-06-09 01:10:39   .lastupdated    2020-06-08 23:10:39
     2020-06-09 01:10:39   .lastupdated_local 2020-06-09 01:10:39
     2020-06-09 01:16:40   battery         100
     2020-06-09 01:16:40   batteryPercent  100
     2020-06-09 01:10:39   batteryState    ok
     2020-06-09 01:10:39   fire            0
     2020-06-09 01:16:40   reachable       1
     2020-06-09 01:10:39   state           nofire
     2020-06-09 01:10:39   tampered        0
   helper:
     devtype    S
     reachable  0
     update_timeout 1
     configList:
     json:
       e          changed
       id         4
       r          sensors
       t          event
       uniqueid   00:0d:6f:00:14:df:cb:14-01-0500
       config:
         battery    100
     setList:
Attributes:
   IODev      deCONZ
   alias      FIRE_Schlafzimmer`

SwoopX, 2020-06-09:
@niceguy0815  Great that it works now. However, I was of the opinion that such a condition would have been gone by now due to the recent code changes.

It would be great if you could stick with me here for some further testing. Would be sufficient to run deconz with `-dbg-info=2 --dbg-aps=2 > dbg` (to have the debug output written to file), reset the device and re-join it.

niceguy0815, 2020-06-09:
Yes i can test it for you.
Do i need to update deconz ore something else before?
than reset device -> delet it from deconz -> close deconz gui and open it again with the parameters.
Than only do the pairing over poscon and see if in deconz all the things are now automaticaly added?

SwoopX, 2020-06-09:
Nothing really needed. The device should automatically leave when you push the pairing button and schould first react with 3 long  light signals. Then release and do it again, it should rapidly start blinking to indicate pairing.

niceguy0815, 2020-06-09:
OK.
I stoped and started deconz via terminal:
[image]

Pushed the pairing button to unpair the sensor.
go to Poscon app -> sensors "add new sensor" -> other -> pushed the pairing button at the sensor again to repair it.
In Deconz it shows not enroled and  has no CIE Address:
[image]

But Res API has Data / Readings:

config |  
-- | --
battery | 100
on | true
pending | []
reachable | true
ep | 1
etag | "50eb71e7b1768c7bafcc9647f29ba763"
lastseen | "2020-06-09T14:25:53.062"
manufacturername | "Heiman"
modelid | "SmokeSensor-EM"
name | "Rauchmelder"
state |  
fire | false
lastupdated | "2020-06-08T23:10:39.723"
lowbattery | false
tampered | false
swversion | "1.1.1"
type | "ZHAFire"
uniqueid | "00:0d:6f:00:14:df:cb:14-01-0500"

do i need to wiat some time?
Where can i find the logfile for you?

SwoopX, 2020-06-09:
You find the log in your fhem folder, where you started deconz.

No need to wait, the mechanism has failed and I'd like to understand why... Also, the `pending` field shouldn't be empty, maybe a rare case missed.

niceguy0815, 2020-06-09:
have done it again here is the logfile.
I hope it helps..
[dbg.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4752903/dbg.txt)

SwoopX, 2020-06-09:
Thanks. Very strange that I do not see anything which should have taken care of the automatic enrollment. Got to double check the code for it.

I'll let you know when I got some new insights.

SwoopX, 2020-06-10:
@niceguy0815 I really don't get it. If it's not asked to much, could you please delete the fire sensor via API and then reset/re-pair with the debug enabled? Eventually, the remnant sensor is the root cause for this.

niceguy0815, 2020-06-10:
Hi, yes no Problem.
Have deleted the sensor from deconz and reset it and re-paired it.
[dbg2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4758167/dbg2.txt)

SwoopX, 2020-06-11:
Thanks again. This time, some messages of interest are there, but unfortunately not those I was hoping for. I also don't see that the device has left the network so it got a physical reset.

Anyway, the solution for a not enrolled sensor is as explained: write the IAS CIE address of the coordinator (needs to be waken/triggered before). In 99% of the cases, that should trigger an enroll request from the device, so directly read the IAS zone cluster attributes again. If it still has not enrolled, wake/trigger it again and send an IAS enroll response command (as you did).

closed 2020-07-10 by stale[bot]