dresden-elektronik/deconz-rest-plugin #2819: please add it to supported devices IMMAX 4in1-Senzor-ZB3.0

apmediask, 2020-05-25:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

apmediask, 2020-05-27:
Hi,
3 out of 4 devices were found, PIR sensor is missing(0406), please help
[image]

apmediask, 2020-06-07:
does not work

apmediask, 2020-06-07:
does not work

SwoopX, 2020-06-18:
Missed your reply. PIR is handled via IAS zone cluster. 

Try the below while sensor seach is running:
Select it, weak the device and read the attributes. Thas should trigger auto-enrollment. Double check while waiting like 10 secs, then do the procedure again.

If that's still no good, double click on the `IAS CIE address` attribute and write the coordinator MAC address while the sensor is awake. Again, wait like 10 secs and then wake the sensor and read the attributes. If the sensor has the address previously entered but is still not enrolled, execute the command above the read button while the sensor is awake. Check once again after 10 secs.

WhistleMaster, 2020-07-14:
I also have an Immax 4in1 device. When integrated in Phoscon App, the device is not visible in "Connected sensors" list.

> If that's still no good, double click on the `IAS CIE address` attribute and write the coordinator MAC address while the sensor is awake. Again, wait like 10 secs and then wake the sensor and read [...]

I had to follow this procedure but in Home Assistant, the presence sensor does not update. The others (temperature, humidity, light) are fine though. Maybe an integration issue @Kane610 ?

Kane610, 2020-07-14:
I don't think it is an integration issue, if it gets created it should also handle updates, I think deconz doesn't send updates

WhistleMaster, 2020-07-14:
> I don't think it is an integration issue, if it gets created it should also handle updates, I think deconz doesn't send updates

Thanks for jumping in ! If the sensor only use "IAS Zone" cluster, how is that reflected in HA vs the "0406 Occupancy sensing" ?

SwoopX, 2020-07-14:
The IAS Zone status is 99% still not enrolled. Has the write of the CIE address been accepted?

WhistleMaster, 2020-07-14:
Seems to be:
[image]

SwoopX, 2020-07-14:
Doesn't it work at all or just occasionally? It should take like 60 secs to jump from motion to nomotion (if there's no motion in between).

WhistleMaster, 2020-07-14:
No, it does not work at all. Temperature, humidity and light are fine though.

SwoopX, 2020-07-14:
Strange. In the IAS Zone, the `Alarm 1` should be ticked upon motion, but only if it's enrolled... Not sure if the device uses a different mechanism for the notifications.
[image]

WhistleMaster, 2020-07-14:
This is what I have:
[image]

Do I just need to check "Alarm 1" and click on "exec" ?

SwoopX, 2020-07-14:
No, you just need to generate motion, then it should update automatically. Not sure if you need to switch back and furth between the clusters to have it visible.

WhistleMaster, 2020-07-14:
I've tried to remove the sensor from Deconz through the VNC interface and add it again. Now I have the following:

[image]

But still no update in Home Assistant for the PIR only...

WhistleMaster, 2020-07-14:
Also, the sensor is not visible in the "Connected sensors" list in Phoscon App and I've tried to look at the log and I can't find any ZHAPresence updates for the sensor.

I have other Philips sensors and I can see that they are updated in the log when there are some movements.

WhistleMaster, 2020-07-15:
@SwoopX & @apmediask Do those sensors work on your side ? Can you see it in the list in the Phoscon App ?

Mimiix, 2020-07-15:
@WhistleMaster Please don't ping users on such short notice. They all (even me) get emails about responses here. Give them some time :)

To answer one question:

If you see devices within deCONZ but not in Phoscon, that is most of the time perfectly normal. Phoscon is closed source and needs devices to be added manually by their Developer. However, the device will still be usable with the API/Home-Assistant/OpenHAB etc. 

Swoop will reply when he can :)

SwoopX, 2020-07-15:
@WhistleMaster Do you run HA? Point is I need to understand what the device is sending and that might be a bit hard if I cannot see the traffic by sniffing then.

WhistleMaster, 2020-07-15:
Yes, I do run HA. The issue is that even in Deconz log, there is no sign of motion update (I have other Philips sensors and I can see that they are updated in the log when there are some movements).

The device does not send anything.

SwoopX, 2020-07-15:
And this is why sniffing the traffic would provide more insights in how far the device maybe behaves differently from others I got some experimental code for that but since you run HA, you cannot use it.

WhistleMaster, 2020-07-15:
Sure, I can send you the device if you need ? I can also send by paypal the amount to buy it. Tell me what would be the easiest for you.

SwoopX, 2020-07-15:
Ping me on our Discord and we figure something out ;)

SwoopX, 2020-08-06:
Bump. Still no feedback from Immax support.

petr-vytlacil, 2020-10-07:
Hi, try somebody connect conbee 2 with this sensor in Phoscon App? When I searching sensor, I dont find it. THX

Mimiix, 2020-10-12:
@petr-vytlacil It's not integrated yet.
We are lacking info.

SwoopX, 2020-10-13:
It is. However, it seems to have some issues for the motion sensor. Got one here now and no idea, why it is so stubborn or what the device issue is.

vahaldor, 2020-10-29:
I have this sensor for couple of days and it detected motion first day, then stopped working. Is this deconz dependent problem?

petr-vytlacil, 2020-10-29:
@vahaldor Hi, can you write me if you use Conbee 2? Beacause i dont see sensor in Deconz and i use Conbee 2. THX

vahaldor, 2020-10-29:
I use ConBee II firmware 26660700, plugged in Rasp pi 4 running hassio. It worked for one or two days, now I am experiencing exactly the kind of behaviour others discuss here.

vahaldor, 2020-10-29:
I received a support case from IMMAX. They think this is a faulty unit... Petr would you write me directly? We can discuss is in czech nikola.schmidt@gmail.com Thanks

petr-vytlacil, 2020-10-29:
@vahaldor Hi, thank you. I have update ConBee II to 26660700 and sensos was detected. THX

ACrazyConcept, 2021-02-28:
I just got this sensor thinking all issues were solved.
It pairs pretty easily. But at first the motion detection was not working. I had it laying around for a couple of days where temp, humidity and lightlevel all updated fine. Then I tried deleting in deCONZ GUI end search for new sensor in Phoscon and afterwards manually reading attributes in deCONZ. Well not super methodical really. But somehow the motion sensor started working.
And it did work. I could also decrease the 'duration' from 60 seconds to 15 seconds with REST command, and that also worked.
However now I just noticed that the temp, humidity and lightlevel had not been updating since I got the motion sensor working. 😅

SwoopX, 2021-03-01:
From my perspective, those devices seem to be cursed. I got one donated by a user and never had so much trouble to get a device fully working as this one here. Everything else than the motion detection does its job, but that thing doesn't send anything on zigbee level although properly set up and configured.

I'm tempted to either suspect a faulty device or firmware...

ACrazyConcept, 2021-03-27:
I think I have figured out a fairly reliable way to get the (my) motion sensor working. Sadly when the motion sensor works, none of the other sensors do.

In any case these are the steps:

Delete the sensor from Phoscon if it is still there.
Start a search for the sensor.
Press and hold the device button untill it starts pairing.
When the sensor has been paired go to deCONZ.
Open the IAS Zone cluster.
Double click the IAS_CIE_Address attribute.
Click read.
Press the device button once.
Close the IAS_CIE_Address attribute window.
Double click the Zone ID attribute.
Press the device button once.
Close the Zone ID attribute window

The IAS Zone cluster attributes then look like this for me:
[image]

After a few seconds motion should now get detected in Phoscon (and Home Assistant)
I am also still able to set the duration option to 15 seconds instead of the default 60 seconds.

If I start poking around in deCONZ and e.g. read the environment sensor clusters the motion sensor will stop responding again and the only way I have been able to get it back is by starting all over as described above.

ACrazyConcept, 2021-05-16:
Is there anything I can do to shed some light on the workings of this device?

I recently paired it again and had the motion detection working (but not the environment sensors). But then after some time I checked and then it was the other way around and all the environment sensors had sent data for several days, but no motion detected..

ACrazyConcept, 2021-06-17:
Still hoping this can be made to work :)

SwoopX, 2021-06-21:
I might have found out how to make that da** thing fully working. Gotta do some more testing to verify that wasn't a one hit wonder...

ACrazyConcept, 2021-06-22:
@SwoopX sounds great! Let me know if I can assist in some way.

ACrazyConcept, 2021-07-02:
> 
> 
> I might have found out how to make that da** thing fully working. Gotta do some more testing to verify that wasn't a one hit wonder...

Any progress on it so far @SwoopX ?

jkalousek, 2021-11-13:
I also bought this sensor before finding out about this issue. I do not really care much about the temperature and luminance reading but I found out that motion sensing does something weird in my case. 
When motion happens, both Alarm 1 in deconz and motion in Home assistant are "on" but if I wave my hand in middle of cooldown period (60s) in front of sensor, home assistant will report no motion after cooldown period from initial/first motion but Alarm 1 in deconz will stay "checked" and will stay on until cooldown from last detected motion (as it should be). So what happens is that home assistant motion state is not the same as state of Alarm 1 and is only triggered when Alarm 1 is switched from off to on but always set to "no motion" after cooldown regardless of motion in front of sensor. In the end if I stay in room constantly moving for 5 minutes home assistant will only see first minute whereas doconz correctly shows expected status. My other sensors works as expected so there is something special about how events are handled by this one.
Is there some additional info that I can provide so we can figure out what can cause this problem?

Mimiix, 2021-11-13:
I think this was fixed in the last beta

jkalousek, 2021-11-13:
@Mimiix Ok, thank You. I will wait for next release.