dresden-elektronik/deconz-rest-plugin #2820: Tradfri TW lights got broken in 5.3.3

HenningGithub, 2020-05-26:


HenningGithub, 2020-05-26:
Sorry, missed description.
Ikea Tradfri tuneable white E27 lights got broken in the latest release.
Scenes that apply xy_color used to work but now produce an error.

2020-05-25 13:54:21 ERROR (MainThread) [homeassistant.components.script] autolights_forceon: Error executing script. Unexpected error for call_service at pos 3: /lights/27/state parameter, xy, not available
[...]
pydeconz.errors.RequestError: /lights/27/state parameter, xy, not available

Setting color_temp instead works around the problem.
The GU10 and E14 versions of the ikea tuneable white lights are affected as well. Same for a Müller Light one. That would be all tuneable white ones I use.

RGB lights of multiple vendors show no issues.

morfei1, 2020-05-26:
Hello! What you mean by latest release... .76 or .77?

HenningGithub, 2020-05-26:
5.3.3 of the deconz-HA-plugin, which includes deconz 2.05.76 according to changelog.
5.3.4 with 2.05.77 shows exactly the same behaviour.

ebaauw, 2020-05-26:
I added a sanity check to PUTting the light state that you can only set attributes that are actually exposed by the light resource.

Afaik the tunable whites (_Color temperature lights_) don't expose `xy`.  I'm surprised they even respond to _Move to Color_ (the command generated on setting `xy`), they advertise that they don't (through _Color Capabilities_) and through the device type.

HenningGithub, 2020-05-28:
I cannot provide anything to counter your surprise - there is a regression in my setup, and it didn't seem intentional, so I report. It is easy enough to work around.
FWIW, there also is a lot of new logspam like:
pydeconz.errors.pydeconzException: /lights/25/state parameter, bri, is not modifiable. Device is set to off.
with a traceback upfront. I don't see how that is providing any insight.
Thanks for all the great work!

closed 2020-05-28 by HenningGithub

MC-1984, 2020-05-29:
Although the issue is closed, ct still doesn't work (again) with .77. According to [https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2804](url) it should work again on Tradfri TW but for me it isn't. 

I am not sure how to set the color if ct isn't working and xy is not exposed?

ebaauw, 2020-05-29:
Please list the `/lights` resource, the REST API call, and the error (if any).

MC-1984, 2020-05-29:
Can you provide information on how to collect these information (e.g. a link)? I am using ioBroker and I am not familiar with the REST API itself.

MC-1984, 2020-05-29:
I can provide the error message from the ioBroker log:
`

deconz.0 | 2020-05-30 00:22:54.391 | warn | (2576) Code 400: Bad request set light state 4: [{"error":{"address":"/lights/4/state","description":"invalid value, 500, for parameter, ct","type":7}}]
-- | -- | -- | --

`

ebaauw, 2020-05-30:
You’re trying to set `ct` to a value that the light doesn’t support.  I think the Trådfri TW lights support values from 250 to 454.

MC-1984, 2020-05-30:
It used to work over months perfectly with a value of 500. But you are absolutely right. It works perfectly in the range from 250 to 454 now. Thank you!

ebaauw, 2020-05-30:
> It used to work over months perfectly with a value of 500.

Because of the practically non-existent parameter checking and error handling in the API.  I want to fix that, one API call at a time, starting with PUT of the light state.

I'll double check how the Hue bridge handles this.  I could also coerce the value requested to the supported range, and report back the coerced value.

I'm also in discussion with @manup why the deCONZ API sends error codes in these cases, instead of 200 like the Hue bridge.  Same for the "error" when trying to change the light state while the light is off.

MC-1984, 2020-05-30:
Thanks for the explanation!