dresden-elektronik/deconz-rest-plugin #2821: Phoscon, doubleclick events for Xiaomi WXKG02LM Rev.2

sjostrand, 2020-05-26:
ConBee on Raspberry 3b+
Firmware: 26350500
Version: 2.05.77
OS: Raspbian Buster
Switch: Xiaomi Aqara WXKG02LM Rev.2

I'm trying to set up actions for double click (and maybe even hold). But I cannot find a way to do in the switch editor. 
I verified in HA that events are being sent (1001, 1002, 1004, 2001, 2002, 2004, 3001....) when clicking, double clicking and holding, so it's probably just be that the settings are not available in the switch editor?

SwoopX, 2020-05-28:
Phoscon repor has been reopened. You may want to raise it there and close the issue here. Thanks!

StephanStS, 2020-05-30:
You find it there: https://github.com/dresden-elektronik/phoscon-app-beta/issues.