dresden-elektronik/deconz-rest-plugin #2822: Philips Hue Dimmer Pairing Issues

meremortals70, 2020-05-26:
Picked up a new Hue Dimmer switch, US style, as they are sold in Australia.

Running the Deconz add-in for Home Assistant

When I try to pair it via the normal method in HA ie. DeConz>>Switch>>Add Switch following the on screen instructions, nothing happens in Home Assistant.

When I go and have a look at Deconz in VNC, I can see a new device has been added however I do not get a second button on the top right of the device.

[image]

I have reset the device and tried to pair again with no result.

I have deleted the device and restarted deconz and tried to pair.

I even tried pairing it within a few cm from the Conbee II

[image]

Nothing seems to be working for me.

Any suggestions please?

Superwuzy, 2020-05-26:
Maybe there is an issue with the newest deConz version. I've experienced problems pairing new devices using the newest software from 05/22/2020.

meremortals70, 2020-05-30:
Update

I was finally able to add the dimmer to deconz.  Since the updates it seems like everytime you need to add a new device it is best to reboot deconz first.

However the only attribute I see in the deconz add in for HA is battery.

How to I program the buttons and associate it with lights?

meremortals70, 2020-05-30:
Deleted device from Deconz and re-pair.  It added a group in Phoscon this time and I get a new group for the Deconz Group in HA for the dimmer.  Still cant do anything with the buttons though.

Mimiix, 2020-06-12:
@meremortals70 What does the API report?

meremortals70, 2020-06-12:
Oh I gave up and downgraded to .75

When the issues are fixed I might upgrade but .76 & .77 were an absolute disaster

Mimiix, 2020-06-12:
I have no issues with my dimmer switch here 🤷

meremortals70, 2020-06-12:
Did you pair it whilst running .76 or .77 or was it paired using a previous version?

Mimiix, 2020-06-12:
It was paired, but i can re-pair later if it helps you?

meremortals70, 2020-06-12:
See not the same. Please don’t make glib comments to people who are having problems

Mimiix, 2020-06-12:
Let me remove it later today and see what happens on re-pairing. 

Just trying to help here..

meremortals70, 2020-06-12:
Well so far in the last 10 days you have closed quite a lot of my posts whilst misconstruing the issue and saying it’s the same as other problems when it wasn’t

You even asked me to close one and then admitted it wasn’t the same issue.

All with not providing any positive advice or comments

closed 2020-06-12 by meremortals70

SwoopX, 2020-06-12:
@meremortals70 let's try to reset all our communication and leave emotions aside. We understand you're having issues, so let's try to get those sorted out together. However, what really helps is a more interactive way of communication. Therefore, I may suggest you join Discord if you not already have and that tag me in the devices channel.

Please note that we havily relie on you providing debug and information. The root cause could have a variety of reasons, so we probably need to dig deep. I'm even happy to offer a teamviever session to expedite if that might help.

Does that sound workable?

Mimiix, 2020-06-12:
Lets sum them up:
-  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2722 > Got a result and closed due to fixed
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2721 > You closed 
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2896 > I asked you to keep open, because one part phoscon one part is a genuine issue. 
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2895 > Closed as is duplicate
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2885 > Closed yourself
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2863 > Closed because there's a general thread for it 
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2850 > Product advice. There's plenty of other places
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2831 > Lacked response of you.
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2823 > Still open
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2777 > Closed yourself

So really, i only see one thread that i can see why you might think this. Other than that, your really making a problem out of nothing...

meremortals70, 2020-06-13:
@SwoopX everything was going fine until @Mimiix decided to take control of the issues and started closing and classifying based on his own cognitive bias.

Seems to me that no one is taken ownership for the extensive issues with the .76/.77 build.

Is deconz api dead?

@Mimiix if you want my cooperation to solve stuff please engage with me don't make sweeping comments without reading all the information.

Otherwise I will drop deconz from my solution and go back to all the forums I had previously supported deconz whilst others were bagging and unequivocally backtrack.

If you want to be helpful drop the attitude.

Mimiix, 2020-06-13:
@meremortals70 I'm trying to see and figure out what still is not solved and what is solved. Going trough 450 issues is not easy and it is easy for me to miss stuff. So if you feel that i closed stuff , please comment on those or reach out to me on Discord.

However, if you feel i have a attitude towards you: I'm sorry, that is not the way i like to come accross. 

On topic: I really do want to see people work together and solve a issue. This one no different. In this particular issue i were trying to solve your issue by testing with my own environment to see if i can reproduce. I like to keep problems within their issue and solve them each on their own. Step by step. If you don't want my help or want me to try stuff, how can i help to get your issues solved?

sunshineautomate, 2021-07-05:
I'm also struggling to pair my new Philips dimmer switch. Is there a trick to this at all?