dresden-elektronik/deconz-rest-plugin #2826: Phoscon Switches and Sensors are empty - Wireless Light without function

duffbeer2000, 2020-05-26:
Hey guys,
after the update to V2_05_76 and now V2_05_76 the sensors and switches in the web frontend are empty, it says that I don't have any of them. By the REST-API I see them and they are working fine. If I switch to the old wirless light the Control tap is empty, Groups says "ERROR! Lost connection, retry in one second ...", Devices says "Error! Loading data, retry in three seconds ..." 
After the update to V2_05_76 I resetted my password (my caps lock was on and I recogniced it to late ;) ) 

Regards
Christian

SwoopX, 2020-05-26:
A restart of deconz should solve it.

Michaelnorge, 2020-05-26:
Try to install the update again, it helped me.

duffbeer2000, 2020-05-26:
I'm running it in the marthoc docker container. I already tried to restart the container but it didn't solves the problem
What I see in the log are those messages

```
19:35:44:027 New websocket 192.168.178.12:60836 (state: 3)
19:35:53:091 Remove websocket 192.168.178.12:60836 after error Unknown error"
```

And I'm not sure if it was with the 75 too but I see a lot of "no button map" entries for different devices:
`19:37:40:863 no button map for: TRADFRI open/close remote ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
19:37:40:863 ZCL attribute report 0x086BD7FFFE22BF09 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:38:03:785 no button map for: TRADFRI open/close remote ep: 0x01 cl: 0x0001 cmd: 0x0A pl[0]: 021
19:38:03:785 ZCL attribute report 0x000D6FFFFE7A745B for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:38:25:124 Current channel 11
19:38:25:156 Device TTL 1212 s flags: 0x7
19:38:25:124 Current channel 11
19:38:25:156 Device TTL 1212 s flags: 0x7
19:38:26:453 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
19:38:26:453 ZCL attribute report 0x00158D0002D43D4B for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
19:38:26:485 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
19:38:26:485 ZCL attribute report 0x00158D0002D43D4B for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000
19:38:31:768 no button map for: SPZB0001 ep: 0x01 cl: 0x0201 cmd: 0x0A pl[0]: 002
19:38:31:768 ZCL attribute report 0x00158D0001FF8CD0 for cluster: 0x0201, ep: 0x01, frame control: 0x1C, mfcode: 0x1037
19:38:31:800 no button map for: SPZB0001 ep: 0x01 cl: 0x0201 cmd: 0x0A pl[0]: 008
19:38:31:800 ZCL attribute report 0x00158D0001FF8CD0 for cluster: 0x0201, ep: 0x01, frame control: 0x18, mfcode: 0x0000
19:38:37:057 don't close database yet, keep open for 900 seconds
19:38:37:852 no button map for: SPZB0001 ep: 0x01 cl: 0x0201 cmd: 0x0A pl[0]: 003
19:38:37:852 ZCL attribute report 0x00158D0001FF8CD0 for cluster: 0x0201, ep: 0x01, frame control: 0x1C, mfcode: 0x1037
19:38:43:071 don't close database yet, keep open for 900 seconds
19:38:43:936 no button map for: SPZB0001 ep: 0x01 cl: 0x0201 cmd: 0x0A pl[0]: 008
19:38:43:936 ZCL attribute report 0x00158D0001FF8CD0 for cluster: 0x0201, ep: 0x01, frame control: 0x1C, mfcode: 0x1037
19:39:15:281 no button map for: FYRTUR block-out roller blind ep: 0x01 cl: 0x0102 cmd: 0x0A pl[0]: 008
19:39:15:281 ZCL attribute report 0x000D6FFFFE30687B for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:39:21:765 no button map for: FYRTUR block-out roller blind ep: 0x01 cl: 0x0102 cmd: 0x0A pl[0]: 008
19:39:21:765 ZCL attribute report 0xD0CF5EFFFE7A0F97 for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:39:22:053 no button map for: FYRTUR block-out roller blind ep: 0x01 cl: 0x0102 cmd: 0x0A pl[0]: 008
19:39:22:053 ZCL attribute report 0xD0CF5EFFFEEA93CA for cluster: 0x0102, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:39:25:126 Current channel 11
19:39:25:159 Device TTL 1152 s flags: 0x7
19:39:25:126 Current channel 11
19:39:25:159 Device TTL 1152 s flags: 0x7
19:39:27:160 no button map for: lumi.sensor_magnet.aq2 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
19:39:27:160 ZCL attribute report 0x00158D0002E738AE for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000
19:39:27:160 no button map for: lumi.sensor_magnet.aq2 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
19:39:27:160 ZCL attribute report 0x00158D0002E738AE for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000
19:39:33:051 no button map for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021
19:39:33:051 ZCL attribute report 0x0017880102C023DD for cluster: 0x0001, ep: 0x02, frame control: 0x18, mfcode: 0x0000`

duffbeer2000, 2020-05-29:
[There is the solution](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2843#issue-627029927)

closed 2020-05-29 by duffbeer2000