dresden-elektronik/deconz-rest-plugin #2827: All deconz versions higher than deconz-2.05.69 wont start on my Conbee II

henne1977, 2020-05-26:
I have a Conbee II running running latest firmware "deCONZ_ConBeeII_0x26580700.bin.GCF"

When I update to a newer version than beta 2.05.69-qt5 ,  I am not able to start the deconz service again (using: "sudo systemctl start deconz"). It is the same problem with the latest stable.

The status says:
" root@raspberrypi:~# sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-05-26 16:43:34 CEST; 15s ago
  Process: 4495 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=1/FAILURE)
 Main PID: 4495 (code=exited, status=1/FAILURE)
   CGroup: /system.slice/deconz.service "

If I go back to 2.05.69-qt5 it is starting again. Does any of you have an idea what I am doing wrong?

Smanar, 2020-05-26:
Hello, how are you updating your deconz version ?

henne1977, 2020-05-26:
sudo systemctl stop deconz
wget http://deconz.dresden-elektronik.de/raspbian/beta/deconz-xxx.deb
sudo dpkg -i deconz-xxx.deb
sudo systemctl start deconz

Smanar, 2020-05-26:
I see nothing strange.
Have you try to update your system ?
Have you take a look in the syslog ?

henne1977, 2020-05-26:
This is what the syslog says:

May 26 22:23:45 raspberrypi systemd[1]: Stopping deCONZ: ZigBee gateway -- REST API...
May 26 22:23:46 raspberrypi deCONZ[7040]: PROTO: CRC error
May 26 22:23:46 raspberrypi systemd[1]: deconz.service: main process exited, code=exited, status=143/n/a
May 26 22:23:46 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
May 26 22:23:46 raspberrypi systemd[1]: Unit deconz.service entered failed state.
May 26 22:24:20 raspberrypi systemd[1]: [/lib/systemd/system/deconz-gui.service:11] Unknown lvalue 'StartLimitIntervalSec' in section 'Service'
May 26 22:24:20 raspberrypi systemd[1]: [/lib/systemd/system/deconz-gui.service:13] Unknown lvalue 'AmbientCapabilities' in section 'Service'
May 26 22:24:20 raspberrypi systemd[1]: [/lib/systemd/system/deconz.service:9] Unknown lvalue 'StartLimitIntervalSec' in section 'Service'
May 26 22:24:20 raspberrypi systemd[1]: [/lib/systemd/system/deconz.service:11] Unknown lvalue 'AmbientCapabilities' in section 'Service'
May 26 22:24:21 raspberrypi systemd[1]: Stopping deCONZ: ZigBee gateway -- WIFI Service...
May 26 22:24:21 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- WIFI Service...
May 26 22:24:21 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- WIFI Service.
May 26 22:24:28 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
May 26 22:24:28 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- REST API...
May 26 22:24:28 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/bin/deCONZ)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Network.so.5: no version information available (required by /usr/bin/deCONZ)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/bin/deCONZ)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/bin/deCONZ)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/bin/deCONZ)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5: no version information available (required by /usr/bin/deCONZ)
May 26 22:24:28 raspberrypi systemd[1]: deconz.service: main process exited, code=exited, status=1/FAILURE
May 26 22:24:28 raspberrypi systemd[1]: Unit deconz.service entered failed state.
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
May 26 22:24:28 raspberrypi deCONZ[15315]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/lib/libdeCONZ.so.1)
May 26 22:24:28 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
May 26 22:24:59 raspberrypi systemd[1]: deconz.service holdoff time over, scheduling restart.

Smanar, 2020-05-27:
Just to test can you try

>sudo systemctl stop deconz-gui
sudo systemctl stop deconz

then
>sudo systemctl start deconz

SwoopX, 2020-05-27:
The messages indicate missing packages. Try `sudo apt -f install`

henne1977, 2020-05-27:
> The messages indicate missing packages. Try `sudo apt -f install`

Thank you, that has been leading me to the solution! 
I used the "sudo apt update" and "sudo apt install deconz" way of updating instead of the manual way I described above. 
If you follow the instructions here it works:  https://phoscon.de/de/conbee2/install#raspbian

closed 2020-05-27 by henne1977