dresden-elektronik/deconz-rest-plugin #2828: Unable to update firmware of Conbee II

roeleboel, 2020-05-26:
Hi,
I am having problems with updating the firmware on my conbee II stick, it times out _every time_ after +-3 seconds, leaving me with a de facto bricked stick...

System:
ubuntu 18.04 on intel nuc
stick inserted in external usb2 hub

log:
```
sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60  -f /home/roel/Downloads//deCONZ_ConBeeII_0x26580700.bin.GCF
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 160930 bytes: |==============error: timeout flashing firmware after 3007 ms
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 160930 bytes: |======error: timeout flashing firmware after 3002 ms
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
Reboot device /dev/ttyACM0 (ConBee II)
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 160930 bytes: |=======error: timeout flashing firmware after 3005 ms
Reboot device /dev/ttyACM0 (ConBee II)
retry, failed

2139: Error: uart reset failed, check retry
```

this log is repeatable, I have tried 50+ updates in the meantime.

What can I do?

Smanar, 2020-05-26:
Here you have a solution https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493

You need to make a firmware update with a "-t" parametre to increase the timeout.

BTW where are you from ?

roeleboel, 2020-05-26:
In the command I use there is already '-t 60'
The intel nuc is a 'real' pc, running ubuntu 18.04
I am from Belgium :)

Smanar, 2020-05-26:
Oups, I haven't see your command, my bad, I have just read the symptom.

But in this situation, I m sorry, with real PC (and usb 2.0 Hub), you realy have the better condition possible ...

roeleboel, 2020-05-26:
unfortunately -t 120 did not change anything. also plugging directy into the nuc does not help.
I think I can consider this stick to be bricked yes?

Smanar, 2020-05-26:
It have already worked ?
But I think you can contact the support yes ....

peer69, 2020-05-26:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1440. 
Try `systemctl stop modem-manager` before starting the Update.

roeleboel, 2020-05-26:
@peer69 Yes! Thank you very much, this did the trick!
Firmware is updated & stick works as it should!

peer69, 2020-05-26:
This for sure should be somewhere in the wiki then.

Smanar, 2020-05-26:
Sure, thx a lot, realy good to know.

It's something only present on ubuntu ?

peer69, 2020-05-26:
I have only „tested“ this on Ubuntu 18.04.

peer69, 2020-05-27:
I have added this to the wiki page: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually. 
Might be obsolete in future versions of deCONZ or Ubuntu but for now adding it there might provide a solution faster than digging through issues or creating a new one.

closed 2020-05-27 by roeleboel

MickL, 2022-06-18:
I stopped the MoemManager and still get the same error:

> flashing 127650 bytes: |======error: timeout flashing firmware after 3000 ms

```sh
pi@homebridge:/tmp $ sudo GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee

pi@homebridge:/tmp $ sudo GCFFlasher_internal -t 60 -f deCONZ_Rpi_0x26400500.bin.GCF -d /dev/ttyS0
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyS0 (RaspBee)
Reboot device /dev/ttyS0 (RaspBee)
Reboot device /dev/ttyS0 (RaspBee)
action: reset device RaspBee
wiringPi 2.52 initialized
Reboot device /dev/ttyS0 (RaspBee)
action: reset device RaspBee
RaspBee Bootloader premium
Vers. 1.02
uild 2013/ (52 ms)
uild 2013/08/01
 (52 ms)
uild 2013/08/01
R (390 ms)
uild 2013/08/01
RE (390 ms)
uild 2013/08/01
REA (391 ms)
uild 2013/08/01
READ (391 ms)
flashing 127650 bytes: |=======error: timeout flashing firmware after 3000 ms
```