dresden-elektronik/deconz-rest-plugin #2830: Conbee II worked once, now firmware not connected

NickWaterton, 2020-05-26:
I just got a new conbee II, and set it up in a Ubuntu 18.04 VM. Connected to a USB 2.0 powered hub by an extension cable.

Everything worked well, was able to pair a ST multipurpose sensor with no problems. I then installed zigbee2mqtt, which also worked with no issues.

Great!

Then I moved on to a Phillips tap switch, got nowhere with that, so I changed the pan_id in zigbee2mqtt to 0xffff. After that, nothing works at all.

I have tried everything, (unloading all software and reinstalling, power cycling etc.) but the deconz web interface still says 'firmware disconnected' (originally it gave the version number), and I cannot get a response. the conbee II lists OK:
```
nick@Ubuntu-18-04:~$ sudo GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2147290  | ConBee II
```
but it times out flashing firmware.
In desperation I loaded up a new RPi 3B+, installed deconz, but still have the same 'firmware disconnected' message, and I can't pair anything (even a new, unused sensor). I can however flash the firmware successfully. I have tried old and new firmware, with no success. I have tried with extension/without extension cable. There is nothing else plugged into the RPi.
```
pi@raspberrypi:~ $ sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x26580700.bin.GCF
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 264A0700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
unlock!  (6223 ms)
flashing 160930 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```
I have ordered another conbee II just in case mine has died for some reason.
This is the log on `deconz` startup:
```
May 26 17:01:05 raspberrypi kernel: [ 2105.260509] usb 1-1.3: USB disconnect, device number 18
May 26 17:01:06 raspberrypi kernel: [ 2105.559211] usb 1-1.3: new full-speed USB device number 19 using dwc_otg
May 26 17:01:06 raspberrypi kernel: [ 2105.693674] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
May 26 17:01:06 raspberrypi kernel: [ 2105.693690] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 26 17:01:06 raspberrypi kernel: [ 2105.693700] usb 1-1.3: Product: ConBee II
May 26 17:01:06 raspberrypi kernel: [ 2105.693709] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
May 26 17:01:06 raspberrypi kernel: [ 2105.693719] usb 1-1.3: SerialNumber: DE2147290
May 26 17:01:06 raspberrypi kernel: [ 2105.694909] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
May 26 17:01:06 raspberrypi mtp-probe: checking bus 1, device 19: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
May 26 17:01:06 raspberrypi mtp-probe: bus: 1, device: 19 was not an MTP device
May 26 17:01:06 raspberrypi mtp-probe: checking bus 1, device 19: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
May 26 17:01:06 raspberrypi mtp-probe: bus: 1, device: 19 was not an MTP device
May 26 17:01:24 raspberrypi systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
May 26 17:01:24 raspberrypi systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
May 26 17:01:24 raspberrypi systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
May 26 17:01:24 raspberrypi systemd[1]: /lib/systemd/system/deconz-gui.service:10: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
May 26 17:01:31 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
May 26 17:01:31 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
May 26 17:01:31 raspberrypi deCONZ[5997]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
May 26 17:01:31 raspberrypi systemd[1]: deconz-init.service: Succeeded.
May 26 17:01:31 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
May 26 17:01:31 raspberrypi deCONZ[5997]: libpng warning: iCCP: known incorrect sRGB profile
May 26 17:01:32 raspberrypi deCONZ[5997]: This plugin does not support propagateSizeHints()
May 26 17:01:32 raspberrypi deCONZ[5997]: This plugin does not support propagateSizeHints()
May 26 17:02:01 raspberrypi deCONZ[5997]: This plugin does not support propagateSizeHints()
```

Do you have any suggestions on what further I can try?
It worked so well out of the box! then the next day - dead...

I have run out of ideas.

Smanar, 2020-05-27:
Hello, just by curiosity can you try

>$ ps ax | grep deCONZ

NickWaterton, 2020-05-27:
I just downloaded the RPi pre-built disk image to try, but same thing.

```
pi@phoscon:~ $ ps ax | grep deCONZ
  307 ?        Ssl    1:58 /usr/bin/deCONZ -platform minimal --http-port=80
  324 ?        Ss     0:06 /bin/bash /usr/bin/deCONZ-update2.sh
  340 ?        Ss     0:18 /bin/bash /usr/bin/deCONZ-WIFI2.sh
10396 pts/1    S+     0:00 grep --color=auto deCONZ
```
the syslog gives these errors now
```
May 27 11:30:09 phoscon systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
May 27 11:30:09 phoscon systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 85.
May 27 11:30:09 phoscon systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 27 11:30:09 phoscon systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 27 11:30:09 phoscon hostapd[3921]: Configuration file: /etc/hostapd/hostapd.conf
May 27 11:30:09 phoscon hostapd[3921]: Interface name not specified in /etc/hostapd/hostapd.conf, nor by '-i' parameter
May 27 11:30:09 phoscon hostapd[3921]: : interface state UNINITIALIZED->DISABLED
May 27 11:30:09 phoscon hostapd[3921]: : AP-DISABLED
May 27 11:30:09 phoscon hostapd[3921]: : CTRL-EVENT-TERMINATING
May 27 11:30:09 phoscon hostapd[3921]: hostapd_free_hapd_data: Interface  wasn't started
May 27 11:30:09 phoscon hostapd[3921]: Failed to initialize interface
May 27 11:30:09 phoscon systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
May 27 11:30:09 phoscon systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 27 11:30:09 phoscon systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 27 11:30:12 phoscon systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
May 27 11:30:12 phoscon systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 86.
May 27 11:30:12 phoscon systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
May 27 11:30:12 phoscon systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
May 27 11:30:12 phoscon hostapd[3934]: Configuration file: /etc/hostapd/hostapd.conf
May 27 11:30:12 phoscon hostapd[3934]: Interface name not specified in /etc/hostapd/hostapd.conf, nor by '-i' parameter
May 27 11:30:12 phoscon hostapd[3934]: : interface state UNINITIALIZED->DISABLED
May 27 11:30:12 phoscon hostapd[3934]: : AP-DISABLED
May 27 11:30:12 phoscon hostapd[3934]: : CTRL-EVENT-TERMINATING
May 27 11:30:12 phoscon hostapd[3934]: hostapd_free_hapd_data: Interface  wasn't started
May 27 11:30:12 phoscon hostapd[3934]: Failed to initialize interface
May 27 11:30:12 phoscon systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
May 27 11:30:12 phoscon systemd[1]: hostapd.service: Failed with result 'exit-code'.
May 27 11:30:12 phoscon systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
```
Which is new, but I'm thinking is more to do with the gateway that anything else.

Smanar, 2020-05-27:
You are using this image on a real pi or a VM ?

NickWaterton, 2020-05-27:
I tried it on both.
I am focusing on the Pi for now (as it's more convenient).
I finally got the gui interface working (had to edit the DISPLAY variable), and can connect to the conbee II, but the interface just says `joining...` at the top. As far as i can tell, it's supposed to switch to `leave` or `join`, but it's just stuck at `joining...`.

In the network config screen `Predefined PAN ID` is selected and set to `0xffff`. I can change and save it, but after a reboot of the conbee II, it reverts back to `0xffff` and I don't seem to be able to do anything. Firmware flashing etc. makes no difference.

It looks like the internal NVRAM has some invalid settings, but I can't erase/reset the internal NVRAM.

Any suggestions?

SwoopX, 2020-05-27:
Are you trying to run deconz and zigbee2mqtt in parallel on the same Conbee?

NickWaterton, 2020-05-27:
No,
I initially set up with deconz, then when that worked, I stopped (and disabled) deconz, I loaded zigbee2mqtt, and that worked.

I then changed the `pan_id` in zigbee2mqtt to 0xffff in an attempt to get a phillips tap button to pair. After that, everything stopped working.

I tried a lot of things, finally giving up on my VM for the time being, and set up a completely vanilla RPi, with the deconz-disk image as a test bed. There is nothing else loaded on, or connected to this pi 3B+.

I can get deconz loaded, and deconz-gui up, but the conbee II is stuck at `joining...` and if I change the network parameters, they just revert back to what you see in the screen shot after I reboot the conbee II.

deconz is the only thing loaded on this Pi.

SwoopX, 2020-05-27:
ok. So let's stop deconz service with `sudo systemctl stop deconz` and then start it again like that `/usr/bin/deCONZ -platform minimal --http-port=80 --dbg-info=2 --dbg-error=2 > debug`. Let it run for a minute or so, that should provide more debug output in that file.

Are you using LAN to connect? Looks like there's a wifi issue...

ebaauw, 2020-05-28:
> Then I moved on to a Phillips tap switch, got nowhere with that, so I changed the pan_id in zigbee2mqtt to 0xffff. After that, nothing works at all.

Why would you do that?  Did you take a note of the previous _PAN ID_?

> In the network config screen Predefined PAN ID is selected and set to 0xffff.

Duh, that’s what you set it to using z2m.  These settings are stored in NVRAM on the device.  Also note the weird value for _APS use ext pan id_

> I can flash firmware etc. but none of that make any difference.

No, firmware update doesn’t touch NVRAM.

> I can get deconz loaded, and deconz-gui up, but the conbee II is stuck at joining... 

Pretty sure that 0xFFFF is not a valid PAN ID.

> and if I change the network parameters, they just revert back to what you see in the screen shot after I reboot the conbee II.

The “normal” procedure to change these NVRAM parameters is change, save, leave network, join, read back.  Best set both _PAN ID_, _APS use ext PAN ID_, and _Network Key_ to all 0, and clear the _Use predefined PAN ID_.  That should cause the ConBee II to create a new network.

NickWaterton, 2020-05-28:
> Then I moved on to a Phillips tap switch, got nowhere with that, so I changed the pan_id in zigbee2mqtt to 0xffff. After that, nothing works at all.

> Why would you do that? Did you take a note of the previous PAN ID?

Previous PAN ID was 0x1a62, I changed it to 0xffff because that is the broadcast PAN ID, and the Phillips Tap switch uses the `Green_Power` profile that broadcasts on PAN ID 0xffff.

> In the network config screen Predefined PAN ID is selected and set to 0xffff.

> Duh, that’s what you set it to using z2m. These settings are stored in NVRAM on the device. Also note the weird value for APS use ext pan id

Yes I know, z2m also sets the APS use ext pan id. I didn't think that there would be no way to ever change them back to other numbers though.

> I can flash firmware etc. but none of that make any difference.

> No, firmware update doesn’t touch NVRAM.

So the question remains, how do I reset the NVRAM to factory default?

> I can get deconz loaded, and deconz-gui up, but the conbee II is stuck at joining...

> Pretty sure that 0xFFFF is not a valid PAN ID.

It is the broadcast PAN ID.

> and if I change the network parameters, they just revert back to what you see in the screen shot after I reboot the conbee II.

> The “normal” procedure to change these NVRAM parameters is change, save, leave network, join, read back.

This is where I am stuck. I can't join, so how can I leave?

> There's a hidden page in Phoscon somewhere to restore the network settings in case they are corrupted. I think ctrl-click on the Settings. If your old settings aren't available there, best set PAN ID, APS use ext PAN ID, and Network Key to all 0, and clear the Use predefined PAN ID. That should cause the ConBee II to create a new network.

I tried that, the hidden page doesn't show (it's alt-click on settings), and if I change the values as you suggest, Network key reverts to `0x1030507090b0d0f00020406080a0c0d` on saving, and the other values revert back on reboot.

This is the debug output:
```
pi@phoscon:~ $ /usr/bin/deCONZ --http-port=80 -dbg-info=2 --dbg-error=1
libEGL warning: MESA-LOADER: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)

libEGL warning: MESA-LOADER: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)

libEGL warning: MESA-LOADER: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)

libEGL warning: MESA-LOADER: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)

libpng warning: iCCP: known incorrect sRGB profile
22:04:40:925 Otau image path does not exist: /home/pi/otau
22:04:41:965 UPNP socket not bound, state: 0
22:04:46:095 GW update firmware not found:
22:04:48:917 CTRL got nwk update id 0
22:04:49:048 CTRL got nwk update id 0
23:03:53:290 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
23:03:54:340 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
23:03:55:390 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
23:03:56:440 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
23:03:57:491 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
23:03:58:514 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
23:04:13:319 CTRL got nwk update id 0
23:04:13:434 CTRL got nwk update id 0
00:03:59:514 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
00:04:00:514 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
00:04:01:514 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
00:04:02:514 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
00:04:03:515 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
00:04:04:515 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
00:04:19:421 CTRL got nwk update id 0
00:04:19:550 CTRL got nwk update id 0
01:04:05:291 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
01:04:06:341 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
01:04:07:392 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
[... 74 lines cut ...]
10:05:04:015 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:05:19:395 CTRL got nwk update id 0
10:05:19:515 CTRL got nwk update id 0
11:05:05:014 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:05:06:014 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:05:07:015 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:05:08:014 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:05:09:014 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:05:10:014 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
11:05:25:379 CTRL got nwk update id 0
11:05:25:492 CTRL got nwk update id 0
12:05:11:292 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
12:05:12:341 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
12:05:13:392 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
12:05:14:443 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
12:05:15:492 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
12:05:16:515 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
12:05:31:321 CTRL got nwk update id 0
12:05:31:479 CTRL got nwk update id 0
12:33:04:960 CTRL got nwk update id 0
12:33:24:397 CTRL got nwk update id 0
12:33:24:519 CTRL got nwk update id 0
12:33:31:825 CTRL got nwk update id 0
12:34:20:318 CTRL got nwk update id 0
```

NickWaterton, 2020-06-09:
I got a new conbee II which works fine. I think the old one is unrecoverable, due to being unable to reset the NVRAM - it's stuck on a pan_id of ffff.

So I'm junking the unrecoverable one.

closed 2020-06-09 by NickWaterton