dresden-elektronik/deconz-rest-plugin #2831: Manually Upgrading Conbee II Firmware

meremortals70, 2020-05-27:
I have searched high and low for a guide to manually upgrade the firmware on the Conbee II for my implementation

Ubuntu Linux>>HA installed in Docker>>Deconz add-in for HA.

Nothing seems to explain upgrading the firmware when you are running the Deconz add-in on HA where HA is in a docker container. Not a separate docker container for Deconz.

Can anyone help?

stickpin, 2020-05-27:
@meremortals70, you can follow this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

I've done an upgrade to my Conbee II two days ago using the Docker method and everything went well.

meremortals70, 2020-05-27:
Sorry, that will not work.

It is for the marthoc's deCONZ Docker image

not the add-on installed in HA which is a docker image.

That is a separate docker image for deconz standalone.

stickpin, 2020-05-27:
@meremortals70 ok.
If you running HASS.IO, you can install the Portainer, but please be careful, you can destroy your setup if you will do something wrong with your dockers.

In the Portainer go to the `Settings` - `Settings` and remove all the filters from `Hidden containers`.
Now go to the `Containers`, you would see `addon_core_deconz`, press `exec console` button and connect using `/bin/bash`.

Now execute the following:
- `/usr/bin/GCFFlasher_internal -l`
You would see list of the devices. Something like this:
[image]
- `ls -1 /usr/share/deCONZ/firmware` to see list of available firmware in this docker. Should look something like this:
[image]

Now be careful and make sure you choose the correct firmware and correct device.
If you sure the next command should be:
- `/usr/bin/GCFFlasher_internal -d <device_path> -f /usr/share/deCONZ/firmware/<firmware_filename>`

So if you want to flash Conbee II, your command should look like this:
- `/usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26580700.bin.GCF`

Please enter and pray. :)

I hope it helps.
You can remove Portainer Addon after that if you don't need it.

Smanar, 2020-05-27:
And the methos explained on the marthoc's deCONZ Docker Github ?
https://github.com/marthoc/docker-deconz

stickpin, 2020-05-27:
> And the method explained on the marthoc's deCONZ Docker Github ?
> https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

Your access to `bash` is limited once you running managed Hass.IO image. So you cannot do what is explained in this link.

Smanar, 2020-05-27:
Ha ok, my bad I have read he use a container for "deconz", not for the "deconz HA add on"

Mimiix, 2020-06-12:
As there was no response from @meremortals70 on this, i'll be closing this up!

meremortals70, 2020-06-13:
There was no response because no one was actually listening to my problem.

Mimiix, 2020-06-13:
@meremortals70 : @stickpin gave some comprehensive input on your issue. I'm happy to reopen if you need some more debugging on this with him.

meremortals70, 2020-06-13:
Actually no because I am not running that docker image as previously explained

Mimiix, 2020-06-13:
As far as i know, the Addon in HA does run in a seperate containers.  I'm still on the old firmware, i'll try his method later today.

meremortals70, 2020-06-13:
Can’t paste an image here as I am currently on an iPad. However I only have three containers in my docker and they are HA, Hassio multicast and assistant relay.

With the Mathic image, you create and container by loading the image.

What I have is I loaded the HA image and then added the Deconz add-in from supervisor.

Mimiix, 2020-06-13:
I've had that too(i'm trying now), but did you go to settings and remove the labels? That made deconz show up for me.

meremortals70, 2020-06-13:
I think this is mute for the moment as I have downgraded to .75 because .76 and .77 are so troublesome

meremortals70, 2020-06-13:
But what do you mean by removing the labels

Mimiix, 2020-06-13:
@stickpin Wrote: 

**In the Portainer go to the Settings - Settings and remove all the filters from Hidden containers.**

I just did that, but not i'm getting another error on trying to flash. Not sure how to fix that though. I'm using a Nas so that complicates things in terms of usb devices. Maybe its trough that, i don't know. 

I've created a issue for that:  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2935

oat1, 2020-07-08:
After the last step i recive:
/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)

How to solve this ?

> @meremortals70 ok.
> If you running HASS.IO, you can install the Portainer Addon from the Addons Store, but please be careful, you can destroy your setup if you will do something wrong with your dockers.
> 
> In the Portainer go to the `Settings` - `Settings` and remove all the filters from `Hidden containers`.
> Now go to the `Containers`, you would see `addon_core_deconz`, press `exec console` button and connect using `/bin/bash`.
> 
> Now execute the following:
> 
> * `/usr/bin/GCFFlasher_internal -l`
>   You would see list of the devices. Something like this:
> 
> [image]
> 
> * `ls -1 /usr/share/deCONZ/firmware` to see list of available firmware in this docker. Should look something like this:
> 
> [image]
> 
> Now be careful and make sure you choose the correct firmware and correct device.
> If you sure the next command should be:
> 
> * `/usr/bin/GCFFlasher_internal -d <device_path> -f /usr/share/deCONZ/firmware/<firmware_filename>`
> 
> So if you want to flash Conbee II, your command should look like this:
> 
> * `/usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26580700.bin.GCF`
> 
> Please enter and pray. :)
> 
> I hope it helps.
> You can remove Portainer Addon after that if you don't need it.

> @meremortals70 ok.
> If you running HASS.IO, you can install the Portainer Addon from the Addons Store, but please be careful, you can destroy your setup if you will do something wrong with your dockers.
> 
> In the Portainer go to the `Settings` - `Settings` and remove all the filters from `Hidden containers`.
> Now go to the `Containers`, you would see `addon_core_deconz`, press `exec console` button and connect using `/bin/bash`.
> 
> Now execute the following:
> 
> * `/usr/bin/GCFFlasher_internal -l`
>   You would see list of the devices. Something like this:
> 
> [image]
> 
> * `ls -1 /usr/share/deCONZ/firmware` to see list of available firmware in this docker. Should look something like this:
> 
> [image]
> 
> Now be careful and make sure you choose the correct firmware and correct device.
> If you sure the next command should be:
> 
> * `/usr/bin/GCFFlasher_internal -d <device_path> -f /usr/share/deCONZ/firmware/<firmware_filename>`
> 
> So if you want to flash Conbee II, your command should look like this:
> 
> * `/usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26580700.bin.GCF`
> 
> Please enter and pray. :)
> 
> I hope it helps.
> You can remove Portainer Addon after that if you don't need it.

After last step, I'm receiving 

/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)

How to solve this ?

Mimiix, 2020-07-08:
@oat1 Please open your own Bug report for this :)

SwoopX, 2020-07-08:
The messages seems to indicate missing packages. Try `sudo apt -f install`

Mimiix, 2020-07-08:
@meremortals70 Did you manage to upgrade yet?

meremortals70, 2020-07-11:
@Mimiix no I haven't and I have dropped the Deconz add-on for ZHA in HA

Mimiix, 2020-07-11:
Okido! Then I will proceed and close this issue :)

closed 2020-07-11 by Mimiix

bgmn, 2020-07-28:
> @meremortals70 ok.
> If you running HASS.IO, you can install the Portainer Addon from the Addons Store, but please be careful, you can destroy your setup if you will do something wrong with your [...]

Hi!

I tried upgrading with portrainer but it gives me the following error
[image]

If I try to kill deCONZ process, the whole container will reboot.
I have supported VMDK running on my Synology NAS. Conbee II is working otherwise just fine.

Mimiix, 2020-07-29:
@bgmn As per documentation (and what i wrote before) : VM's are not claimed to be supported to update the device upon. Please refer to [This page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-update-manually).

bgmn, 2020-07-29:
> @bgmn As per documentation (and what i wrote before) : VM's are not claimed to be supported to update the device upon. Please refer to [This page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-update-manually).

Thank you for the quick reply. I was hoping that it would be possible to update them easily via docker container but it's not so big task to update them manually on another computer.

Mimiix, 2020-07-29:
@bgmn You can, but not within a VM. 

VM's add some complexity to serial connections.

Harry-1976, 2020-09-15:
> @meremortals70 ok.
> If you running HASS.IO, you can install the Portainer Addon from the Addons Store, but please be careful, you can destroy your setup if you will do something wrong with your [...]

Thanks for this guide. It assumes the firmware is already in the /firmware folder. How do i pull the latest firmware to this folder with Portainer?

Mimiix, 2020-09-15:
@Harry-1976 They should come with the latest versions. HA isn't putting those in because of their own reasons. You should open a issue in their Addon repository for this question.

stickpin, 2020-09-15:
@Harry-1976 just go to the folder:
`cd /usr/share/deCONZ/firmware/`
and then do:
`wget http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_<FIRMWARE_YOU_NEED>.bin.GCF`

and then it supposed to work fine. :)

Harry-1976, 2020-09-15:
@stickpin Thanks! That worked.
Update failed though.
Now to find a solution for the 781 error (same error as @bgmn posted) i got.

stickpin, 2020-09-15:
@Harry-1976 it seems that you running on some other kind of virtualization.
I am also lately moved to vmware esxi approach and also was not able to upgrade the firmware today. 

So my solution was to use some old windows machine to update my Conbee II.

I hope it helps.

Harry-1976, 2020-09-15:
@stickpin Yes, that's how i use to do it. I guess i'll have to stick to it for now.

Cheers.

Mimiix, 2020-09-16:
@Harry-1976 @stickpin  upgrading on virtual installations is not supported / recommended.

Harry-1976, 2020-09-17:
@Mimiix I understand that Mimiix. I’m running with the standard pi4 installation which i believe is not a virtual installation.