dresden-elektronik/deconz-rest-plugin #2833: Conbee 2 firmware Update via Phoscon

EduardSarjadjuk, 2020-05-27:
Hello, 
i have a quastion. 
Why is it not possible to update the firmware 0x26580700 for the Conbee 2 via Phoscon?
Does this option follow?
I would appreciate any feedback.
The manual installation (see Wiki) unfortunately does not work for me.
Firmware version 0x26580700 with routing fixes and stability improvements included for manual installation for ConBee II.

stickpin, 2020-05-27:
Hi @EduardSarjadjuk,

If you running HASS.IO, may my comment in parallel thread can help: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2831#issuecomment-634674928

Smanar, 2020-05-27:
Why the manual installation don't work ?
You have an error message ?

SwoopX, 2020-05-27:
> Why is it not possible to update the firmware 0x26580700 for the Conbee 2 via Phoscon?
> Does this option follow?

This will probably follow with the next release. However, what do you expect on this? If the manual update has already failed, so will it by pushing a button in Phscon most likely as well.

Maybe we can help troubleshooting anyway if you provide some further information as suggested.

meremortals70, 2020-05-28:
Thanks @SwoopX 

If we could have it in Phoscon we would not need to manually update.

My expectation if I have an update process within the Phoscon Add-On. It knows where its root directory is and can work everything else from there

EduardSarjadjuk, 2020-05-28:
The firmware update does not work via my Linux system (Proxmox VM with Debian).
When I start with the firmware update I get the message: Conbee Stick 2 reboot + (old firmware).
After that nothing happens and I have to restart the VM.
Is it not possible to update the firmware via a virtual machine?

Regardless of that, I was now able to import the update via Windows. There were no problems here?

peer69, 2020-05-28:
I successfully flashed Conbee IIi on Ubuntu 18.04 virtual machine using VMWare. Shouldn't be a general problem with virtualised systems.

SwoopX, 2020-05-28:
Eventually, this is related to the virtualization one uses. In my experience, VMware always played nice in contrast to others, e.g. virtual box