dresden-elektronik/deconz-rest-plugin #2836: Rest API does report wrong time

yves19, 2020-05-27:
I just mentioned this today.

The time reported by "lastseen" item in end point or routers in wrongly reported compared to the one reported by "localtime":  in the GW config .

The time reported elsewhere than in GW config seems to be GMT time whatever time has been configured in the GW .
For instance there are currently 2 hours shift between France and GMT. 

This would be fine if the time reported in API was computed with the same Time Zone as the one defined in the GW config.

regards

ebaauw, 2020-05-27:
All times reported by the API, except `/config/localtime` are in UTC.   I want to add `Z` to the value to take away any ambiguity, but am a bit worried that might break some clients.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1687.

yves19, 2020-05-28:
Thank you for your quick answer.
Yes it would be more "mathematically" correct if the Time Zone was added to the value returned so tier solution that rely on deCONZ API can manage local time vs UTC time.

closed 2020-05-28 by yves19