dresden-elektronik/deconz-rest-plugin #2837: Philips Hue app complains about "old version bridge"

joeblack2k, 2020-05-27:
When successfully adding Deconz to the official Hue app, the app gives a warning about an update for the bridge that fails 
The latest version of the official bridge is:
1938112040

is it possible to change the emulated Deconz version to 1938112040

townsmcp, 2020-05-28:
+1 for this. The Hue app cant be used until theres an update

hmrc87, 2020-06-02:
I just installed the Hue App because I wanted to test if works with the Windows App "Hue Sync".

Have the same issue - but I am unsure: Was the official Hue App working with deconz before? I never tried it before so cant answer.

How did you find out the emulated Hue Bridge Version of Deconz?

kurzdev, 2020-06-06:
Same here. While trying to connect to the emulated bridge, this is what occurs:
[image]

Would be nice to know either how to possibly fix this so that one can contribute (possibly how older versions have been fixed), or have it fixed. Thank you! 😄

Smanar, 2020-06-06:
You haven't the version gived by deconz ?
1810251352 ?

kurzdev, 2020-06-06:
The hue app needs version 1938112040 to function, i.e. to be used at all. @Smanar

Smanar, 2020-06-06:
Yep but IDK wich one deconz is using.
Perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/rest_configuration.cpp#L990

If you test with changing this value ? Or if you can retreive the value gived by deconz to be sure it's the good line ?

kurzdev, 2020-06-06:
Right, let me check - do you have any idea how to retrieve the version from the API? Otherwise I will try to reverse engineer as good as I can 😄

Smanar, 2020-06-06:
Nope :(, and it seem the code don't answer the same thing depending of the request type.

But you can try to edit the value and recomplile the code, it s easy if you have a linux machine (and not HA or ARM 64)

kurzdev, 2020-06-06:
Yeah I guess I'll do that tomorrow. Thanks so much for your help already, will let you know when I've made any progress or at least know more 👍

kurzdev, 2020-06-06:
@Smanar Ok, I just figured it out. The trick is to set the devicetype while requesting an API key to a name that starts either with "Hue " or "hue_" as seen here:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/31ffad99da46a7a86423567f96ca5022453e2157/authorisation.cpp#L200

After that the the ApiRequestMode ist set to ApiModeHue which is not ApiModeNormal and thus forces the API to output the swversion upon request on http://yourdevice/api/your_api_key/

After doing that, you should find the line "swversion": "1810251352" which is exactly what we are looking for. Tomorrow, I will either try to change the corresponding lines of code or will try to implement a proxy that modifies said data, and will come up with a comprehensive report on possible results. 👍

ebaauw, 2020-06-06:
After setting `swversion`, `apiversion`, and `datastoreversion` to the values of the current Hue bridge gen-2 firmware, the Hue app connects to deCONZ v2.05.77.  I've included these changes in my current PR.

kurzdev, 2020-06-06:
@ebaauw Awesome, thanks so much!

fonix232, 2020-06-30:
Would it be possible to have a user-settable field for this, that is pre-populated with the hardcoded value, instead of just a hardcoded value? It would mean people won't need to wait for the next update to use their Hue apps again, but could manually update the firmware on their end.

pattyland, 2020-07-01:
On my phone the hue app still wants to install an update that I can’t dismiss. I’m using 2.05.78

fonix232, 2020-07-02:
@pattyland - I had the same issue, however it looks like something is broken even deeper - removing all the Hue app data allowed me to set up the virtual bridge, but it was incredibly buggy, I couldn't get away from the room management screen in any way, rooms were not synced from deCONZ (though if I did try to create a room with the same name, it would freak out), rooms wouldn't save (either generic error, or the room seemingly saves, but it disappears in a matter of seconds).

thedoginthewok, 2021-02-11:
This issue is still relevant, the problem still exists.

closed 2021-02-12 by close-issue-app[bot]