dresden-elektronik/deconz-rest-plugin #2840: Gledopto ct colormode stopped working

Jolle666, 2020-05-28:
Hi, 

Since deCONZ v2.05.76 ct colormode stopped working with my Gledopto lights (GL-C-006 and GL-C-008). API error in Node-RED/Home Assistant. 2.05.75 is fine. 

Maybe it's related to [#2798](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2798)?

Thanks!

SwoopX, 2020-05-28:
Update to .77, there's a fix for ct as I recall

ebaauw, 2020-05-28:
Only for IKEA.  What type of GLEDOPTOs are these?  Could you list the resources and screenshots of the _Color Control_ cluster (after reading the attributes)?

Jolle666, 2020-05-28:
The Gledopto GL-C-006 is the WW/CW LED controller (firmware 1.0.6). The GL-C-008 is the RGB+CCT LED controller (firmware 2.0.0).

I run Home Assistant with the Phoscon app on a RPI4 with Conbee II. I'm sorry, but not sure where to find resources and Color Control cluster.

ebaauw, 2020-05-28:
Hm, both should expose `state.ct` and work?!  Did you happen to try .77?

What API error do you see?  Can you find out what API request are sent?

ebaauw, 2020-05-28:
Double-checked the `GL-C-008`: it's working in 2.05.77:
```
$ ph get /lights/5
{
  "ctmax": 65535,
  "ctmin": 0,
  "etag": "34b94379dede7d32b61d6d6d480b7b11",
  "hascolor": true,
  "lastseen": "2020-05-28T19:55:36.214",
  "manufacturername": "GLEDOPTO",
  "modelid": "GL-C-008",
  "name": "GL-C-008",
  "state": {
    "alert": "none",
    "bri": 1,
    "colormode": "hs",
    "ct": 153,
    "effect": "none",
    "hue": 0,
    "on": true,
    "reachable": true,
    "sat": 0,
    "xy": [
      0.3125,
      0.3333
    ]
  },
  "swversion": "1.0.6",
  "type": "Extended color light",
  "uniqueid": "00:12:4b:00:1b:91:ec:83-0b"
}
$ ph put -v /lights/5/state '{"ct": 500}'
{
  "ct": 500
}
$ ph put -v /lights/5/state '{"ct": 153}'
{
  "ct": 153
}
```

Jolle666, 2020-05-28:
@ebaauw thank you so much.

Just testet once more. This is the JSON in Node-RED:
[image]

With 2.05.75 the LED controller is working and with no error in Node-RED:
[image]

With 2.05.77 the LED controller is not working and I get this API error in Node-RED:
[image]
with this debug message:
[image]

This is the info in the Phoscon app:
[image]

ebaauw, 2020-05-28:
Ah OK: you cannot set `colormode`: it's a read-only attribute.  It's set automatically when you set the colour (temperature) trough `xy`, `hue` and `sat` (`hs`), or `ct`.

It has never done anything, but was silently ignored.  I fixed the handling of errors for a PUT of the light state, so you're only seeing the error since v2.05.76.

Jolle666, 2020-05-28:
@ebaauw Ah, now it all makes sense and is working perfect. Thank you so much!

closed 2020-05-28 by Jolle666