dresden-elektronik/deconz-rest-plugin #2841: Privacy Issue: Gateway of another party revealed due to shared public AP Adress on Dual-Stack Lite Internet Access

MarcProe, 2020-05-28:
When I open Phoscon in the browser, I can see 2 different gateways, one of which is not on my network and not reachable in my network.

[image]

The ``192.x.x.x`` gateway is defintely **not** on my network. 

Using the REST-API https://dresden-light.appspot.com/discover, I receive the following (redacted):
```
[
  {
    "id":"00212EFFxxxxxxxx",
    "internalipaddress":"10.x.x.x",
    "macaddress":"00212EFFxxxxxxxx",
    "internalport":80,
    "name":"Phoscon-GW",
    "publicipaddress":"185.a.a.a"
  }, 
 {
    "id":"00212EFFyyyyyyyy",
    "internalipaddress":"192.y.y.y.y",
    "macaddress":"00212EFFxxxxxxxx",
    "internalport":80,
    "name":"Phoscon-Some-Family-Name",
    "publicipaddress":"185.b.b.b"
  }
]
```

**Here, the public IPv4-Addresses are identical.**

I am on a Fiber Link with [DS-Lite](https://avm.de/service/fritzbox/fritzbox-7590/wissensdatenbank/publication/show/1611_Was-ist-DS-Lite-und-wie-funktioniert-es/), wich means that I share a public IPv4-Address with several other 
customers of the ISP.

It seems the gateways are registered with there public IPv4-Addresses and this is the primary and only key with which they are identified.

So the internal Information of the other persons Phoscon Gateway are exposed to me.
 * The family name of the person (as they named their gateway that way)
 * The internal network information of the person

Also, the unknown person will most likely see this information from me.

I consider this a severe privacy issue that should be fixed ASAP.

ebaauw, 2020-05-28:
> It seems the gateways are registered with there public IPv4-Addresses and this is the primary and only key with which they are identified.

Yes.

> Also, the unknown person will most likely see this information from me.

Yes.

Note that this is exactly the same mechanism as used by the Hue bridge.

> I consider this a severe privacy issue that should be fixed ASAP.

I doubt that that's going to happen anytime soon.  In the meantime, you can disable the announcement by setting `config.discovery` to false.  You need to specify `Accept: application/vnd.ddel.v1` in the http header to unhide this hidden attribute.  Easiest is to use `ph`:
```
ph -p put /config '{"discovery": false}'
```
Note that it might take an hour or so, before the server removes the old announcement.

Note that Phoscon doesn't use UPnP for discovery, so it will no longer discover your gateway.  You need to specify it in the url instead: http://phoscon.de/pwa/login.html#host/192.x.x.x:80

MarcProe, 2020-05-28:
Thank you for the workaround.

dsjoho, 2020-05-29:
Is there any documentation of the hidden attributes?
I'm specifically interested in any other phoning-home settings, and what/how deconz writes to storage and settings to change it.

For anyone else just googling this, here's a standard way to turn it off without finding the ph-binary:

# Get API KEY, returned username is the API-key
curl -X POST localhost/api --data '{"devicetype": "turn off discovery"}'
# Disable discovery, replace API_KEY with the returned username
curl -H 'Accept: application/vnd.ddel.v1' -X PUT localhost/api/API_KEY/config --data '{"discovery": false}'

ebaauw, 2020-05-29:
> Is there any documentation of the hidden attributes?

UTSL:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/720a324f33dcb31efc086f2f052b35eba3948794/rest_configuration.cpp#L1567

MarcProe, 2020-05-29:
> Is there any documentation of the hidden attributes?

You can also ``GET /api/KEY/config``

ebaauw, 2020-05-29:
That only returns the non-hidden attributes; you need to specify the Accept header to see the hidden ones.

Mimiix, 2020-06-10:
Can we close this issue :)?

MarcProe, 2020-07-03:
Now I can already see 2 foreign systems...

[image]

holdestmade, 2020-07-06:
I had this same issue a while ago, and at the same time some of my lights were randomly turning on and off.
Not seen any additional gateways for a few months now, seems to coincide with me installing and running a VPN ?

MarcProe, 2020-07-27:
I can see 3 foreign systems now!
[image]

andriej, 2020-08-18:
Stale bot seems to be the only one who care.

holdestmade, 2020-08-18:
> I'm running the HA add-on, how do I turn off the discovery ? Thanks

N5A, 2020-10-04:
172.X is a class B network, used by Docker, if you have that installed it can be coming from there.

169.X is a TCPIP auto-configure IP Address, when no DHCP server exists the TCPIP stack falls back and randomly grabs a 169.X address and uses it. If it collides with another active 169 it drops off, grabs another and gets back on.

192.168's are Class C's used by home routers by default.  They are not web routable, so you just won't see anyone elses that are not in your actual network. If you and a neighbor share a router then yes,  you'll see theirs. There is nothing that can be done about that unless you control the router and its a high end one with internal VPN capability and can set up your own VPN within it to isolate your gear from the neighbors.

I will bet ALL of these are yours just found at different stages and are just stuck in Phoscon's memory.

manup, 2020-10-05:
The issue of showing foreign gateways or non responding gateways should not appear anymore. There was a bug in the discovery server, fixed a while ago. The discovery can be enabled/disabled via REST-API configuration endpoint.

```
PUT /api/<apikey>/config

{
    "discovery": false
}
```

For example by using curl:

`curl -X PUT -H "Content-Type: application/json" -d '{"discovery":false}' 192.168.1.2:8080/api/<apikey>/config`

Note: 192.168.1.2:8080 needs to be replaced by your deCONZ gateway IP address and port. As `apikey` you can use the one from the Phoscon App &rarr; in your Browser open developer console (F12) and enter `apikey`.

MarcProe, 2020-10-05:
@N5A 
> I will bet ALL of these are yours just found at different stages and are just stuck in Phoscon's memory.

No.

@manup 
Thanks for confirming. It's been awhile since I checked, currently I cannot see any foreign systems.

closed 2020-10-05 by MarcProe