dresden-elektronik/deconz-rest-plugin #2843: IKEA TRÅDFRI open/close button - shutter

duffbeer2000, 2020-05-29:
Hey together,
I recognized that after the update from V2_05_75 to V2_05_77 my IKEA TRÅDFRI open/close buttons stopped working. Everytime I press the button I see in the log:
no button map for: TRADFRI open/close remote ep: 0x01 cl: 0x0102 cmd: 0x01 pl[0]: 000

ebaauw, 2020-05-29:
Could you list the ZHASwitch `/sensors` resource for the remote?  What system are you running deCONZ on?

duffbeer2000, 2020-05-29:
I'm running the marthoc docker image on an amd64 system.
I think I found the error! I updated my Buttons with the latest firmware two weeks ago, after the update the manufacturename is shown like this:
[image]
I already updated the firmware again but the manufacturename still is not Ikea and contains illegal charakters.

That's from the API:
    "40": {
        "config": {
            "alert": "none",
            "battery": 74,
            "group": "7",
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "54189c69875e97d9a2795e4a3b579610",
        "lastseen": "2020-05-29T08:46:12.865",
        "manufacturername": "KE",
        "mode": 1,
        "modelid": "TRADFRI open/close remote",
        "name": "TRÅDFRI auf/zu Taster - Rollos",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2020-05-13T13:21:25.000"
        },
        "swversion": "2.2.008",
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:ff:fe:7a:74:5b-01-1000"
    },

"47": {
        "config": {
            "alert": "none",
            "battery": 74,
            "group": "9",
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "2eec9d9b0697776e83251de7987f023c",
        "lastseen": "2020-05-29T08:55:35.477",
        "manufacturername": "KE",
        "mode": 1,
        "modelid": "TRADFRI open/close remote",
        "name": "TRÅDFRI auf/zu Taster - Wohnung",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2020-05-13T18:58:17.000"
        },
        "swversion": "2.2.008",
        "type": "ZHASwitch",
        "uniqueid": "08:6b:d7:ff:fe:22:bf:09-01-1000"
    }

duffbeer2000, 2020-05-29:
Is there a way to fasten the firmware update? Packet Spacing maybe?

ebaauw, 2020-05-29:
Indeed, the code assigning the button map checks that the `manufacturername` starts with `"IKEA"`.

I don't think that's the latest firmware.  I'm on 2.2.010 and the `manufacturname` is normal (see below).

Did you re-read the _Basic_ cluster attributes after updating the firmware?

```
$ ph get /sensors/511
{
  "config": {
    "alert": "select",
    "battery": 100,
    "group": "316",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "0ad26fcadd5c844958e1be0314828725",
  "lastseen": "2020-05-29T09:33:46.812",
  "manufacturername": "IKEA of Sweden",
  "mode": 1,
  "modelid": "TRADFRI open/close remote",
  "name": "Bathroom Blind Control",
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2020-05-25T21:56:33.284"
  },
  "swversion": "2.2.010",
  "type": "ZHASwitch",
  "uniqueid": "00:0d:6f:ff:fe:e3:f4:90-01-1000"
}
```

duffbeer2000, 2020-05-29:
Yes I re-read the Basic cluster already a few times. Everytime with the same response.

duffbeer2000, 2020-05-29:
I got it! Had to start a sensor search without deleting it, reset the switch and wait 2-3 minutes. After that it worked. (If you don't wait a little bit the info by cluster reading is still wrong)
Shouldn't it show now the correct firmware version? In the OTAU Plugin it shows 0x22010631 after a query, that should be 2.2.010 but in the API it shows 2.2.008

ebaauw, 2020-05-29:
You need to read the attribute separately on battery-powered devices.  Only the first couple of attributes (up to _Date Code_) are read on _Read_.  Double-click the attribute for the pop-up window.

duffbeer2000, 2020-05-29:
Ah, that did the trick! Do you know why the firmware update for philips devices is realy fast even a battery powered dimmer but the ikea switches are really slow?

ebaauw, 2020-05-29:
The update rate is dictated by the device.  And further slowed by how many hops the device is from the coordinator.  When the coordinator is the parent the update might go quicker.

duffbeer2000, 2020-05-29:
Perfect, thank you for your assistance! I'll close the Issue.

closed 2020-05-29 by duffbeer2000

bosjp, 2021-12-02:
hello,

i'm using phoscon gateway on raspbee II version 2.12.06 
I discovered two buttons but they don't give event when I use it

attributes seams correct 
[image]

can you help me ?

thx 

jean-pierre

Mimiix, 2021-12-02:
They are broken since last firmware update of ikea. Will be fixed in 2.13.3

bosjp, 2021-12-02:
ok many thanks for your swift answer...
do you have an idea about the timing ?

Mimiix, 2021-12-02:
Half december.

bosjp, 2021-12-03:
great thanks a lot !