dresden-elektronik/deconz-rest-plugin #2847: Is there a list of devices which can be use as router or repeater?

xinyang-creates, 2020-05-29:
One of the selling point for ConBee II is "Signal range", however, no devices I have at the moment can be used as a repeater. I have some wired wall switch from Aqara but they are not recognized as either routers or switchs. Currently, the Zigbee coverage in the house is pretty bad. It causes sensors have a slow response and sometime no response at all. 

I would love to add some repeater in the house but I am not sure which device I should get to extend the signal coverage. Is there a list of devices which can be use as router or repeater?

What's the differences between router and repeater in the following context?

> Devices further away are connected through the Zigbee Mesh network. In this network, all mains powered Zigbee devices, such as lights and sockets, act as repeaters and can route the signal.

ebaauw, 2020-05-29:
Any full-feature device (FFD) acts as a Zigbee router.  These are the yellow nodes in the GUI.  Typically these are mains-powered: lights, wired switches, plugs.  The word "repeater" is typically used for routers that have no other function.

Smanar, 2020-05-29:
Aquara wired switch with neutral/zero line act as router.

xinyang-creates, 2020-05-30:
Thanks everyone. Closing the issue.

closed 2020-05-30 by xinyang-creates