dresden-elektronik/deconz-rest-plugin #2848: Setting action value not working properly

MC-1984, 2020-05-30:
I am having an issue with setting an action value that only affects Ikea color bulbs (not Hue). I am using ioBroker and current stable versions of all relevant systems.

If I use this command `"on": true, "bri": 25, "xy": [0.6915,0.3083]` brightness does only change for a few milliseconds although the value seems to be set correctly in the bri datapoint in ioBroker. Xy and color works perfectly. Only after manually sending the correctly sent value from the bri datapoint again the bulb changes the brightness.

ebaauw, 2020-05-30:
I think you're hitting a bug in the IKEA firmware.  It ignores commands when a transition is still in progress.  The API command results in (I think - need to sniff/double-check the code - haven't yet refactored the handling of group action) two Zigbee commands in close succession: a _Move to Level (with On/Off)_ and a _Move to Colour_.

Possible workarounds:
- You might try adding `"transitiontime": 0` to the API command, but that will look ugly;
- If you send two separate API commands, it might just work for the default transitiontime of 0.4s, otherwise you need to delay the seconds command a bit;
- Recall a Zigbee scene instead.

MC-1984, 2020-05-30:
> you might try adding `"transitiontime": 0` to the API command

That works! Thank you very much for your fast replies!

Can you please explain what you mean with 

> * Recall a Zigbee scene instead.

ebaauw, 2020-05-30:
Create a scene with the desired brightness and colour, and recall that scene instead of sending the brightness and colour.  See http://dresden-elektronik.github.io/deconz-rest-doc/scenes/.

MC-1984, 2020-05-30:
Okay, thanks again!

closed 2020-05-30 by MC-1984