dresden-elektronik/deconz-rest-plugin #2851: No battery state for newer IKEA Tradfri remotes

Wildbill-Z, 2020-05-31:
Beside of many lights, motions sensors and repeaters, I have some older IKEA remotes (with the larger pairing button) and some of the newer ones (small pairing button). I also use iobroker which is connected to the Deconz-API with [this adapter](https://github.com/iobroker-community-adapters/ioBroker.deconz). None of the newer remotes shows a correct battery state in iobroker. It stays at 0% for weeks. Everything is working OK, all buttons are recognized. All other battery devices show a battery level between 34% and 100%, but the new ones stay at 0%. Even changing batteries to new ones didn't change a thing.

I just set the adapter in iobroker to debug. This is was I got:

```
deconz.0	2020-05-31 12:11:50.921	debug	(18377) Websocket message: {"e":"changed","id":"26","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2020-05-31T10:11:50"},"t":"event","uniqueid":"14:b4:57:ff:fe:65:16:1a-01-1000"}
deconz.0	2020-05-31 12:11:42.805	debug	(18377) Websocket message: {"e":"changed","id":"26","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2020-05-31T10:11:42"},"t":"event","uniqueid":"14:b4:57:ff:fe:65:16:1a-01-1000"}
deconz.0	2020-05-31 12:11:38.263	debug	(18377) Websocket message: {"e":"changed","id":"26","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2020-05-31T10:11:38"},"t":"event","uniqueid":"14:b4:57:ff:fe:65:16:1a-01-1000"}
**deconz.0	2020-05-31 12:11:38.137	debug	(18377) Websocket message: {"config":{"alert":"none","battery":74,"group":"39139","on":true,"reachable":true},"e":"changed","id":"26","r":"sensors","t":"event","uniqueid":"14:b4:57:ff:fe:65:16:1a-01-1**
deconz.0	2020-05-31 12:11:24.529	debug	(18377) Websocket message: {"e":"changed","id":"33","r":"sensors","state":{"buttonevent":3002,"lastupdated":"2020-05-31T10:11:24"},"t":"event","uniqueid":"14:b4:57:ff:fe:50:54:f7-01-1000"}
deconz.0	2020-05-31 12:11:13.249	debug	(18377) Websocket message: {"e":"changed","id":"33","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2020-05-31T10:11:13"},"t":"event","uniqueid":"14:b4:57:ff:fe:50:54:f7-01-1000"}
```

I pressed two times on a new remote, and then on an older remote which shows battery state. As i can see, with the older one there were two messages at first press, the first which I marked with two * with other messages like battery state. so it seems to be a problem in Deconz I think as it doesn't send out this first message which contains battery state? Is there anything else I can check or do?

Greetings, Jürgen

ebaauw, 2020-05-31:
Most likely, the attribute reporting for the battery hasn't been setup properly.  Does the iobroker adapter log an event when you read the _Battery Percentage Remaining_ attribute of the _Power Configuration_ cluster of the remote in the deCONZ GUI?

Wildbill-Z, 2020-05-31:
Just tested. Iobroker didn't log events when just reading out attributes in Deconz GUI. Neither with older nor the newer remotes. But I am not sure, what exactly to do to _read the Battery Percentage Remaining attribute of the Power Configuration cluster of the remote in the deCONZ GUI_. Then just pressed another older remote which was not used for a couple of days and immidiately got a double message, one with the battery state and the other with the button pressed. Then tried the newer remote and just got the one message about butten press:
`deconz.0 | 2020-05-31 15:34:05.184 | debug | (3405) Websocket message: {"e":"changed","id":"33","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-05-31T13:34:05"},"t":"event","uniqueid":"14:b4:57:ff:fe:50:54:f7-01-1000"}
-- | -- | -- | --
deconz.0 | 2020-05-31 15:33:58.317 | debug | (3405) Websocket message: {"e":"changed","id":"33","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-05-31T13:33:58"},"t":"event","uniqueid":"14:b4:57:ff:fe:50:54:f7-01-1000"}
deconz.0 | 2020-05-31 15:33:53.136 | debug | (3405) Websocket message: {"e":"changed","id":"26","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-05-31T13:33:53"},"t":"event","uniqueid":"14:b4:57:ff:fe:65:16:1a-01-1000"}
deconz.0 | 2020-05-31 15:33:50.506 | debug | (3405) Websocket message: {"config":{"battery":100,"group":"20532","on":true,"reachable":true},"e":"changed","id":"24","r":"sensors","t":"event","uniqueid":"00:17:88:01:04:af:d5:63-02-fc00"}
deconz.0 | 2020-05-31 15:33:48.751 | debug | (3405) Websocket message: {"e":"changed","id":"26","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-05-31T13:33:48"},"t":"event","uniqueid":"14:b4:57:ff:fe:65:16:1a-01-1000"}
`
As I have the effect of not getting battery updates for the newer ones for weeks, there seems to be something changed with them.

ebaauw, 2020-05-31:
Open the _Cluster Info_ panel in the GUI, click the right drop-down button on the node for the remote, select the _Power Configuration_ cluster.  The _Cluster Info_ panel is now populated with the attributes.  Press _Read_ while pressing a button on the remote to wake it.  This might take multiple attempts, but should show the value of the _Battery Percentage Remaining_ attribute (0x0021).

Wildbill-Z, 2020-05-31:
You are great, that worked. Before doing the readout the new remote (all of them) showed a 0 in attribute 0x0021, for weeks now. After touching a couple of buttons and clicking read it changed to 87% and in that moment I got a readout in iobroker and it also showed 87%. All OK now.
Before I close this issue please tell me, why is that behavior with the new remotes? All older ones showed a battery state just after pairing, the new ones all show 0. And have I to do this readout from time to time or will it update from now on automatically?

Grettings, Jürgen

ebaauw, 2020-05-31:
Probably the attribute reporting and associated bindings weren't setup when pairing the new remotes.  You might want to do that manually in the GUI, to make sure the battery percentage is updated regularly.  See the user manual (under help) how to setup a binding from the _Power Configuration_ cluster to endpoint 0x01 on the coordinator.  Then configure reporting for the _Battery Percentage Remaining_ attribute with settings 7200, 7200, 0, for a report every two hours.

Wildbill-Z, 2020-05-31:
I din't even know, what I could do within Deconz-GUI. I'll have a look, if the battery level updates regularly from now on and if not, will follow your suggestion to create a manual binding.
For now I close this issue. Many many thanks!
Greeteings, Jürgen

closed 2020-05-31 by Wildbill-Z