dresden-elektronik/deconz-rest-plugin #2852: harmony hub home automation no longer works

steff75, 2020-05-31:
Since about 2 weeks the control of my lamps/plugs via the Harmony Hub does not work anymore.

The hub has been successfully paired with my Conbee 2 and it worked until recently.

I can also see all devices connected to the Conbee 2 in the Harmony App. However, if I want to switch on a lamp via Harmony, for example, this is not done.  Instead, the app briefly displays "no connection".

Here are the things I've done since when the Harmony control was still working:

- several new Ikea devices (lamps/switches/shutters) paired with Phoscon
- Update of the Conbee 2 with firmware 26580700
- Update Phoscon/Rest-api to v2.05.77
- maybe there were also updates for the Harmony hub?
- I have already unpaired and repaired Phoscon with Harmony Hub several times

In Node-Red I have installed the harmony-websocket node, where I can see in its automation-node that the harmony hub can read the status from the devices paired with the Conbbe 2. Therefore a connection to the devices must be possible. 

Is there a simple way to find out the cause of the problem?
System: marthoc/docker-deconz on synology DS218+, Conbee 2

Thanks!

[image]

[image]

[image]

Mimiix, 2020-06-06:
Please refer to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2486
Probably has to do with that.

w00druff, 2020-06-09:
Exactly the same problem here.

- Conbee 2 with firmware 26580700
- Phoscon/Rest-api v2.05.77

Working again fine after downgrade to v2.05.75 !

steff75, 2020-06-10:
> Working again fine after downgrade to v2.05.75 !

Could you leave the Conbee 2 firmware at version 26580700, or did you have to downgrade the stick as well?

I would still not like to downgrade the plugin because my IKEA blinds seem to be more stable now.
I hope there is a fix that allows me to have best of both worlds.

Thanks!

w00druff, 2020-06-10:
Firmware stays on 26580700 !

Mimiix, 2020-06-10:
As this seems different, i re-open.

mrbjoernson, 2020-06-17:
Same problem here, after the update to v2.05.77. i hope i will be fixed.

w00druff, 2020-06-18:
Works again with v2.05.78 !

steff75, 2020-06-19:
Yes, I can confirm that.
I have compiled the binary file in my Docker container and replaced the old plugin with it. After restarting the container, I was able to operate my lamp with the Harmony again.

@Mimiix Should I close the issue now or wait for an official release?

Mimiix, 2020-06-19:
@steff75 I'm back to the HA addon. Let's see what happens on that version too when it updates. I'll confirm if it works again there too.

Mimiix, 2020-07-13:
I forgot to reply on this. I'll check today :)

Mimiix, 2020-08-03:
HA Addon works too. #bitlate again

closed 2020-08-03 by Mimiix