dresden-elektronik/deconz-rest-plugin #2853: No colour temperature (ct) possible within Innr RB 285 C

ChrisPl-GH, 2020-05-31:
ConBee 1, Firmware 26350500
Since version 2.05.76 the color temperature of the Innr RB 285 C (E27) can no longer be adjusted.
Error message: invalid value, 421, for parameter, ct
Via REST-API ctmin and ctmax are output with 0.

> {
  "ctmax": 0,
  "ctmin": 0,
  "etag": "da87d________bae7dc662423947e31e",
  "hascolor": true,
  "lastseen": "2020-05-31T11:35:31.585",
  "manufacturername": "innr",
  "modelid": "RB 285 C",
  "name": "x_e27_cdt_1",
  "state": {
    "alert": "none",
    "bri": 82,
    "colormode": "xy",
    "ct": 344,
    "effect": "none",
    "hue": 8514,
    "on": false,
    "reachable": true,
    "sat": 195,
    "xy": [0.4416, 0.4215]
  },
  "swversion": "2.0",
  "type": "Extended color light",
  "uniqueid": "00:15:8d:00:xx:xx:xx:a8-01"
}

The update to version 2.05.77 did not solve the problem. With version 2.05.75 and the current firmware it worked.

ebaauw, 2020-05-31:
Hm, the resource reports `ctmin` and `ctmax` as 0, so the only valid `ct` value would be 0.  Can you read the _Color Control_ cluster attributes in the GUI, and check what values the light reports for _Color temperature min_ (0x400B) and _Color temperature max_ (0x400C)?

ChrisPl-GH, 2020-05-31:
I cannot use the GUI because no monitor is connected to the Raspi.

ChrisPl-GH, 2020-05-31:
With the previous versions the CT adjustment has always worked without problems. The CT value changes by adjusting the RGB colour and then the white is switched with the reported CT, but an adjustment is not possible.

ChrisPl-GH, 2020-06-20:
Works again with version 2.05.78
Thanks

closed 2020-06-20 by ChrisPl-GH