dresden-elektronik/deconz-rest-plugin #2854: Philips Hue dimmer switch not working reliable

Gerben321, 2020-05-31:
So first of all, nice software so far. I've been using it since yesterday and I've moved over all my Aqara stuff, and so far 7 Hue lights (the other ones are a PITA to re-pair because of their enclosure...).

Seems to be working fine so far, but my Hue dimmer switch isn't really working like it used to I think. First of all, before I could press the dim up/down buttons once to make a very small dim change. Now a single press turns the lights on or off, that seems odd and not logical? Because that's what the big on/off buttons are for in my opinion.

Secondly, when I long press the dim up/down buttons my lights are unevenly dimming! 2/4 lights go up, but the other 2 stay very dimmed. How is that possible when they're identical bulbs and used to work perfectly before this? They're E27 simple dimming Ikea lights. Also, the on button has to be clicked 2 of 3 times sometimes before it registers the click. I almost never had this with the Hue hub. My links all seem to be healthy, and even have multiple lines to the lights and dimmer, so I would say the signals should be going through fine?

This is how my map looks right now. Everything is green, that's good right? Should I give it some time to all calm down and connect? Otherwise I have to keep my Hue hub for the dimmer and those 4 lights I'm afraid!
[image]

Smanar, 2020-05-31:
How have you "linked" your bulbs and the remote ?
Have you used the remote group visible on home page after the remote inclusion ?

Gerben321, 2020-05-31:
I have made a group in Phoscon in which I added the 4 lights. In that group I've added a switch.

[image]

What do you mean with ' Have you used the remote group visible on home page after the remote inclusion ?'?

Smanar, 2020-05-31:
Ok, so I don't know this remote, but if it works like other remote, there is another way to do that.

When you have included the remote, have you a new group with the same name than the remote on the home page ?

If yes, don't make another group, just add all your bulb in this group.

Gerben321, 2020-05-31:
As far as I know I can't add switches to groups directly? So no, unless I don't fully understand what you mean. But I mean this should work fine IMO. Just had to press the on button again twice before it responded. Seems like the commands are not reliable. That's too bad...

Gerben321, 2020-05-31:
I somehow think Phoscon is working poorly. For example: I have a Aqara switch connected as well. I have Home Assistant connected, and I read deconz_event events. I see the events coming in without the lights being changed sometimes. When I control the lights via the deconz_events in HASS it works a lot better. Is Phoscon somehow missing some of these events?!

Smanar, 2020-05-31:
No, add switch to a bulb group is not the same thing than adding a bulb in the remote group. In the second situation you don't create group.
Groups created by remote are special. But not all remote create this group.

On my side I never link switch and bulb using phoscon too, I make group using it, but all my automation are make with the application.

Gerben321, 2020-05-31:
Okay so I found what you meant I guess. Via the old interface I added the lights to the dimmer group (that was invisible in Phoscon). Would be great if I could do this via Phoscon though. Really confusing.

Smanar, 2020-05-31:
Harf, yep, I don't know this remote, so I can't say if it s visible or not in phoscon.