dresden-elektronik/deconz-rest-plugin #2855: Device visible in deconz but not in Phoscon

X015, 2020-05-31:
Hello GitHub, my issue is that when i pair any device its visible in deconz but never in Phoscon

Here is the Log when i pair two Tradfri Lights:

`      

15:18:13:809 APS-DATA.request id: 47, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:13:820 APS-DATA.confirm id: 47, status: 0x00 SUCCESS

15:18:13:820 APS-DATA.confirm request id: 47 -> confirmed, timeout 1590931093

15:18:13:824 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:13:824 APS-DATA.indication request id: 47 -> finished

15:18:13:825 APS-DATA.request id: 47 erase from queue

15:18:13:825 ZDP status = 0x00 -> SUCCESS

15:18:13:825 ZDP Mgmt_Lqi_rsp zdpSeq: 88 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:13:825     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:13:825     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:17:169 Mgmt_Lqi_req zdpSeq: 89 to 0x00212EFFFF059B79 start index 0

15:18:17:169 APS-DATA.request id: 63, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:17:182 APS-DATA.confirm id: 63, status: 0x00 SUCCESS

15:18:17:182 APS-DATA.confirm request id: 63 -> confirmed, timeout 1590931097

15:18:17:186 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:17:186 APS-DATA.indication request id: 63 -> finished

15:18:17:186 APS-DATA.request id: 63 erase from queue

15:18:17:186 ZDP status = 0x00 -> SUCCESS

15:18:17:186 ZDP Mgmt_Lqi_rsp zdpSeq: 89 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:17:187     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:17:187     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:20:529 Mgmt_Lqi_req zdpSeq: 90 to 0x00212EFFFF059B79 start index 0

15:18:20:529 APS-DATA.request id: 78, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:20:542 APS-DATA.confirm id: 78, status: 0x00 SUCCESS

15:18:20:542 APS-DATA.confirm request id: 78 -> confirmed, timeout 1590931100

15:18:20:546 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:20:546 APS-DATA.indication request id: 78 -> finished

15:18:20:546 APS-DATA.request id: 78 erase from queue

15:18:20:546 ZDP status = 0x00 -> SUCCESS

15:18:20:546 ZDP Mgmt_Lqi_rsp zdpSeq: 90 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:20:546     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:20:547     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:23:890 Mgmt_Lqi_req zdpSeq: 91 to 0x00212EFFFF059B79 start index 0

15:18:23:890 APS-DATA.request id: 93, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:23:901 APS-DATA.confirm id: 93, status: 0x00 SUCCESS

15:18:23:901 APS-DATA.confirm request id: 93 -> confirmed, timeout 1590931103

15:18:23:905 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:23:905 APS-DATA.indication request id: 93 -> finished

15:18:23:906 APS-DATA.request id: 93 erase from queue

15:18:23:906 ZDP status = 0x00 -> SUCCESS

15:18:23:906 ZDP Mgmt_Lqi_rsp zdpSeq: 91 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:23:906     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:23:906     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:26:658 APS-DATA.request id: 106, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00

15:18:26:658 send permit join, duration: 59

15:18:27:250 Mgmt_Lqi_req zdpSeq: 92 to 0x00212EFFFF059B79 start index 0

15:18:27:250 APS-DATA.request id: 110, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00

15:18:27:261 APS-DATA.confirm id: 110, status: 0x00 SUCCESS

15:18:27:261 APS-DATA.confirm request id: 110 -> confirmed, timeout 1590931107

15:18:27:265 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:27:265 APS-DATA.indication request id: 110 -> finished

15:18:27:265 APS-DATA.request id: 110 erase from queue

15:18:27:265 ZDP status = 0x00 -> SUCCESS

15:18:27:265 ZDP Mgmt_Lqi_rsp zdpSeq: 92 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:27:265     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:27:265     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:27:315 APS-DATA.confirm id: 106, status: 0x00 SUCCESS

15:18:27:330 aps request id: 106 finished, erase from queue

15:18:30:609 Mgmt_Lqi_req zdpSeq: 93 to 0x00212EFFFF059B79 start index 0

15:18:30:609 APS-DATA.request id: 125, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:30:619 APS-DATA.confirm id: 125, status: 0x00 SUCCESS

15:18:30:619 APS-DATA.confirm request id: 125 -> confirmed, timeout 1590931110

15:18:30:623 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:30:623 APS-DATA.indication request id: 125 -> finished

15:18:30:624 APS-DATA.request id: 125 erase from queue

15:18:30:624 ZDP status = 0x00 -> SUCCESS

15:18:30:624 ZDP Mgmt_Lqi_rsp zdpSeq: 93 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:30:624     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:30:624     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:33:969 Mgmt_Lqi_req zdpSeq: 94 to 0x00212EFFFF059B79 start index 0

15:18:33:969 APS-DATA.request id: 141, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:33:981 APS-DATA.confirm id: 141, status: 0x00 SUCCESS

15:18:33:981 APS-DATA.confirm request id: 141 -> confirmed, timeout 1590931113

15:18:33:985 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:33:985 APS-DATA.indication request id: 141 -> finished

15:18:33:985 APS-DATA.request id: 141 erase from queue

15:18:33:985 ZDP status = 0x00 -> SUCCESS

15:18:33:985 ZDP Mgmt_Lqi_rsp zdpSeq: 94 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:33:985     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:33:986     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:37:329 Mgmt_Lqi_req zdpSeq: 95 to 0x00212EFFFF059B79 start index 0

15:18:37:329 APS-DATA.request id: 156, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:37:341 APS-DATA.confirm id: 156, status: 0x00 SUCCESS

15:18:37:341 APS-DATA.confirm request id: 156 -> confirmed, timeout 1590931117

15:18:37:345 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:37:345 APS-DATA.indication request id: 156 -> finished

15:18:37:345 APS-DATA.request id: 156 erase from queue

15:18:37:345 ZDP status = 0x00 -> SUCCESS

15:18:37:346 ZDP Mgmt_Lqi_rsp zdpSeq: 95 from 0x00212EFFFF059B79 total: 2, startIndex: 0, listCount: 2

15:18:37:346     * neighbor: 0x086BD7FFFE04E984 (0xDBD7), LQI: 129, relation: 0x01 rxOnWHenIdle: 1

15:18:37:346     * neighbor: 0x086BD7FFFE37152C (0x4B0B), LQI: 255, relation: 0x01 rxOnWHenIdle: 1

15:18:40:690 Mgmt_Lqi_req zdpSeq: 96 to 0x00212EFFFF059B79 start index 0

15:18:40:690 APS-DATA.request id: 172, addrmode: 0x03, addr: 0x00212effff059b79, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

15:18:40:702 APS-DATA.confirm id: 172, status: 0x00 SUCCESS

15:18:40:702 APS-DATA.confirm request id: 172 -> confirmed, timeout 1590931120

15:18:40:706 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0

15:18:40:706 APS-DATA.indication request id: 172 -> finished

15:18:40:706 APS-DATA.request id: 172 erase from queue

15:18:40:706 ZDP status = 0x00 -> SUCCESS
`

I tried 6 Tradfri light and 4 Xiaomi Door Sensor always the same Visible in deconz but never in Phoscon

ebaauw, 2020-05-31:
> my issue is that when i pair any device its visible in deconz but never in Phoscon

Do you open the network from Phoscon or from the GUI?  You need to open the network through the API (using Phoscon or another API client) for any REST resources to be created.  When the resource is created, the node label in the GUI changes from the NWK address to the resource name.

Even then, Phoscon doesn't support all devices support by the REST API plugin, but the Trådfri lights should be visible.

X015, 2020-05-31:
> Do you open the network from Phoscon or from the GUI? You need to open the network through the API (using Phoscon or another API client) for any REST resources to be created. When the resource is [...]

I open the network over Phoscon

ebaauw, 2020-05-31:
Ok, deCONZ hasn't read the _Simple Descriptor(s)_ of the (I assume) lights.  You can tell from the missing right drop-down circle in the nodes.  You might try and read them manually from the left drop-down circle, or by pressing `8` while the node is selected.  That should bring up the right circle and populate the endpoint(s) and clusters.  Next, search for device in Phoson, and read the attributes of the _Basic_ cluster in the GUI.  That should trigger the creation of the REST resources.

X015, 2020-05-31:
> Ok, deCONZ hasn't read the _Simple Descriptor(s)_ of the (I assume) lights. You can tell from the missing right drop-down circle in the nodes. You might try and read them manually from the left [...]

If i press the Read Simple Descripters "Button" it stays on for 1 second and goes away, is that normal?

meremortals70, 2020-06-08:
Not the only one experiencing this issue.

Smanar, 2020-06-08:
Hello, just to be sure it's not a connexion problem, have you an usb cable extension to try with it ?

X015, 2020-06-08:
> Hello, just to be sure it's not a connexion problem, have you an usb cable extension to try with it ?

Hello, yes it is all the time attached to a usb extension cable