dresden-elektronik/deconz-rest-plugin #2856: ctmax attribute at 65535 for IKEA lights?

wvuyk, 2020-05-31:
a user reported an issue with his ikea light:

{
"type": "Color temperature light",
"name": "Candeeiro do Nuno",
"modelid": "TRADFRI bulb E14 WS opal 600lm",
"swversion": "2.0.023",
"uniqueid": "00:0d:6f:ff:fe:bb:34:4a-01",
"manufacturername": "IKEA of Sweden",
"state": {
"on": false,
"bri": 211,
"ct": 454,
"alert": "none",
"colormode": "ct",
"reachable": true
},
"etag": "8ecc54cd9d8f359312bb01ca81a275af",
"hascolor": true,
**"ctmax": 65535,**
"ctmin": 0
}

ctmax should be no higher then 500?

ebaauw, 2020-05-31:
There are actually some lights that support higher `ct` values than 500.  IKEA typically support 250 to 454.

I think 0 to 65535 are default values, in case the light doesn't expose _Color temperature min_ and _Color temperature max_ or deCONZ hasn't read these attributes.  I vaguely remember an issue where both were reported as 0 or 65535, causing issues, but I cannot find it.  I think the old (ZLL) IKEA firmware firmware doesn't expose these, but recent (ZB3) firmware does

ChrisPl-GH, 2020-05-31:
The Paulmann RGBW E27 also supplies ctmin 0 and ctmax 65535:

> {
  "ctmax": 65535,
  "ctmin": 0,
  "etag": "a4e145_______dc4c743e0ee4a34a998",
  "hascolor": true,
  "lastseen": "2020-05-31T22:40:02.831",
  "manufacturername": "Paulmann Licht GmbH",
  "modelid": "RGBW light",
  "name": "x_e27_cdt_3",
  "state": {
    "alert": "none",
    "bri": 0,
    "colormode": "xy",
    "ct": 647,
    "effect": "none",
    "hue": 2838,
    "on": false,
    "reachable": true,
    "sat": 249,
    "xy": [0.5789, 0.3858]
  },
  "swversion": "1218-0909",
  "type": "Extended color light",
  "uniqueid": "00:15:8d:00:xx:xx:xx:fb-01"
}

wvuyk, 2020-06-01:
So one cannot rely on the contents of ctmax, but errors are still thrown by rest api if one sents a value higher then ctmax when it is set?

ebaauw, 2020-06-01:
Throwing an error was probably a bridge too far.  The `ct` value is silently maximised to `ctmin` and minimised to `ctmax` when these are > 0.  So setting `ct` to 153 on an IKEA bulb would actually send 250, but return 153.

Another discussion is why these message should be http errors in the first place.  The Hue bridge returns status 200 in these cases.  I think this makes more sense: as long as there’s a valid JSON response, return 200.

wvuyk, 2020-06-01:
That sounds a lot better to me for sure. It would make it more fool proof 👍

ebaauw, 2020-06-01:
According to the ZCL spec, the maximum value for _ColorTemperature_, _Color Temperature Min_, and _Color Temperature Max_ is actually 65279 (0xFEFF), same as for _Current X_ and _Current Y_.

closed 2020-07-11 by stale[bot]