dresden-elektronik/deconz-rest-plugin #2857: RGBgenie ZB-5001 pairing issue

gmitch64, 2020-05-31:
Currently running 2.05.77

I broke out my RGBgenie remote to try and pair it up, since I believe that it's supported. I did have it paired up before

> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1256

but since the buttons weren't working at the time, I removed it from my setup.
I've now come back to it, and I'm getting it to pair up, but it's not getting past the NWK address stage, no matter how I poke and prod it (I've removed it from the network, done a factory reset on it, included it again, pressed buttons on the remote when trying to read the basic cluster etc. And there's also a new battery in the remote as well, just in case)

One thing that I did notice, is that when I come to open the network to include it, as soon as I press the 2nd button for pairing, the network closes (I have it set for 10 mins open at the moment). I've repeated the inclusion process several times, and every time, then network immediately closes on pressing the 2nd button.

I don't know if that that's new behavior, but in the past when I've included new devices, the network has remained open for the 10 mins duration that I set. The only thing that I've added to the network recently has been a fan, and I don't remember whether that showed the same issue or not.

I don't have anything that's not already included to test with at the moment, but I'm wondering if anyone else is seeing issues pairing with 2.05.77?

[image]

gmitch64, 2020-05-31:
And just to clarify, it does seem to read the additional buttons on the remote (it goes all the way up to 5:) but not 11).

[image]

SwoopX, 2020-06-04:
Can you please paste a screenshot from all available endpoints/clusters? The device seems to have more than just the 2 thought endpoints.

> One thing that I did notice, is that when I come to open the network to include it, as soon as I press the 2nd button for pairing, the network closes (I have it set for 10 mins open at the moment).

What does that mean? How does it close? Do you get a success message?

> but it's not getting past the NWK address stage

What is the NWK stage?

gmitch64, 2020-06-06:
Sorry it's taken a couple of days to get back with you - work has been a bear this week.

> What is the NWK stage?

Maybe not the best description, but I get as far as this...

[image]

Where as other recognized devices get past the NWK stage, and present this..

[image]

> Can you please paste a screenshot from all available endpoints/clusters?

Yup.. Hopefully these come out ok. If there's any other info you need, please let me know.

Thanks.

G

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]
[image]

[image]

[image]

[image]

[image]

[image]
[image]

[image]

SwoopX, 2020-06-06:
Thanks. So it appears that data from the initial issue was incomplete and the support with I previously mentioned might have been for a different device with a similar modelID (or it was best effort). Let's jump over to the original issue then and continue from there. 

@Mimiix please close, moving over to #1256

closed 2020-06-07 by Mimiix