dresden-elektronik/deconz-rest-plugin #2859: Phoscon: Add Group always says the group already exists.

Cornelius-Riemenschneider, 2020-05-31:
Hi,

when adding a group the message
"The group already exists." is shown in the popup all the time.

Smanar, 2020-06-01:
Hello,

And the group don't already exist ? It is invisible ? Have you a capture ?

Cornelius-Riemenschneider, 2020-06-01:
The text is shown in the popup even before entering anything, and just stays there while I type.
The browser is firefox, but I just tested and the problem exists in Chrome as well.
Do you want a screen capture? Is there any more information I can provide?

Smanar, 2020-06-01:
Yes, can you give me the procedure you are using ?
- Go on phosocn homepage
- Press the "add group" button

And after ?

Cornelius-Riemenschneider, 2020-06-01:
1. I go to http://192.168.1.3/pwa/ (my raspberry pi's address)
2. I click on 192.168.1.3 for my gateway
3. I enter my password
4. I click on "Add group"
5. Without typing, it already says "The group already exists."
If I type something, this doesn't change.

I'm running deconz/phoscon on docker using the marthoc image.

Smanar, 2020-06-01:
Ha, ok I have made try, and have same ^^

But it's not a problem, you still can create a group ?

Cornelius-Riemenschneider, 2020-06-01:
Yes, it's just a cosmetic mistake, adding groups works fine.

YKO-de, 2020-06-05:
Is fixed with the next release.

closed 2020-06-05 by YKO-de