dresden-elektronik/deconz-rest-plugin #2862: DeConz: Repetitive Logfile entries to out of memory

FrankRenp, 2020-06-01:
I'm running deconz as a service on a raspberry, installed 3 weeks ago (only hint to version I have).

Got in /var/log/syslog MBytes of messages:
May 31 23:02:52 ioBroker deCONZ[31455]: QIODevice::write (zmHttpClient): device not open.

After approx 3min in dmesg:
[826783.676215] Out of memory: Kill process 31455 (deCONZ) score 169 or sacrifice child
[826783.676253] Killed process 31455 (deCONZ) total-vm:273432kB, anon-rss:160908kB, file-rss:0kB, shmem-rss:0kB
[826783.711134] oom_reaper: reaped process 31455 (deCONZ), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
Linux ioBroker 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
model name      : ARMv7 Processor rev 4 (v7l)
Hardware        : BCM2835
Revision        : a020d3

closed 2020-07-11 by stale[bot]

TungstenE2, 2020-12-21:
any news on this?

having similar issue

[image]

Smanar, 2020-12-21:
I don't see memory problem on your capture ?
And pls make your own issue.

TungstenE2, 2020-12-21:
referring to this message:
deCONZ[xxx]: QIODevice::write (zmHttpClient): device not open.

Mimiix, 2020-12-21:
The device not open has to do with ports on your device (https://community.openhab.org/t/problem-setting-up-deconz-integration/74172/5

 However, try with --platform (instead of 1 -) see where that goes. 
 
Anyway: 
 > And pls make your own issue.

I've asked you this before. Please stop hijacking old and closed issues.

TungstenE2, 2020-12-21:
@Mimiix this issue was not resolved and closed automatically because no one was following up. I was searching for the message and added a comment. Is this what you call hijacking?

But thx for the hint. Will give it a try. May be this also helps others searching for the issue.

Have a nice day.

Mimiix, 2020-12-21:
@TungstenE2 No issues are the same. Without any context, we can't help anymore and thus the bot closes it as nobody seems to be interested anymore. 

A lot has changed since 1 june. New versions came out etc, that's why we encourage users to open new issues with filled templates. That way, we can help users.

Also,i am the only one reading up every comment. If you reply on old closed issues, not a lot of people see it.

SwoopX, 2020-12-21:
@TungstenE2 have you noted that you made a small mistake while editing your service file? You're missing a space between the 2 port definitions...

Mimiix, 2020-12-21:
> @TungstenE2 have you noted that you made a small mistake while editing your service file? You're missing a space between the 2 port definitions...

Good catch!

TungstenE2, 2020-12-21:
thx! to be honest I was wondering the same, but c&p from this page.
https://wiki.metal.zone/display/MET/ConBee+Zigbee+USB-Gateway

will fix this asap!

SwoopX, 2020-12-21:
Pls follow Mimiix advice if that's not the solution ;)