dresden-elektronik/deconz-rest-plugin #2863: Wow what has happened to Deconz

meremortals70, 2020-06-01:
Since upgrading from .75 nothing but problems. 12 months problem free and now it’s like everything is falling apart. Suddenly I have sensor continually going off line and now my Philips lights are playing up as well. 😟

Smanar, 2020-06-01:
You have updated only deconz or the firmware too ?

meremortals70, 2020-06-01:
Just Deconz not firmware

yabbah, 2020-06-01:
I have the same issue both with hue and ikea tradfri. The get "reachable: false" in the API. If I press 0 on the node in deconz gui it come back again, most of the time. In .75 I had no issues at all. Something vital must have been changed. I have also just updated deconz, not the conbee II firmware.

joeblack2k, 2020-06-01:
same here, went back to version 3 months old and the FYRTUR roller blind works again!

gmitch64, 2020-06-01:
It looks like a couple of code changes for some new devices has brought some bugs and old coding practices out of the woodwork (from what I can gather reading between the lines (and PRs)).

G

zigistar, 2020-06-01:
same here. returned to .74

Smanar, 2020-06-02:
Ok so All is working fine on Version 74, 75 but problems on Version 77 (and I think 76)
Only with updating deconz without changing firmware ?

But there is a problem, it's possible check for difference in the API, but not in deconz core

yabbah, 2020-06-02:
Im back on 2.05.75, all is stable as a rock again. In 2.05.77 I had lots of nodes disconnecting and blink red in deCONZ GUI, haven't seen it yet with .75, and never had that issue before either. Also had many rows in my log on 2.05.77 that said something like "remove dead link ....:". That not happened now either. So it must be something in the core that cause this issues for so many. 

Also seen a thread on a swedish facebook-group with many people that also complained about issues and disconnected lights with 2.05.76/77.

willpall7, 2020-06-02:
Same here. I'm rolling back to .75. Everything is either intermittently going unavailable or taking a few seconds to respond to commands. Door sensors get stuck "open" when they're closed. Others just go offline indefinitely. I guess I'm just commenting so I can see updates on this :(

yabbah, 2020-06-03:
> Same here. I'm rolling back to .75. Everything is either intermittently going unavailable or taking a few seconds to respond to commands. Door sensors get stuck "open" when they're closed. Others just go offline indefinitely. I guess I'm just commenting so I can see updates on this :(

You mean you still have issues with 2.05.75, or the issues you describe was in the latest version?

meremortals70, 2020-06-03:
Yes looks like I don't have a choice but to rollback to get some stability.  Means six off my sensors are useless as they were added in .76 :(

willpall7, 2020-06-03:
> You mean you still have issues with 2.05.75, or the issues you describe was in the latest version?

2.05.75 was mostly stable for me. The issues I'm having now started when I updated to .77. One thing I've really noticed is that when Home Assistant issues a `turn_on` command to a deCONZ light group, the individual bulbs now illuminate one after another, instead of all at once. It takes a couple seconds whereas before it was instant. 

I need to see what debugs I can pull to show it... I haven't rolled back yet.

meremortals70, 2020-06-03:
I have decided to have another look at ZHA, ordered another zigbee stick to run on another channel so I can have the two side by side.

djashjones, 2020-06-03:
I've moved everything over to hubitat

meremortals70, 2020-06-03:
Ok great, upgraded the Deconz add-in for HA which downgrades Deconz to .75.
Lost all my Xaiomi sensors.

mkarnebeek, 2020-06-04:
> Lost all my Xaiomi sensors.

Same, but they did come back once the sensor reported something (motion, temperature, ...)

Smanar, 2020-06-04:
Not always, I have problem too if the conbee stay offline too much time with Xiaomi.

Mimiix, 2020-06-05:
As there are other ways to discuss these issues, i close this issue for now.

closed 2020-06-05 by Mimiix

meremortals70, 2020-06-05:
That’s rude, we raised legitimate issues which need to be addresses

Mimiix, 2020-06-05:
@meremortals70 Theres a general thread #2875 in which this continues. Having multiple threads for the same issue is not beneficial in any way.

MattWestb, 2020-07-29:
Great work done @jcallaghan !!!!