dresden-elektronik/deconz-rest-plugin #2864: RaspBee II RTC doesn't work on Pi 4B

ebaauw, 2020-06-01:
Got a new Pi 4B, running Raspbian buster and installed the RaspBee II to it.

I followed all the installation steps from https://phoscon.de/en/raspbee2/install, double-checked, triple-checked them, but cannot get the real-time clock to work.  It seemed to work fine on the Pi 3B running stretch, to which the RaspBee II was installed before.

The RaspBee II seems to work fine with deCONZ.  I2C is enabled, and it looks like the rtc driver is registered on address 0x51:
```
$ uname -a
Linux pi6 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
$ i2cdetect -l
i2c-1	i2c       	bcm2835 I2C adapter             	I2C adapter
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 UU 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77
```

However, reading the clock fails:
```
$ sudo hwclock -v
hwclock from util-linux 2.33.1
System Time: 1591041638.171913
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Last drift adjustment done at 1591041382 seconds after 1969
Last calibration done at 1591041382 seconds after 1969
Hardware clock is on UTC time
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
ioctl(3, RTC_UIE_ON, 0): Invalid argument
Waiting in loop for time from /dev/rtc0 to change
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument
...synchronization failed
```

The `/etc/rtc0` device seems to do something, when writing the clock:
```
$ sudo hwclock -w -v
hwclock from util-linux 2.33.1
System Time: 1591041693.705954
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Last drift adjustment done at 1591041382 seconds after 1969
Last calibration done at 1591041382 seconds after 1969
Hardware clock is on UTC time
Assuming hardware clock is kept in UTC time.
RTC type: 'rtc-pcf85063'
Using delay: 0.000000 seconds
missed it - 1591041693.715314 is too far past 1591041693.000000 (0.715314 > 0.001000)
1591041694.000000 is close enough to 1591041694.000000 (0.000000 < 0.002000)
Set RTC to 1591041694 (1591041693 + 1; refsystime = 1591041693.000000)
Setting Hardware Clock to 20:01:34 = 1591041694 seconds since 1969
ioctl(RTC_SET_TIME) was successful.
Not adjusting drift factor because the --update-drift option was not used.
New /etc/adjtime data:
0.000000 1591041693 0.000000
1591041693
UTC
```

I'm afraid this is not my cup of tea.  Please let me know what other info is needed for troubleshooting.

dxxxm, 2020-06-03:
Doesn't `/boot/config.txt` need a explicit `i2c-rtc` config entry, depending on the actual hw, e.g.
`dtoverlay=i2c-rtc,ds3231`
`/boot/overlays/README` lists all known 
The raspbee2 install doesn't mention it, though

ebaauw, 2020-06-03:
Thank you for the suggestion.  I really don't know; as I said: not my cup of tea.  I did find a reference to that through google, from 2016 or so.  I didn't have to specify that for the Pi 3B, but I suppose I could try anyways.

ebaauw, 2020-06-03:
Since the service definition does:
```
echo pcf85063 0x51 > /sys/class/i2c-adapter/i2c-1/new_device
```
I think the line should be:
```
dtoverlay=i2c-rtc,pcf85063,addr=0x51
```
But no joy: `i2cdetect -y 1` doesn't show any device and system doesn't find the hardware clock at all:
```
$ sudo hwclock -v
hwclock from util-linux 2.33.1
System Time: 1550139146.807026
Trying to open: /dev/rtc0
Trying to open: /dev/rtc
Trying to open: /dev/misc/rtc
No usable clock interface found.
```
After starting the service, I get the same error as before.

ebaauw, 2020-06-09:
Updated the RaspBee II firmware to 0x26610700, but no change.

ebaauw, 2020-06-12:
Cross-referencing https://github.com/dresden-elektronik/raspbee2-rtc/issues/3.

ChrisHae, 2020-06-12:
what is the output of i2cdetect -y 1 when deconz is running? I can confirm that the rtc is not running on a pi4 but on a pi3 it runs great. But I have another output of i2cdetect:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- 51 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

It should be UU instead of 51 and all other fields empty (--).
The installation routine in the makefile should be correct no need to add other lines in config.txt or wherever. At least it works on a pi3. I don't know yet if the routine must differ on a pi4. The only thing is that the kernel version of the pi4 is another as of the pi3. That matters when compiling the rtc file. But the makefile should automaticaly choose the right version or else the make command would fail before the make install command.

Seems to me that the proper configuration of the i2c interface on the pi4 is not correct yet. But I have to investigate further.

ebaauw, 2020-06-12:
> what is the output of i2cdetect -y 1 when deconz is running?

Same as when it's not running:
```
$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 UU 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77                         
$ sudo systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/deconz-gui.service.d
           └─override.conf
   Active: active (running) since Thu 2019-02-14 11:12:07 CET; 1 years 3 months ago
 Main PID: 706 (deCONZ)
    Tasks: 16 (limit: 4915)
   Memory: 54.2M
   CGroup: /system.slice/deconz-gui.service
           ├─706 /usr/bin/deCONZ --http-port=80 --dbg-info=2 --dbg-aps=2 --dbg-error=1
           ├─768 dbus-launch --autolaunch 9105a5cfb368456ab9a44b39e552d395 --binary-syntax --close-stderr
           └─769 /usr/bin/dbus-daemon --syslog-only --fork --print-pid 5 --print-address 7 --session

Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:271 aps request id: 188 finished, erase from queue
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:293 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 17, rssi: 21
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:293         asdu: 187401004000420a30783236363130373030
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:293 Node data 0x00212effff05bbdb profileId: 0x0104, clusterId: 0x0000
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:293 0x00212EFFFF05BBDB: update ZCL value 0x01/0x0000/0x4000 after 0 s
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:873 poll node 00:21:2e:ff:ff:05:bb:db-01
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:873 Poll light node RaspBee II
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:921 read attributes of 0x00212EFFFF05BBDB cluster: 0x0000: [ 13:46:27:921 0x4000 13:46:27:921 ]
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:921 add task 1760467 type 19 to 0x00212EFFFF05BBDB cluster 0x0000 req.id 194
Jun 12 15:46:27 pi6 deCONZ[706]: 13:46:27:921 Poll APS request 194 to 0x00212EFFFF05BBDB cluster: 0x0000
```

> At least it works on a pi3.

Yes, in did in my case.

>  I don't know yet if the routine must differ on a pi4. 

I did see a `Warning: modules_install: missing 'System.map' file. Skipping depmod.
` during `sudo make install`.  Don't remember if I saw that on the Pi3 as well.
```
$ cd raspbee2-rtc-master/
$ make
checking deb package dependencies...
i2c-tools
build-essential
raspberrypi-kernel-headers
build rtc module...
make -C /lib/modules/4.19.118-v7l+/build M=/home/pi/raspbee2-rtc-master modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-v7l+'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.19.118-v7l+'
rtc module successfully build, you may run 'sudo make install' now
$ sudo make install
install rtc module...
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-v7l+'
  INSTALL /home/pi/raspbee2-rtc-master/rtc-pcf85063.ko
  DEPMOD  4.19.118-v7l+
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.118-v7l+'
systemctl disable fake-hwclock
Synchronizing state of fake-hwclock.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable fake-hwclock
enable rtc module...
rtc_pcf85063
enable i2c interface...
raspi-config nonint do_i2c 0
install rtc service...
systemctl start rtc-pcf85063.service
systemctl enable rtc-pcf85063.service
Done - Please reboot your machine now
```

ChrisHae, 2020-06-15:
> I did see a Warning: modules_install: missing 'System.map' file. Skipping depmod. during sudo make > install. Don't remember if I saw that on the Pi3 as well.

Yes, this is also on a pi3. Thats why I added "depmod -A" to the makefile right after the make install step.

cwildfoerster, 2020-06-15:
Lights/Switches to not connect after update to 26610700. Needed to relearn them once. Connecting again even after reboot.

ChrisHae, 2020-06-17:
So after sudo apt update und sudo apt upgrade I reinstalled the rtc via makefile make und sudo make isntall and I was able to fix the problem I had. The installation also used linux-headers v. 4.19.118-v7l+ so basically I used the same routine as you did.
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
timedatectl
               Local time: Wed 2020-06-17 08:43:52 CEST
           Universal time: Wed 2020-06-17 06:43:52 UTC
                 RTC time: Wed 2020-06-17 06:43:53
                Time zone: Europe/Berlin (CEST, +0200)
System clock synchronized: no
              NTP service: inactive
          RTC in local TZ: no

ChrisHae, 2020-06-17:
Because of your output of i2cdetect where it looks that every gpio pins are active I would suggest that it is a RaspBeeII or FW error. But since you say it works well on a RPI3 with the same RaspBeeII and the same FW this probably could not be the case.

ebaauw, 2020-06-17:
> There are people with similar errors (not related to raspbee2):

Thanks for the link, Chris, that got me on the right track:

> My guess is something is pulling SDA down to zero.

Upon inspection, the SDA pin (physical pin 3) on my Raspberry Pi 4B is soldered wrongly, like 2mm down compared to the other pins.  While it makes good contact with the RaspBee II, it also touched the bottom part of my aluminium heatsink case, I suppose, effectively pulling it down. I cut the bottom of the pin, so it no longer makes contact (I didn't want to dust off my soldering iron to reseat the pin).  Now `i2cdetect` shows `--` for all addresses (except for 51, of course), and the clock works.

For completeness:
```
$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
```
$ sudo hwclock -v
hwclock from util-linux 2.33.1
System Time: 1592423786.844341
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Last drift adjustment done at 1591969922 seconds after 1969
Last calibration done at 1591969922 seconds after 1969
Hardware clock is on UTC time
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
ioctl(3, RTC_UIE_ON, 0): Invalid argument
Waiting in loop for time from /dev/rtc0 to change
...got clock tick
Time read from Hardware Clock: 2020/06/17 19:56:28
Hw clock time : 2020/06/17 19:56:28 = 1592423788 seconds since 1969
Time since last adjustment is 453866 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2020-06-17 21:56:27.876519+02:00
```

closed 2020-06-17 by ebaauw