dresden-elektronik/deconz-rest-plugin #2865: Conbee II impossible to install on Ubuntu 20.04 focal

iris1801, 2020-06-02:
Hi!
I was on 18.04 and everything worked like a charm for deconz, but after upgrading the os, i started to install deconz software for my zigbee devices.

I imported the public key given by phoscon, i add the repo, and i apt update, but it answer me:

tomillr, 2020-06-02:
Same problem here when trying to install a Conbee II stick on an Ubuntu Server 20.04.

```
sudo apt update
Hit:1 http://de.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://de.archive.ubuntu.com/ubuntu focal-updates InRelease                                                              
Hit:3 http://de.archive.ubuntu.com/ubuntu focal-backports InRelease                                                            
Ign:4 http://phoscon.de/apt/deconz focal InRelease                                                                             
Hit:5 http://de.archive.ubuntu.com/ubuntu focal-security InRelease                                                             
Err:6 http://phoscon.de/apt/deconz focal Release                                                                              
  404  Not Found [IP: 144.76.96.194 80]
Hit:7 https://downloads.plex.tv/repo/deb public InRelease                                      
Ign:8 http://download.webmin.com/download/repository sarge InRelease     
Hit:9 http://download.webmin.com/download/repository sarge Release
Reading package lists... Done
E: The repository 'http://phoscon.de/apt/deconz focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

```

```
sudo apt install deconz
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package deconz
```

iris1801, 2020-06-02:
I solved running deconz on docker, if you can't solve this problem, give docker a shot

Smanar, 2020-06-02:
Have you tried with theses files http://deconz.dresden-elektronik.de/ubuntu/beta/

iris1801, 2020-06-02:
Yes, tried also with the beta release, didn't worked for me

Malesimi, 2020-06-02:
manually change
/etc/apt/sources.list.d/deconz.list to:
deb [arch=amd64] http://phoscon.de/apt/deconz **bionic**-beta main

works fine in ubuntu 20.04. LTS for me.

iris1801, 2020-06-02:
Thanks, I'll give it a try tomorrow

closed 2020-07-03 by stale[bot]