dresden-elektronik/deconz-rest-plugin #2866: Mesh not forming. Conbee One, 2.05.77, 26350500, pi3

Klagio, 2020-06-02:
Hi, I  made a new installation a on pi3 witrh Conbee one, latest deconz, upgraded firmware, uploaded a backup I had .. everything starts but MESH is NOT forming, all devices are there, but no connewction.

What can I do?

Smanar, 2020-06-02:
Hello, by no connection you mean no green line beetwen devices ?

Can you show a capture pls ?

Klagio, 2020-06-02:
> Hello, by no connection you mean no green line beetwen devices ?
> 
> Can you show a capture pls ?

yes no lines between nodes. What is a capturew, a screen shot?

Klagio, 2020-06-02:
Is it defective stick? How can I check this?

Klagio, 2020-06-02:
Now all evening without light in the house. I am so close to throw in the garbage the  Conbee, and all the Zigbee smart light concept

Smanar, 2020-06-03:
I think you have loose all your device
For conbee 1 just use Phoscon to update the conbee, if the firwmare is not show in phoscon, don't use it.

Hava you tried the hidden config ? > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

Else It seem some user have problem with version 77, so they rollback to 75 > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2863

Klagio, 2020-06-04:
> I think you have loose all your device
> For conbee 1 just use Phoscon to update the conbee, if the firwmare is not show in phoscon, don't use it.
> 
> Have you tried the hidden config ? > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues
> 
> Else It seem some user have problem with version 77, so they rollback to 75 > #2863
> 
> Try to compare your config setting with this post [#768 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/768#issuecomment-418111119)
> 
> And if it's a new configuration, you can try the USB cable extension.

Hi, I already have a USB extension, and I downgraded deCONZ to75 (inside Home Assistant addon).

Will check now your suggestion hidden config. If all fails how do I do to reset everythinng (I also tried to reset the Gateway but it did not solve, probably just made things worse, but I did already)

Smanar, 2020-06-04:
You have already used backup/restore feature, so you are fine.
On the capture, the cconbee seem working (theres is "in network".

So I hope for a setting problem.

You can try too, run deconz with debug enabled.

Klagio, 2020-06-04:
> You have already used backup/restore feature, so you are fine.
> On the capture, the conbee seem working (theres is "in network").
> 
> So I hope for a setting problem.
> 
> You can try too, run deconz with debug enabled.

how is that?

Smanar, 2020-06-04:
So it s still not working ?

On the desktop, close deconz, use the terminal and type
deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1

Klagio, 2020-06-04:
I use deCONZ as addon in Home Assistabnt, which gives a log, not sure is the same

0:12:23:896 Current channel 11
20:12:23:943 Device TTL 2794 s flags: 0x7
20:12:23:966 delay sending request 184 dt 3 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:12:24:066 delay sending request 184 dt 3 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:12:24:166 delay sending request 184 dt 3 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:12:24:266 delay sending request 184 dt 4 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:12:24:366 delay sending request 184 dt 4 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:12:24:467 delay sending request 184 dt 4 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:12:24:505 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:12:24:505 max transmit errors for node 0x00158D0001530785, last seen by neighbors 132 s
20:12:24:766 channel is 11 but should be 15, start channel change
20:12:24:766 Skip automatic channel change, TODO warn user
20:12:28:347 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:12:28:347 max transmit errors for node 0x00158D0001530785, last seen by neighbors 135 s
20:12:33:730 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:12:33:733 max transmit errors for node 0x00158D0001530785, last seen by neighbors 141 s
20:12:34:766 channel is 11 but should be 15, start channel change
20:12:34:766 Skip automatic channel change, TODO warn user
20:12:39:723 0xF0F8 seems to be a zombie recv errors 14
20:12:39:723 LightNode removed 0x00158d0001530785
20:12:39:723 SensorNode id: 3 (lumi.plug) no longer available
20:12:39:723 SensorNode id: 70 (lumi.plug) no longer available
20:12:39:723 SensorNode id: 2 (lumi.plug) no longer available
20:12:39:723 SensorNode id: 69 (lumi.plug) no longer available
20:12:39:723 Node zombie state changed 0x00158d0001530785
20:12:44:766 channel is 11 but should be 15, start channel change
20:12:44:766 Skip automatic channel change, TODO warn user
20:12:52:802 GW firmware version: 0x26350500
20:12:52:802 GW firmware version is up to date: 0x26350500
20:12:54:766 channel is 11 but should be 15, start channel change
20:12:54:767 Skip automatic channel change, TODO warn user
20:13:04:767 channel is 11 but should be 15, start channel change
20:13:04:767 Skip automatic channel change, TODO warn user
20:13:04:786 no button map for: lumi.plug ep: 0x02 cl: 0x000C cmd: 0x0A pl[0]: 055
20:13:04:786 ZCL attribute report 0x00158D0001530785 for cluster: 0x000C, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
20:13:09:791 don't close database yet, keep open for 900 seconds
20:13:12:474 0x00158D0001530785 error APSDE-DATA.confirm: 0xE9 on task
20:13:14:766 channel is 11 but should be 15, start channel change
20:13:14:774 Skip automatic channel change, TODO warn user
20:13:17:873 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:13:23:862 Current channel 11
20:13:23:896 Device TTL 2734 s flags: 0x7
20:13:24:767 channel is 11 but should be 15, start channel change
20:13:24:767 Skip automatic channel change, TODO warn user
20:13:26:953 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:13:32:719 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:13:34:767 channel is 11 but should be 15, start channel change
20:13:34:767 Skip automatic channel change, TODO warn user
20:13:42:202 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:13:44:767 channel is 11 but should be 15, start channel change
20:13:44:767 Skip automatic channel change, TODO warn user
20:13:47:935 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:13:47:935 max transmit errors for node 0x00158D0001530785, last seen by neighbors 215 s
20:13:54:766 channel is 11 but should be 15, start channel change
20:13:54:766 Skip automatic channel change, TODO warn user
20:13:57:285 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:13:57:286 max transmit errors for node 0x00158D0001530785, last seen by neighbors 224 s
20:14:02:321 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:14:02:321 max transmit errors for node 0x00158D0001530785, last seen by neighbors 229 s
20:14:04:766 channel is 11 but should be 15, start channel change
20:14:04:766 Skip automatic channel change, TODO warn user
20:14:12:474 0x00158D0001530785 error APSDE-DATA.confirm: 0xE9 on task
20:14:14:767 channel is 11 but should be 15, start channel change
20:14:14:767 Skip automatic channel change, TODO warn user
20:14:21:915 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task
20:14:23:896 Current channel 11
20:14:23:927 Device TTL 2674 s flags: 0x7
20:14:24:769 channel is 11 but should be 15, start channel change
[... 7 lines cut ...]
20:14:33:875 ZCL attribute report 0x00158D000322BA27 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:14:34:468 delay sending request 236 dt 2 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:34:569 delay sending request 236 dt 2 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:34:666 delay sending request 236 dt 2 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:34:766 channel is 11 but should be 15, start channel change
20:14:34:770 Skip automatic channel change, TODO warn user
20:14:34:770 delay sending request 236 dt 2 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
[... cut ...]
20:14:36:068 delay sending request 236 dt 3 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:36:166 delay sending request 236 dt 3 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:36:267 delay sending request 236 dt 4 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:36:366 delay sending request 236 dt 4 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:36:466 delay sending request 236 dt 4 ms to 0x00158D0001530785, ep: 0x01 cluster: 0x0400 onAir: 1
20:14:36:516 0x00158D0001530785 error APSDE-DATA.confirm: 0xA7 on task

Klagio, 2020-06-04:
after 5 minutes, just 2 nodes

[image]

Smanar, 2020-06-05:
Yep it's the same log.
I don't see something special, except the device seem working but without connexion to devices ....

You can try select a node then press L, but I m almost sure the device will not receive the command.
You can try to power cycle a router.

Have you compared your config with this one on a previous link ?

On the hidden config, you have see a change ? like a channel change ?
It's the same conbee, but with a firmware upgrade ?

closed 2020-07-03 by stale[bot]

makke1971, 2021-08-23:
Hallo lese gerade dein Bericht hier .Vor ca 1 Jahr hattest du Probleme mit dem Verbinden deiner Geräte . Hast du das lösen können ?Ich habe jetzt das gleiche Problem . Und weiss nicht was ich machen soll. Kannst du mir helfen