dresden-elektronik/deconz-rest-plugin #2869: New Device Request: SmartThings SmartSense Multi Sensor

cjs28, 2020-06-03:
Hello,

I am looking to get help with the SmartThings SmartSense Multi Sensor, this the first SmartThings sensor that was delivered as part of the kickstarted campaign back in 2013. The sensor supports  movement, vibration, orientation, and temperature.

[SmartThings SmartSense Multi PDF GUIDE](https://support.smartthings.com/hc/article_attachments/115006467966/SmartSense%20Multi%20User%20Guide.pdf)

Node:
[image]

Cluster Info
[image]

Node Info Panel
[image]

OTAU Cluster Info
[image]

SwoopX, 2020-06-04:
It looks like the sensors hasn't been paired correctly or, to be more precise, some clusters seem to be missing in the first screenshot.

Can you please try to wake the device (just trigger it by open/close) and then read the simple descriptors? Must happen within a few seconds.

cjs28, 2020-06-05:
It has a button on the inside used for pairing that can be used to wake it. I tried that as well as triggering open/close. I even tried to move the sensor closer to the conbee II stick and tried the same steps, all while clicking Read Simple Descriptors and nothing has changed. I can see the interface lights on the node blink red and green in the Decons VNC interface, when performing these steps.

Heres a screen shot after trying it a few times, incase I am missing something:
[image]

SwoopX, 2020-06-09:
Hm, I'm afraid purely based on that information, chances are very low to get it working. It's probably just too old.

You may give it another try reading the simple desctiptors once again. It's the black bullet next to the selected one.

cjs28, 2020-06-14:
I tried a bunch of things to get this to sync, but it really just does the same thing no matter how I try to join it to the network. It seems this one has some "special" requirements. I found this forum post from the hubitat forum where they were able to get it to work, but seems it took a special driver.

https://community.hubitat.com/t/issues-with-multipurpose-sensor/264/26

I suppose it's time to look at some new door sensors.

SwoopX, 2020-06-15:
Yeah, took a look at that but it doesn't really help. Especially remotely, I can only act on the info the node is willing to present, which, in this case, is barely anything. Sorry to sayy that but with a new device, there's also a higher chance of getting some more "convenience" features.

SwoopX, 2020-07-09:
Closing this as we can't get the required information out of the device.

closed 2020-07-09 by SwoopX