dresden-elektronik/deconz-rest-plugin #2870: New device request: Sengled smart sensor

clockbrain, 2020-06-03:
Hi, it would be great if support for the Sengled smart sensor E1D-G73 could be added.

Clusters (though they don't look right - maybe mixed up with a different Sengled sensor floodlight I also recently added to my network?)
[image]

Node info:
[image]

Basic cluster:
[image]

I've added the model to de_web_plugin.cpp and got it paired. https://github.com/dresden-elektronik/deconz-rest-plugin/compare/master...clockbrain:sengled_e1d
It shows in the web API but not in Phoscon. The sensor shows and registers state changes in Home Assistant via the deconz addin.

Is there something else I need to change to get it showing in Phoscon? Also, the battery level isn't showing in the Api.

```
    "19": {
        "config": {
            "battery": null,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "5f39d80585f247d2bc3f29ef511f9bc6",
        "lastseen": "2020-06-03T04:23:08.753",
        "manufacturername": "sengled",
        "modelid": "E1D-G73",
        "name": "E1D-G73",
        "state": {
            "lastupdated": "2020-06-03T02:55:12.418",
            "lowbattery": false,
            "open": false,
            "tampered": false
        },
        "type": "ZHAOpenClose",
        "uniqueid": "b0:ce:18:14:00:0e:72:6a-01-0500"
    },
```

SwoopX, 2020-06-04:
> Clusters (though they don't look right - maybe mixed up with a different Sengled sensor floodlight I also recently added to my network?)

What should have been mixed up exactly?

Attribute reporting hasn't been taken care of yet, so no battery. What does the device offer after the power configuration attributes have been read?

clockbrain, 2020-06-04:
> What should have been mixed up exactly?

I had been simultaneously playing with a Sengled E13-A21 floodlight and as per https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2155#issuecomment-598442898 I was trying to get the 0500 IAS motion detection zone showing for that device.

So I was surprised when this contact sensor showed an 0500 IAS zone. But now that I look more closely at the cluster for this contact sensor it looks ok.
[image]
[image]

Here's the Power Configuration cluster attributes for the E1D-G73:
[image]

SwoopX, 2020-06-04:
Ah, ok. Does it work for the bulp now?

clockbrain, 2020-06-04:
Nope. I got the metering to work but can't get the 0500 IAS cluster to appear on that bulb no matter how many times I try. I've factory reset it and paired it again but no better.

SwoopX, 2020-06-04:
I'll jump into the original issue

clockbrain, 2020-06-04:
Is it worth also adding the IAS zone cluster check for this E1D-G73 contact sensor  https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L4127

I see there is a bunch of other contact sensors being checked there.

I could add it myself to my setup but I've no idea what that cluster actually achieves, i.e. how I would test it.

SwoopX, 2020-06-05:
Already did that with ad423ce

SwoopX, 2020-06-18:
Closing this as PR has been merged. Supported with .78

closed 2020-06-18 by SwoopX

stevenwfoley, 2020-09-17:
Hello,
I have a couple of these Sengled E1D-G73 sensors that are not working in deconz v2.05.82. The device shows up in the deconz gui, but is not reading/populating the clusters. The dot for the drop-down cluster menu is not even showing on the object in the gui.

SwoopX, 2020-09-17:
@stevenwfoley Please raise a seperate issue on this one to follow up, thanks!