dresden-elektronik/deconz-rest-plugin #2872: When opening http://..../pwa/login.html i get This Browser is not supported :-(

bkemmler, 2020-06-03:
I get with the actual Firefox and chrome browser the same message:

This Browser is not supported :-(

Please use one of the supported browsers:

    Chrome
    Firefox
    Safari
    Samsung Browser
    Edge

This browser identifies itself as:
Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:76.0) Gecko/20100101 Firefox/76.0

i checked the console output

This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. login.html
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.
login.html
Error: SyntaxError: invalid assignment left-hand side login.html:654:13
Error: ReferenceError: Backbone is not defined login.html:654:13
SyntaxError: invalid assignment left-hand side
backbone-min.js:1:19
ReferenceError: Backbone is not defined
scanner.js:3:113
jQuery.Deferred exception: app.backButton is null @http://..../pwa/js/login.js?a1493a:3:9697
j@http://..../pwa/library/js/jquery-min.js?0baf29:2:29999
g/</k<@http://..../pwa/library/js/jquery-min.js?0baf29:2:30313
 undefined jquery-min.js:2:31579
Error: TypeError: app.backButton is null login.html:654:13
TypeError: app.backButton is null

This happened after updating (apt update/upgrade) 

This is my current OS version: 
~# uname -arm
Linux orangepizero 5.4.43-sunxi #20.05.1 SMP Sat May 30 20:32:43 CEST 2020 armv7l GNU/Linux

Any idea how i can fix the problem? Thanks.

Mimiix, 2020-06-04:
You are better off posting in the Phoscon repo:
https://github.com/dresden-elektronik/phoscon-app-beta

Mimiix, 2020-06-05:
Ignore my last comment, this issue will be moved!