dresden-elektronik/deconz-rest-plugin #2873: Conbee 1 -> Conbee 2 cloned. Need info on how to sniff with Conbee 1

drbytes, 2020-06-03:
Hello

Because of reasons I have now replaced my Conbee with a Conbee 2. To get there I backed up the config via PWA and 'cloned' the Conbee with the ALT-Advanced click/Zigbee configuration.

It works.

Now I'd like to use the old Conbee to snoop on Zigbee traffic but how do I tweak the Conbee so that it has another mac, or whatever, so it doesn't interfere with the Conbee 2?

ebaauw, 2020-06-03:
Normally you'd set the MAC address through the network settings in the deCONZ GUI.  Not sure if this is needed for the ZShark firmware though - I think it only listens in on the radio traffic, but doesn't send anything.

drbytes, 2020-06-04:
Ok, thx, I'll give it a shot

closed 2020-07-03 by stale[bot]