dresden-elektronik/deconz-rest-plugin #2874: deCONZ hassio-addon 2.05.77 to 2.05.75 downgrade

luke3butler, 2020-06-04:
This morning home assistant updated on my server. 

deconz lights (all Phillips Hue bulbs) no longer worked. Meshes looked fine when viewed through VNC, and controlling lights through the Phoscon web app worked fine, but no lights were working from HA and were listed as offline. Switches worked fine, though.

After "updating" to 5.3.5 [2.05.75] (and rebooting 3 times) everything was working again, except for functionality included in recent updates (no more Xaiomi switch of course).

I didn't see a bug report in place for this other than [this bug in hassio-addons](https://github.com/home-assistant/hassio-addons/pull/1375) along with the unclear references, hence this new issue.

Mimiix, 2020-06-04:
It is being investigated by the devs and DE on the discord server. I(or somebody else) might create a issue later in which i keep updates from there.

Mimiix, 2020-06-04:
Updates will made here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2875

closed 2020-06-05 by Mimiix