dresden-elektronik/deconz-rest-plugin #2876: Device Support: Centralite 3-series Switches

disconn3ct, 2020-06-04:
The Centralite 3-series switches provide short and long detection of up and down in a battery powered device. I've been using it successfully in smart things for a year or more, but I'm not sure if I had to do anything stupid to get it working. (Nothing jumps out in the dev console, so it might have Just Worked.)

[2datasheet-3seriessmartswitch.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4730581/2datasheet-3seriessmartswitch.pdf)
[image]
[image]
[image]
[image]
[image]

disconn3ct, 2020-06-26:
Activity

disconn3ct, 2020-09-11:
activity. I mean.. what else should I provide? Are there docs for doing this ourselves?

Mimiix, 2020-09-14:
@disconn3ct  Bumping it up ;)

I'll keep it open.

disconn3ct, 2022-07-08:
Is there any documentation about adding new devices ourselves?

Mimiix, 2022-07-08:
Check the ddf cheat sheet in the wiki :)