dresden-elektronik/deconz-rest-plugin #2878: Raspbee 2 is not recognized on Raspberry

ebrger, 2020-06-04:
Hello, I have a Raspbee 2 since today. I installed it with the help of the manual. Phoscon or deconz is also reachable. But it does not show me any information under device.
What am I doing wrong?
i have already made an lsusb, it does not appear there. 
thank you
[image]

StephanStS, 2020-06-06:
Hello @ebrger, 
which Pi hardware do you use? 
Did you went through the installation via https://phoscon.de/de/raspbee2/install
Or did you take this: https://phoscon.de/de/conbee2/install

As far as I assume the Raspbee II is connected via the GPIO connector (-> I2C) and not via USB (this would be a Conbee II). So a lsusb will not show the Raspbee2.
In every case, if you have a Raspbee II, [this](https://phoscon.de/de/support#conbee2-connection-issues) will not help. 

Hello Dennis (@Mimiix), 
is the Rasbee II connected via I2C? Then I propose to add a section 9 "Testen der Ansprechbarkeit des Rasbee II Moduls" in the [installation manual](https://phoscon.de/de/raspbee2/install) regarding checking whether the module is detected by the Pi hardware (via i2cdetect command).

The same could be in the Conbee II installation (you have a perfect description already [there](https://phoscon.de/de/support#conbee2-connection-issues)). Checking this during the installation gives a clear step-by-step advise. Of course this is needed for all different OS types...

ebaauw, 2020-06-07:
The RaspBee and RaspBee II use the Pi's internal serial port on GPIO pins 14 and 15 (Broadcom numbers, not the physical pins).  See https://pinout.xyz/pinout/raspbee.  In additional to that, the RaspBee II uses I2C for the real-time clock.  I think they also use another GPIO pin for hardware reset.

StephanStS, 2020-06-08:
@ebaauw: Hello Erik, do you know whether there is a small command line test program checking the correct presence of a Raspbee II board which eberger might test?

ebaauw, 2020-06-08:
```
$ sudo GCFFlasher_internal -r
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
deCONZ firmware version 26520700
Bootloader version 0x00030003, app crc: 0xD4E57E35
```

closed 2020-06-14 by ebrger

StephanStS, 2020-06-14:
@ebrger: Would you post here whether and what solved your issue?
Thanks for closing it.

ebrger, 2020-06-14:
Switched to conbee2 with usb after a connection could not be established even with a second raspbee and an image from Phoscon.

Tim-AQ, 2021-02-21:
Hi,
I'm havint the same issue. When using the CGFFlasher to reset it i get this output:

```
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)

952: Error: failed to open device, status: Permission error while locking the device (2)
```

Any idea the issue is?

StephanStS, 2021-02-21:
Did you use ‚sudo‘?

Tim-AQ, 2021-02-21:
> 
> Did you use ‚sudo‘? ----------------------------
> […](#)
> I'm havint the same issue. When using the CGFFlasher to reset it i get this output: GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh Reboot device RaspBee (RaspBee) 952: Error: failed to open device, status: Permission error while locking the device (2) Any idea the issue is?

Sorry, deleted the original comment as a managed to resolve the issue.

I did use sudo, but got the error because deconz was running. The issue turned out to be the powersupply. Didnt notice any stability issues other than raspbee behaving inconsistent. However, replacing it solved the issue.