dresden-elektronik/deconz-rest-plugin #2879: New device request - TS0601 / TZE200_nkjintbl

bkupidura, 2020-06-04:
Hey,
I got new zigbee wall switches, but they are not supported by deconz.

Is there any chance to add support for those switches?
[image]
[image]
[image]
[image]
[image]

DJBenson, 2020-06-06:
This is almost identical to the ticket I opened today (#2884). I think these are all Tuya based devices and there are tonnes of them out there, under lots of different brand names, so it would be great to get them in deCONZ.

Your device mac address has the same prefix as mine (0xEC1BBD) but your model number is different to mine (mine is TS0012 and TS0013 - there's also a TS0011, the last number signifies the number of gangs on the switch e.g. 1/2/3).

Smanar, 2020-06-07:
But they are light, so they can work natively, the device is not visible at all in the API ?

DJBenson, 2020-06-07:
I appreciate we are having a cross platform discussion as I’m also discussing this on Discord but how do we check if the device is in the API and as I also mentioned, the device isn’t showing any clusters to allow the device to be controlled (e.g. on/off) - is that normal?

Smanar, 2020-06-07:
Yes was exactly that.

So can run run deconz in debug mode and make a new inclusion pls ?

Smanar, 2020-06-07:
No need
I have understand the problem, I can make a modification on code, do you know how to test it ?

DJBenson, 2020-06-07:
> Yes was exactly that.
> 
> So can you run deconz in debug mode and make a new inclusion pls ?
> 
> Edit:
> No need, I have just see the problem.
> 
> No information how this remote make request to other devices ?
> Using cluster EF00 ?
> Using not detected cluster in deconz ?

I seem to have an email for a reply that’s no longer here. The last email was about a code fix and testing...

I’m fairly competent compiling the source if I can test it for you?

Smanar, 2020-06-07:
Yep, I have edited my post.
I know how to make the device visible, but not sure the cluster I will create can be used by the device.

I m searching info on other zigbee projecy atm.

DJBenson, 2020-06-07:
Great, thank you so much for looking into this. If I can do anything I’m happy to help. As I mentioned on Discord I have the 2 and 3 switch models so can test those if/when you find a solution.

DJBenson, 2020-06-07:
Found this article on decoding the Tuya cluster format;

https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d

It’s linked from the zigbee2mqtt repository so you’ve probably already stumbled upon it but adding here for anybody else who is interested.

Smanar, 2020-06-07:
I have my answer, I m on a modification on xml file to test.

Smanar, 2020-06-07:
Ok so now there is job for you ^^

https://github.com/Smanar/deconz-rest-plugin/commit/a5cad08f9d451bfaaec0fd16b9924a7c9455e859

It's just an xml file to edit, so you just need a text editor and reload deconz.

If I m right on your node on cluster EF00 previously "unknow" you will have now something new.
For the moment there is only one command to try

Status : always 0
TransID : buton id : 13 / 14 / 15
Dp: state : ??? : 257 /258 /259
fn : always 0
data : state : "1" for "on" or "0"

For the moment you can only send command, not receive them, ofc the switch need to be wired with a bulb to test.

I m not sure at all for values, for the data for exemple IDK if you need to use hexa or not.

DJBenson, 2020-06-07:
mmm the cluster is still showing as unknown.

The general.xml file I edited was at /usr/share/deCONZ/zcl - which is what is referenced in the deCONZ UI.

I switched from service (headless) mode to GUI mode to check on the network overview.

[image]

DJBenson, 2020-06-07:
There's a typo in your cluster ID (0xfc00) which I presume should be 0xef00 - amended and reloaded and the cluster shows up.

Smanar, 2020-06-07:
Oups my bad, Wrong number https://github.com/Smanar/deconz-rest-plugin/commit/d2498dcac3dd6385ed27dc89f01a1c55c5d74be9

DJBenson, 2020-06-07:
n/p I manually amended it when I saw it looked odd. Cluster is there now. Will see what it does.

Smanar, 2020-06-07:
Will be hard with random mode, sorry.

But later I will make something to retreive value from the device, will be easier to find missing information.
It will be visible using deconz log (I think, but not sure because this cluster is not implemented at all) or cli plugin too.

DJBenson, 2020-06-07:
Tried sending some data in both decimal and hex and the device didn’t respond I’m afraid. I will have a play later when my kids have gone to bed.

SwoopX, 2020-06-07:
You most likely need to have a binding for that cluster as well, otherwise you don't necessarily see everything.

Smanar, 2020-06-07:
Yep, but this command will at least turn the light off or on, even without return.

I don't have enought time today, but tomorrow I will make the code write some log according to cluster return, will be usefull, because using "magic number" .....

Smanar, 2020-06-08:
Ok so first try https://github.com/Smanar/deconz-rest-plugin/commit/8b6e0ffe4710faba2c4437f870aacc2671f223f0

If someone can try this code ? Realy experimental, but nothing hard to understand.

With this code, you will have debug line with information when you press a button on the device, try to mimic the value in the deconz GUI.

DJBenson, 2020-06-08:
So I'm running deCONZ with the debug options? What level is required? From the wiki;

deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1

...but that seems very busy - is that level necessary?

Smanar, 2020-06-08:
I m using -dbg-info=1 for my debug message.
But -dbg-info=2 is enought.

All my log will start by "Tuya debug" and a number, you can use that as filter.

You haven't problem for compilation ? Because I haven't tested the code on my side, so I can have make mistake.

DJBenson, 2020-06-08:
No it compiled fine.

Going to go and connect the light and see what I get back.

DJBenson, 2020-06-08:
If I grep the output by "Tuya debug" will that capture what we're looking for? Even with -dbg-info=2 the output is very chatty.

Smanar, 2020-06-08:
NP, I think "-dbg-info=1" is enought for my log, you can try "=0" on other, IDK the defaut value.

But yes I know, if you have lot of devices, it s a real problem.

DJBenson, 2020-06-08:
Mmm I'm not seeing any debug output from the switch.

1) Do I need to leave the general.xml change in place?
2) Do I need to delete and re-bind the switch?

Smanar, 2020-06-08:
No, you can use the general.xml you want, better to use the modified one.

And no you don't need to delete and re-include the device, and from the link it works without bind.
Every time you send a command 0x00, it will answer by 0x02.

You have take the code in the tuya branch ? not the master ?

DJBenson, 2020-06-08:
So;

- Using same XML as yesterday
- Code cloned from "tuya" branch (confirmed my local version has your two changesets in)
- I've sent 0x00 from the GUI (no other values changed)
- Toggled the state of the physical switches

I don't get any Tuya related debug at level 1 or 2.

DJBenson, 2020-06-08:
Just started putting random combinations of values in those boxes and we have some movement - still no debug but the switch is responding from the UI;

Status = 0x00
TransID = 0xD
Dp = 0x101
fn = 0x00
Data = 0x00

Results in switch 1 toggling on/off (sending the same payload toggles the state)

Status = 0x00
TransID = 0xD
Dp = 0x102
fn = 0x00
Data = 0x00

Results in switch 2 toggling on/off (sending the same payload toggles the state)

SwoopX, 2020-06-08:
You may need the cluster binding. It was required for the WAXMAN leak sensor as well.

Presumably, the device uses a different data format than in the link (or at least different values)

DJBenson, 2020-06-08:
TransId (0xD,0xE,0xF) seems to make no difference to the outcome - the switch value seems to be in Dp.

DJBenson, 2020-06-08:
Further info - I've paired the device with zigbee2mqtt with a cheapo sniffer and TransId is just incremental.

DJBenson, 2020-06-08:
**Toggling switch 1 on then off**:
`
zigbee2mqtt:debug 2020-06-08 23:12:44: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":46,"dp":257,"fn":0,"data":{"type":"Buffer","data":[1]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:12:44: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":47,"dp":258,"fn":0,"data":{"type":"Buffer","data":[1]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:12:44: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":48,"dp":259,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:12:46: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":48,"dp":257,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:12:46: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":49,"dp":258,"fn":0,"data":{"type":"Buffer","data":[1]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:12:46: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":50,"dp":259,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0`

**Toggling switch 2 on then off**;
`zigbee2mqtt:debug 2020-06-08 23:16:12: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":54,"dp":259,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:16:12: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":53,"dp":258,"fn":0,"data":{"type":"Buffer","data":[1]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:16:12: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":52,"dp":257,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:16:15: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":56,"dp":259,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:16:15: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":55,"dp":258,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0,
zigbee2mqtt:debug 2020-06-08 23:16:15: Received Zigbee message from '0xec1bbdfffe69e06b', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":54,"dp":257,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0`

Smanar, 2020-06-09:
But data is string not hexa decimal value, from my memory I have disable the Hexa mode for this field.

And I don't see the problem I can have in y code.
- Perhaps there is no manufacture field in the request, so can you change
```
        case TUYA_CLUSTER_ID:
            // Tuya manfacture cluster:
            if (zclFrame.manufacturerCode() == VENDOR_EMBER)
            {
                handleTuyaClusterIndication(ind, zclFrame);
            }
            break;
```

in 
```

        case TUYA_CLUSTER_ID:
            // Tuya manfacture cluster:
            handleTuyaClusterIndication(ind, zclFrame);
            break;
```

- Have you tried to bind this cluster to the conbee ?

BTW, have you a zigbee sniffer that can be used to sniff the request made by the device ?

DJBenson, 2020-06-09:
I will try the code mod a bit later. 

Yes I have a C2530 dongle, no idea what firmware is on it but it worked with zigbee2mqtt. 

When you say “bind to Conbee” what do you mean? The device is in my network if that’s what you mean.

Smanar, 2020-06-09:
Nope, There is a bind feature in deconz https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1979

But I don't think it s usefull, from the post the working mode is natif, but it's something you can try if nothing is working.

DJBenson, 2020-06-09:
OK so I can see the debug messages now, but really not sure what I'm looking for, what you need me to send etc.

I've sent some random commands in the data field as string;

on: Tuya : debug 023:20:26:**887**
off: Tuya : debug 023:20:46:**887**
test: Tuya : debug 023:22:11:**957**
toggle: Tuya : debug 023:23:01:**976**
dim: Tuya : debug 023:23:56:**888**

I have also flashed the sniffer firmware and set it looking for traffic to or from the address (0x9715) of the switch but it's not capturing anything (with the filters off I can see the rest of my network but I'm a novice with Wireshark so not sure how best to filter). Currently I'm using zbee_nwk.src == 0x9715 or zbee_nwk.dst == 0x9715 but that isn't returning anything.

DJBenson, 2020-06-09:
I have some pcap files now if you want them (saved two files, one for toggling switch 1 on/off and one for toggling switch 2 on/off).

Smanar, 2020-06-11:
>I'm not massively familiar with calling the API directly but I fired up Homebridge and homebridge-hue pulled in the switch and I can toggle it from HomeKit so that's great work!

So you can send order the the device (at least the first button) ?

>I guess you might expect this but the API isn't keeping sync with the switch which is causing some weird behaviour - I can describe them in detail if required but I think I'm probably getting ahead of where the code is actually at.

Yep, for the moment I have just start the part to send request, so for the moment the code don't use the received request, like state.

For your ghost device, it's a long story, there is an issue about it https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2064 , if you are lucky, you just need to wait some days and retry to delete it.

For the pcap, we will see later, don't worry, for the moment I ha ve all the informations I need for this device.

For the log, It's strange you haven't the "23:48:38:250 Tuya : debug " like the previous one ?
But I think the "force poll" is something normal. Deconz detect this device don't use attribute reporting, so it use pool, but unfortunately for him, the clusters aren't standard ^^.

Now the bigger problem, how to integrate the device in the API, you have 3 "light" but only 1 endpoint, first time I see that.
I think the better way is making 3 devices, but what to use as UniqueID ?

DJBenson, 2020-06-11:
> So you can send order the the device (at least the first button) ?

Yes indeed, I could turn the light on and off via API calls (using homebridge-hue)

> For the log, It's strange you haven't the "23:48:38:250 Tuya : debug " like the previous one ?

This is strange but no, there were no Tuya debug events this time.

> Now the bigger problem, how to integrate the device in the API, you have 3 "light" but only 1 endpoint, first time I see that.
> I think the better way is making 3 devices, but what to use as UniqueID ?

Is there an existing multi-switch device which is supported by deCONZ which can be used? Elsewhere I saw that unique ID is being made up of the model number and suffixed with the NWK ID.

Smanar, 2020-06-11:
Ok I have find why there is not debug line ^^ my fault.

I m searching another device with this kind of "hack"

schumi2004, 2020-06-11:
Just checking in. Going to order a few of these switches.

Smanar, 2020-06-12:
But the integration will take some time ^^.
ebaaw have give me a solution, you will have 3 device, one by button, with fake Endpoint. But I m sure it will not work at first try.

Smanar, 2020-06-13:
Ok so new version to try https://github.com/Smanar/deconz-rest-plugin/tree/tuya

It's a critic version for 2 raisons:
- It s my first test to force the creation of 2 more node, and the code is a little agressive for me.
- I have put so much log in the code to have the maximum information

So you will nee to re-include the device and get all the debug line during the inclusion.
You can check if the code have working > you will have 3 more device, but no more change on working mode, so you don't nee to test it.

DJBenson, 2020-06-13:
3 devices? Is that 3 for the 3-button switch, one device per physical switch? Should I expect 2 devices for the 2-button switch?

Are status updates working? Should a change to the physical switch update the software switch?

Just trying to understand how much I need to test or if you literally want me to include and capture the debug, or if I should capture debug controlling the device too.

Smanar, 2020-06-13:
Yep, we will have one entry by device.

Nope, just include the device, and capture logs during inclusion, this code is just for testing the  device creation and give me some info about the device, I haven't make changes on the rest of code.

It,s the harder part for me, because it's the first device on deconz that will use this method.

DJBenson, 2020-06-13:
It seems logical to me to have one device per switch so I kinda like that. 

Same as usual, I will test this later this evening when the kids are asleep 🤣

DJBenson, 2020-06-13:
Unfortunately I get a compiler error;

`de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::addLightNode(const deCONZ::Node*)’:
de_web_plugin.cpp:1600:41: error: no matching function for call to ‘QList<deCONZ::SimpleDescriptor>::append(const deCONZ::SimpleDescriptor**) const’
     node->simpleDescriptors().append(&sd);
                                         ^
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qobject.h:49,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qcoreapplication.h:46,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QCoreApplication:1,
                 from de_web_plugin.cpp:11:
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qlist.h:579:27: note: candidate: ‘void QList<T>::append(const T&) [with T = deCONZ::SimpleDescriptor]’
 Q_OUTOFLINE_TEMPLATE void QList<T>::append(const T &t)
                           ^~~~~~~~
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qlist.h:579:27: note:   no known conversion for argument 1 from ‘const deCONZ::SimpleDescriptor**’ to ‘const deCONZ::SimpleDescriptor&’
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qlist.h:962:13: note: candidate: ‘void QList<T>::append(const QList<T>&) [with T = deCONZ::SimpleDescriptor]’
 inline void QList<T>::append(const QList<T> &t)
             ^~~~~~~~
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qlist.h:962:13: note:   no known conversion for argument 1 from ‘const deCONZ::SimpleDescriptor**’ to ‘const QList<deCONZ::SimpleDescriptor>&’
make[1]: *** [Makefile.Release:638: release/de_web_plugin.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/domoticz/code/deconz-rest-plugin-dev'
make: *** [Makefile:40: release] Error 2`

Smanar, 2020-06-14:
Arf, my bad, theses lines was useless, just a bad copy/paste.

I have updated the code, with some more loging

DJBenson, 2020-06-14:
With the new code I get a segmentation fault when deCONZ tries to load. If I switch back to the stock version of libde_rest_plugin.so, deCONZ loads fine.

Smanar, 2020-06-14:
Ha ? You haven't more information on error message ?

I have pull a new version, It have worked on my machine. If you have the problem when deconz load, I think it's ok now.

DJBenson, 2020-06-14:
Same issue - I've run deCONZ using dgb and it outputs the following when the crash occurs (which is on trying to connect to the Conbee II from the device list screen)

#0  0xb6f38fb4 in deCONZ::ZclCluster::id() const () from /usr/lib/libdeCONZ.so.1
#1  0xa76be5e8 in DeRestPluginPrivate::addLightNode(deCONZ::Node const*) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#2  0xa76dd99c in DeRestPluginPrivate::nodeEvent(deCONZ::NodeEvent const&) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#3  0xa77b4b64 in DeRestPluginPrivate::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#4  0xb606f244 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#5  0xb6f59db0 in deCONZ::ApsController::nodeEvent(deCONZ::NodeEvent const&) () from /usr/lib/libdeCONZ.so.1
#6  0x0005eac0 in zmController::loadNodesFromDb() ()
#7  0x00069b54 in zmController::readParameterResponse(ZM_State_t, ZM_DataId_t, unsigned char const*, unsigned short) ()
#8  0x00046b8c in zmMaster::processPacked(zm_command*) ()
#9  0x000475ac in zmMaster::onPacket() ()
#10 0x000b5d18 in zmMaster::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#11 0xb606fbec in QObject::event(QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#12 0xb69abdb4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#13 0xb69b42a8 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#14 0x020f4958 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Just so you know how I've been testing so far, in case something is wrong here;

Stop deCONZ (service mode)
Make a backup of /usr/share/deCONZ/plugins/libde_rest_plugin.so
Copy the newly compiled plugin back to the same directory
Start deCONZ in GUI mode

Smanar, 2020-06-15:
No you are right, but like I have said the code is critic ATM and I need to use lot of security check to prevent this kinf of bug.

for exemple on precedent code I m using 
>sd->inClusters()[4].id()

Because for me, you have this cluster, but if you haven't c++ don't forgive and it s a crash.

I have male some tries on my side with fake value, And I think the code is good now

DJBenson, 2020-06-15:
Yeah I'm totally cool working through these.

I've just compiled and started up deCONZ and captured the attached (--dbg-info=2)

[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4781413/deconz.log)

Smanar, 2020-06-15:
Ok, so all was nice for me, all debug validated (I hope for good device)
>15:31:42:000 Tuya : debug 10
15:31:42:000 Tuya : debug 11
15:31:42:000 Tuya : debug 12
15:31:42:000 Tuya : debug 13

Have you 3 devices after a new Tuya inclusion ?

DJBenson, 2020-06-15:
Yep

[image]

[image]

Smanar, 2020-06-15:
Ok so nice new.
I can finish the code.

Smanar, 2020-06-15:
Ok new version to test.
Some explanation if you want edit the code yourself.

I m making 3 devices with endpoint 01 to 03.

When you send a command I take the endpoint and send comand according to it
>           qint16 button = 0x0101;
            if (ep == 0x02) { button = 0x0102; }
            if (ep == 0x03) { button = 0x0103; }

101 is 257 in decimal, it's the value you have used in deconz

When you receive a command I take the button value and choose the endpoint according to it
>                uint ep = 0x01;
                if (dp == 0x0102) { ep = 0x02; }
                if (dp == 0x0103) { ep = 0x03; }

It's the theory ^^, this time, you don't need to re-include the device, because I will use the same then previous inclusion, and you can test as you want, looking for debug message if something is not working.

The code is compiling, and I don't think you can have crash, just not works.

DJBenson, 2020-06-15:
Think I need to re-bind the device - the switch doesn't appear to be responding.

[image].

Will try binding again shortly.

DJBenson, 2020-06-15:
BTW the switch(es) are visible in the Phoscon app;

[image]

[125 more replies]

closed 2020-10-01 by manup