dresden-elektronik/deconz-rest-plugin #2880: Trying to use HTTP basic authentication on the Conbee2

acaral, 2020-06-05:


acaral, 2020-06-05:
I am trying to use the API to automate the installation of my Home Assistant.
I don,t want to interact with the Phoscon APP at all. I am planning to add the devices and the authorization using a  script. The problem is I need username:password to add to the header of the request but I cannot find the username. Can you help me with that?
Also, can I add a switch using this API?

SwoopX, 2020-06-05:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

acaral, 2020-06-05:
Hello SwoopX, I searched that doc before asking. I tried the the basic authorization but it not worked for me. I could not find the username of the conbee2. I tried with the name of the device but I always got user not authorised. I am using Postman to communicate with the API. Can you tell me is the username is the name of the device?

SwoopX, 2020-06-05:
I'm not sure how much more explicit it can be if you thoroughly read the documentaiton based on the provided information

[image]

Kane610, 2020-06-06:
@SwoopX he wants to login to Deconz with username and password prior to generating the api key

SwoopX, 2020-06-06:
Why not say so directly. Would have never deduced that it could be about Phoscon, especially since you don't need any username for first login.

Anyway, default username/password for both webapps is `delight`

acaral, 2020-06-06:
Now it is working. Thanks. I guess you  should put that information on the Doc.
anyway, can I scan and add devices using the API?

SwoopX, 2020-06-06:
Yes, it's described in the API documentation.

Seems like you caught on of the rare cases where you're not directly asked to set a new pw. Usually. that should be done on deconz restart...

acaral, 2020-06-06:
I am ask to set a new password but is does not say anything about username. But, this method is perfect because I don't want to interact with the Phoscon App.

parruda, 2020-09-11:
@acaral How did you figure that out? When I make a POST request to `/api` I get an username back. Then on authorization it says I should authenticate with Basic (base64 of username:password). Well, where's the password?

Kane610, 2020-09-11:
http://{host}:{port}/api/{api_key}