dresden-elektronik/deconz-rest-plugin #2881: Develco smoke sensor SMSZB-120: `config.battery` missing

ebaauw, 2020-06-05:
`config.battery` is missing from the ZHATemperature resource for the Develco smoke sensor SMSZB-120 after startup.  When the battery value is retrieved, it's added to the resource.  The ZHAFire resource has `config.battery` alright from startup.

The _Power Configuration_ cluster is on the same endpoint as the _IAS Zone_ cluster (used for the ZHAFire); the other endpoint with the _Temperature Measurement_ cluster doesn't have a _Power Configuration_ cluster.  Consequently, it's missing from the resource's fingerprint, causing `config.battery` not to be created on startup.

I'm not sure what the proper behaviour should be.  I don't know any other battery-powered devices with resources on multiple endpoints.  The behaviour should be consistent, though: either expose `config.battery` from startup, or don't expose it.

SwoopX, 2020-06-05:
Good observation and agree on consistency. I'd have a look later and indicate my preference

ebaauw, 2020-06-05:
For now, I patched my database to include the cluster in the fingerprint.

Mimiix, 2022-01-12:
Isn't this fixed already>?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5547

ebaauw, 2022-01-12:
Yes

closed 2022-01-12 by ebaauw