dresden-elektronik/deconz-rest-plugin #2882: Close / Open deconz with conbee II results in loss of the switches

daluma, 2020-06-05:
Ihm running conbee II with deconz - using restapi

versions: 2.05.75 / 08.03.2020  / 26580700

Everything is working fine with some quara sensors, vibration and temp + I added 2 Hue On / Off Plugin Units and the outdoor motion detector - everything works as expected so far - but in case I close deconz - at the next startup the two hue On/Off Plugin Units are missing as devices - if this happens they are also not available through the rest api. 

To fix this I need to re-search again for the switches - after they found everything is OK - until I close open deconz - same again.

Why the hue on / off plugin units missing at each startup and I need to search again for them. As described all other sensors static and never missing at startup of the deconz.

Seems this is a failure.

Thanks,

SwoopX, 2020-06-05:
I'm not aware if those are supported yet. Please check the supported devices list in the wiki. In case they're not, please make a device support request out of it (info also available in the wiki). Thanks!

daluma, 2020-06-05:
@SwoopX - the device is supported -

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

It is the - 

Philips | Hue Smart Plug LOM001

SwoopX, 2020-06-05:
That is, as to my experience, typical if the resources are not properly created and saved to the DB. Could you share a screenshot of the available clusters from deconz GUI? Also, ist LOM001 the model identifier from basic cluster? Couldn't find any issue related to the device on the first glimpse.

daluma, 2020-06-05:
@SwoopX - Here are some screenshots but I'm unsure what exactly I should provide for / from the cluster? You mentioned not properly saved to DB - how to solve that? in case this is the issue?

[image]
[image]
[image]

duffbeer2000, 2020-06-05:
That are screenshots of Phoscon and not from deconz GUI [Please read this first](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies)

Smanar, 2020-06-05:
It's something visible on log no ? 
I had a similar problem > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2390

daluma, 2020-06-05:
@duffbeer2000, Thanks that was helpful, Hope I grabbed all information needed
@Smanar, How did you solved your issue?

[image]
[image]
[image]
[image]
[image]

daluma, 2020-06-05:
Independent of the above I noticed that if I create a Group this Group is lost too on the close / open of deconz - is the problem that deconz is not saving for whatever reason the groups and the devices into the database?

Smanar, 2020-06-06:
Yep it's solved, but the same issue, in yous situation your device is "normal", and it's a router, so less problem.

For me, I think the only way to find something usefull is using log. In it you can see
- if deconz save the database
- if deconz load it after a restart
- if deconz freeze on a command

daluma, 2020-06-06:
@smanar, Thanks for reply 

For me the plugs useless in case I need to reboot the system I loose the plugs from a functional perspective - so they turn at max into routers / repeaters for the network.

I looked into the zll.db with sqlite - it become clearer now ... the two plugs can be found only under devices added - but not seen e.g. under sensors (assume it should be there due that all other devices independent of class also there) with the values - it turns out that all activities I do e.g. adding - add only under devices but not with the resining values I would expect - doing an remove shows that even this is not working and the devices stay forever - this is reflected even that I remove the devices under runtime the entires stay there and deconz show the devices alway as part of the network at next restart.

Any other component from database perspective behave normal - Adding / Removing / Storing in database e.g. Hue Motion Sensor, Aquara Vibration, Temp etc. I can see under sensors even the deleted status  attempts for other ones.

So the status for the "supported" device is - You can pair them "once" - device stay even trying to remove in the database / network forever due that remove. All activities in deconz kinda runtime for this device. For test purposes i created a copy of the database - and deleted under devices the corresponding "old" entries. id 2, 3 - I reset my plugs manually. Then I again paid with the network - now they changed id to 23, 24 - but everything is the same.

Maybe I'm wrong at any of the points above - but at the end the plugs behave completely different than any other device - proofed by adding and removing other devices too.

Appreciate with the above information any advise. E.g. what kind of log level can be beneficial.

Thanks,

Smanar, 2020-06-06:
If the device is not full integrated in deconz, you will not have sensor device for it (so no power mesure for exemple), but it will be in light section for sure, and it's enought to have it working.

If you check only the light section in database is enought. You need to find it after a restart, or just after closing deconz.

If you can compile the code, we can finish to include it, to have sensor part too, but I don't think it's the problem.

daluma, 2020-06-06:
The device is list under the full support devices so I was under impression it should work.
It is "only" in the lights section from Phoscon under runtime - I can't see in the database a lights table. I can see the devices "only" in the devices table properly added. 

Not sure what do you mean with compile the code - I would assume it should not be in sensors morelikly in lights which not exist - cause the plug is a simple one / off device not measurement capability.

In regard to your last question not really sure - cause - I wanted to simply use deconz / conbee II as a user and only bought things on the support list to end up now with R&D work .... really was not my intention ...

Smanar, 2020-06-06:
Lol, my last question was for someone that know better than me how deconz use database ^^, I m not sure of me.

But you are right, if the plug is a simple one, it can be normal he haven't sensor part.

Wait 10 mn, I will search the line to check on my previous issue.

daluma, 2020-06-06:
@Smanar, as long I do not "rescan" after the start - scan for new "lights" ... the devices simply not available by the API - after the scan and the devices reappeared - the API shows them too and I can use them via API ...

Smanar, 2020-06-07:
Can you start deconz with debug enabled ?

daluma, 2020-06-07:
@Smanar , Thanks for your support  - can you advise on windows command line for Debug - Any preferred Log Level in case there are different ? - where is the log file then located?

daluma, 2020-06-07:
@Smanar - here are some logs and excerpts :

IEEE.txt
- Data of the two plugs

Deconz_Start_Where_are_the_lights.log
 - No lights under light - nevertheless there are information for the IEEE - the device is available and content of the network

Deconz_Start_Excerpt_Deconz_Doing_Something_with_the_plugs.txt
- Excerpt of Start Sequence 

Start_Manual_Search_For_Lights.txt
- Search for new lights manually started - Both Plugs and the formerly created group re-appear

Smanar, 2020-06-07:
For next time, you can use just "--dbg-error=2 --dbg-info=2" for debugging.

BTW have you another light device to make test ?

I ask that because
- The conbee is added a new light (normal) but not saved too, and we can be sure this device is full supported ^^.
- on log we can see entry used, but I never see "light" table data updated or saved
- Absolutely nothing in light section in database when you load deconz.

Deconz sometime update the "node" but never the "light" table. Never
>DB sql exec REPLACE INTO lights (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode)

daluma, 2020-06-07:
Thanks, at the moment I do not have any further light devices available - what would you recommend that I should by as an additional test device?

Smanar, 2020-06-07:
A cheaper one.

Its realy for test, I m trying to compare with my logs but I have a bigger network, and hard to understand something, logs are realy hudge.

But I can see data be saved. When you run deconz with loging, and close it using ctrl + C , you will have one line by sensor saved and 1 lin by light saved 

>12:53:54:173 ~Resource() /lights 0073D1D0

Smanar, 2020-06-07:
Have you a windows machine ?

Can you install deconz on it, plug the key, include the plug and check if you have same problem ?

I m not sure the problem is from the device.

daluma, 2020-06-07:
Everything I have is installed on Windows - Nevertheless i tried again to start with an empty database for test purposes - including / removing the Plug etc ... always ends in the same result.

Smanar, 2020-06-07:
But you have tried on another machine with new install ? or the same install ?

daluma, 2020-06-07:
Yes, on a clean system ... also I left the Rest API aside and deactivated the plugin ... same

daluma, 2020-06-07:
I added the plugs to my HUE system and updated them to latest available version and re-added to the DeConz / Conbee system - same result - only thing I noticed now is that in the Node List in Deconz - the Plugs show No Model and No Version in the list.

SwoopX, 2020-06-07:
I'd recommend to use the debug flags @Smanar previously mentioned to get the debug output. Then, physically reset the plug and when it is ready to pair, start a light search and once the search has finished, share the debug output.

I see some strange ZDP errors in the log files I've yet never seen before.

daluma, 2020-06-07:
Prepared Test - Light Node 3 removed from network 

Debug Start Deconz 
First Light Search Addes Light Node 2 that is still part of the network 
Next Step pair  "Light Node 3" back to network
Log attached

According to the following information : 

seems added for lights? but I do not have light in the database?

00:17:88:01:08:9e:05:32-0b - This is the one i re-paired

00047686	31575.02539063	[22876] 21:39:46:385 LightNode 3: On/Off plug-in unit 3 added
00047687	31575.02539063	[22876] 21:39:46:385 ~Resource() /lights 0073D1B0
00047688	31575.02539063	[22876] 21:39:46:385 DB pushZdpDescriptorDb()
00047689	31575.02539063	[22876] 21:39:46:386 read attributes of 0x00178801089E0532 cluster: 0x0000: [ 
00047690	31575.02539063	[22876] 21:39:46:386 0x0005 
00047691	31575.02539063	[22876] 21:39:46:386 ]

[dbgview_new - repair 3.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4742798/dbgview_new.-.repair.3.log)

SwoopX, 2020-06-07:
Thanks. You got quite a lot of errors from your deconz DB in there. It also fails to update the 2 entries for the Philips plugs + the light node from the coordinator itself.

Have you played around with it?

daluma, 2020-06-07:
No I did not played around - the only devices that behave in that way are the two Philips Plugs where behave "different" all other nodes I still can add / remove without any issues since the beginning - the behaviour of the Philips Plugs where from the beginning - like described here 

- What irritates me shouldn't exist a table lights in my database when the Philips Node were added technically?

SwoopX, 2020-06-07:
> Question - can it be a problem that I had to skip initial lights setup due that I started with sensor - or speaking in other way - is there a must to start necessarily with a Light mandatory to setup the database properly?

No, that's not an issue.

The lights or the devices to be considered as lights are stored in table nodes. Maybe you can post its content and also its setup (`.schema nodes`).

daluma, 2020-06-07:
Created copy of current database for analysis: here are the results

nodes:

.schema nodes

CREATE TABLE nodes (mac TEXT PRIMARY KEY, id TEXT, state TEXT, name TEXT, groups TEXT, endpoint TEXT, modelid TEXT, manufacturername TEXT, swbuildid TEXT);

select * from nodes;

No content

SwoopX, 2020-06-07:
Interesting. The schema seems to be corrput. Mine is as follows (note the last field):

`CREATE TABLE nodes (mac TEXT PRIMARY KEY, id TEXT, state TEXT, name TEXT, groups TEXT, endpoint TEXT, modelid TEXT, manufacturername TEXT, swbuildid TEXT, ritems TEXT);`

daluma, 2020-06-07:
I added manually to the database the missing column - no the nodes is filled with the entries and the devices stay persistent ....

daluma, 2020-06-07:
@SwoopX - the groups table seems corrupt too due that this is not persistent and stored too... can we compare our tables?

Here is my groups table :

CREATE TABLE groups (gid TEXT PRIMARY KEY, name TEXT, state TEXT, mids TEXT, devicemembership TEXT, lightsequence TEXT, hidden TEXT, type TEXT, uniqueid TEXT);

SwoopX, 2020-06-07:
Its already there, check my last post.

daluma, 2020-06-07:
Got it, same here - added missing column - now the entry of the groups is populated.

Can we compare the other schema that I'm consistent ! - below I show the schema for each table that contains no content - maybe there is an additional one that is missing a column 

.schema gateways
CREATE TABLE gateways (uuid TEXT PRIMARY KEY, name TEXT, ip TEXT, port TEXT, pairing TEXT, apikey TEXT, cgroups TEXT);

.schema rules
CREATE TABLE rules (rid TEXT PRIMARY KEY, name TEXT, created TEXT, etag TEXT, lasttriggered TEXT, owner TEXT, status TEXT, timestriggered TEXT, actions TEXT, conditions TEXT, periodic TEXT);

.schema scenes
CREATE TABLE scenes (gsid TEXT PRIMARY KEY, gid TEXT, sid TEXT, name TEXT, transitiontime TEXT, lights TEXT);

.schema schedules
CREATE TABLE schedules (id TEXT PRIMARY KEY, json TEXT);

.schema userparameter
CREATE TABLE userparameter (key TEXT PRIMARY KEY, value TEXT);

.schema zcl_values
CREATE TABLE zcl_values (id INTEGER PRIMARY KEY, device_id INTEGER REFERENCES devices(id) ON DELETE CASCADE, endpoint INTEGER NOT NULL, cluster INTEGER NOT NULL, attribute INTEGER NOT NULL, data INTEGER NOT NULL, timestamp INTEGER NOT NULL);

daluma, 2020-06-07:
Even with asking for more from @SwoopX - I already want to say - THANKS to all here !!!! especially to @Smanar and @SwoopX - Cool community !!!!

SwoopX, 2020-06-07:
Just checked the fields, all ok. Please remember to close if the issue got resolved.

Welcome and enjoy!

closed 2020-06-07 by daluma

Smanar, 2020-06-08:
It works now ?

But what happen to the DB ? Corrupted by a power off during writting ?

daluma, 2020-06-08:
Yes, everything works now ... after adding the missing columns to the database the devices properly appeared ... but you are right the reason seems unclear even to me - the system was never turned of in that time - for sure I started or closed deconz in that time frame more than once  but opening / closing should not cause that ... Also true - i verified again just for curiosity - it happened on a other install too again ... interestingly the same columns - see above - where missing - maybe related to the devices? or order I did again - cancel initial Lights Setup and starting with Sensors only - maybe the reason?

Smanar, 2020-06-08:
For me, there is something realy something strange, and something that can happen to another user.

But, nevermind, we have this issue to compare, if the same problem happen again.

daluma, 2020-06-08:
True, I have more sensors and plugs on the road and then let it run - I will do some testing  also of starting / shutting down etc do see if something cause this again ... and will let you know.