dresden-elektronik/deconz-rest-plugin #2883: After a new installation some devices are not visible

bruesche, 2020-06-05:
Today I had an Issue with my raspberry so I had to install it again.
Before the crash I had two eurotronic thermostats in my config. In the Phoscon App I can't see it but with VNC on server directly in the Deconz "Network MAP". This devices were shown in my hue essantials with an ? Logo on Device Tap because there were unknown. a click an them show "on development". Although the CONBEE II Stick was shown there too. 

After the new installation everything seems to work. But no ? devices are in the Hue Essential App 
The thermostats and the Conbee II stick were not shown. Per VNC in the Network Map the devices and the stick is visible. 
Before the crash I used Homebridge to get all hue devices and sent them to Alexa. There I can control the thermostats. 
Now I can't.
The ? devices were shown with Phoscon version 2.05.75. Works perfect after update to 2.05.77 (? devices still there) but with a new installation off 2.05.77 no ? devices.

How can I get the not known devices (devices with ? logo) back in Phoscon / Hue Essantials so that I can control them als before.

Please help me.

Regards
Thorsten

Smanar, 2020-06-05:
Hello, you have only those 2 devices ?

bruesche, 2020-06-05:
Yes. 2 Thermostats which worked well till today. And the Conbee II stick is also not shown.

Smanar, 2020-06-06:
So now your devices are invisible in phoscon but visible in deconz (using GUI).

I know it's working because you are using deconz, but just to be sure, can you take a look in Phoscon/gateway to check if you have a firmware number displayed ?

So now you have no device at all visible in phoscon ?

You have used the backup/restore feature in Phoscon ?

bruesche, 2020-06-06:
Hi, 
Have tried it again. I resetted the thermostats (included to put Out the batteries). Then started a new search in phoscon web App. Then retry the pairing of the Thermostats (Open the gui, search the hex device, basic0000, Cluster, read). And now it works. The thermostats are back in Hue essentials. I think you must do the hole procedure exactly like I wrote above. 

Thanks for your help.

closed 2020-06-06 by bruesche