dresden-elektronik/deconz-rest-plugin #2884: X71XA Pairing

DJBenson, 2020-06-05:
According to [this](https://zigbee.blakadder.com/deconz.html) the X71XA switches (manufactured by Lerlink, Lenhonso and Zemismart to name but a few) should be supported but I can't get my two switches (a X712A and X713A) to pair. I can put them into pairing mode (by holding down gang/L1 for 5 seconds until the LED flashes) and then it flashes briefly then stops flashing (as if to indicate it had paired) but the switch never shows in Phoscon and doesn't appear to be in deCONZ itself (confirmed by restarting Homebridge to pull down new devices and also viewing the deCONZ GUI in an X window session).

What am I missing? Are they supported? If so, has anybody got any tips for getting these to show in homebridge-hue?

Smanar, 2020-06-06:
I m sorry but the only way to do something is seing the device in deconz.

It's your device ?

Try to run deconz with GUI, and set it in permit join (don't use phoscon).
You realy need to see the device in deconz to do something.

BTW I have already see this king of issue using " X-Forwarding", I know nothing in this procedure, but it was like the system use 2 different "sessions", all that happen in 1 wasn't saved on the other.

DJBenson, 2020-06-06:
X forwarding is just a method of viewing the deconz GUI on my headless system. You can’t connect to two instances as the stick is locked so the only way to use it is to stop the deconz service and start the GUI session so it’s 100% nothing to do with X forwarding. 

My question was is this device supported, the above link says yes (which I assume means someone got it working rather than it being officially supported) because it’s not on the deconz list.

I will try again with the deconz GUI but I didn’t see the device join yesterday despite allowing new devices to join and the device itself exiting pairing mode within seconds (as a device does when it has successfully paired).

Smanar, 2020-06-06:
I don't use "X forwarding", I know the theory too, but I know some users have issue with it, and without explanation since.

And zigbee.blakadder is not as realiable as owner github, most of the website is just copy/paste from other site, if you want to know is a device is compatible just go on the solution github and use search engine.

If you see the device join and leave we can perhaps do something, but if it don't join, there is nothing to do.

And you already have some device working on your gateway ?

DJBenson, 2020-06-06:
@Smanar point taken about X-Forwarding, I apologise if I appeared abrupt (replying by phone is never a good idea when busy with other stuff). I just wanted to make it clear I know how to avoid the issues you refer to around resource clashes.

I'm looking at deCONZ right now and there are two devices in my network map (both offline currently as they are unplugged) but they are unknown to me, I guess these may well be my switches (I powered up and tried to pair 2 of the 3 I have).

I am really new to this low-level of networking of the Zigbee network (moved over from Hue where it's all wrapped in a nice UI and you never have to worry about the network side) so I'm not sure what I'm looking at below;

[image]

Could these be my switches?
If so, how do I get them available via the API? (In fact, is it possible to call the API via a browser to see if these are returned?)

DJBenson, 2020-06-06:
Closing as it turns out this is a duplicate of #2879 as my switch is identical. Would be great to get these added to deCONZ though. I will keep my fingers crossed.

closed 2020-06-06 by DJBenson

Smanar, 2020-06-07:
Right, they have the same MAC address range.

I thought of a problem because they were disconnected and therefore grayed out.

But they are light, so they can work natively, the device is not visible at all in the API ?