dresden-elektronik/deconz-rest-plugin #2885: Philips Hue Outdoor PIR Issues

meremortals70, 2020-06-06:
Hi All

I have three active Philips Hue Outdoor PIR's.

One is very stable and does not cause any issues.

With the other two, when I add another device, they often respond and reset their names to Motion 
Sensor.

All three have the same model identifier and date code

Any hints on how to fix this?

meremortals70, 2020-06-06:
ok an update here, I removed the PIR's added them to a Philips Hue Hub and upgraded them to the latest firmware. 

March 11, 2019
Software version: 6.1.1.27575

Once complete, I reset them and attempted to connect them back into Deconz.

They all now appear without attributes

[image]

Any advice please

SwoopX, 2020-06-06:
Could it be that they' not yet supported? Based on the last screenshot, pairing was not successful.

Smanar, 2020-06-06:
Or a bad inclusion ?
Clic on "Read simple descriptor" and "Read node descriptor"

meremortals70, 2020-06-06:
@SwoopX they worked just fine before the firmware upgrade and they are on the supported list Philips Hue Outdoor Motion Sensor.

@Smanar if you look closely at the screen shot, you will see that read simple descriptors is not selected, doesn’t matter what I do it cannot be selected.

SwoopX, 2020-06-06:
Should have taken a closer look, the node descriptor has not even been picked up. Try reading the node descriptor to see if the node info pane changes. If not, reset the device and re-pair.

If that's still no good, run deconz with `--dbg-info=2` and make sure to read the node descriptor once again and then provide the output. We should then see what data comes in and if that got modified.

meremortals70, 2020-06-06:
I have just removed the batteries from the sensor.

Tried to re-pair, Deconz crashed.

Restarted Deconz and paired again.

Motion sensor is now in Deconz but not in Phoscon/Add-in for HA

meremortals70, 2020-06-07:
I had to re-pair again as an “other” sensor in Phoscon the “Philips” sensor did not work