dresden-elektronik/deconz-rest-plugin #2886: Bitron Home 901020/24 Smoke Sensor/Warning Device, Siren does not work

ChrisPl-GH, 2020-06-06:
ConBee 1, Firmware 26350500, Version 2.05.77 (Headless)
I have several Bitron 901020/24 smoke detectors with integrated siren. There are two devices, one sensor 'ZHAFire' and one light 'Warning device'. The sensor works, but the siren cannot be switched. Using the REST-API I can send an alert:select, but the status does not change and the siren is not activated.

> {
  "etag": "035554033xxxxxxf0684d3b78c0e6905",
  "hascolor": false,
  "lastseen": "2020-06-05T23:03:46.298",
  "manufacturername": "Bitron Home",
  "modelid": "902010/24",
  "name": "x_smoke_6_alarm",
  "state": {
    "alert": "none",
    "reachable": true
  },
  "swversion": "20150615",
  "type": "Warning device",
  "uniqueid": "00:12:4b:00:xx:xx:xx:3b-01"
}

curl -X PUT -i 'http://192.168.xxx.xxx/api/xxxxxxxx/lights/21/state' --data '{"alert": "select"}'

> [{
  "success": {
    "/lights/21/state/alert": "select"
  }
}]

ebaauw, 2020-06-06:
Can the siren for this device be activated remotely?  We’ve seen other devices that expose a non-functioning _IAS WD_.  Can you check in the GUI whether the device reacts to the _Start Warning_ command.  Also double-check that you can read the _Basic_ cluster attributes, maybe deCONZ had lost the route.  You might try setting alert to `"all"`, this sends a broadcast and should not be impacted by routing issues.

SwoopX, 2020-06-06:
According to ma-ca's initial commit back in the days, it can. However, you made some changes on the task options lately. It works for the Develco's but may not for the Bitron...

> I tried different task.options and found that the Bitron smoke detector siren did only start with values 0x02 and 0x12. The latter did also start the blinking red LED.

ebaauw, 2020-06-06:
@SwoopX, could very well be that the Bitron requires different values.  I'll whitelist it again and send 0x12 as before.

ChrisPl-GH, 2020-06-06:
@ebaauw
Yes, the siren can be activated separately, e.g. via QIVICON or Telekom Smart Home.
https://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/optischer-rauchmelder/

I am running deCONZ headless on a Raspberry and therefore cannot use the GUI.

Using the REST API I can send an alert:all, but the status does not change and the siren is not activated.

curl -X PUT -i 'http://192.168.xxx.xxx/api/xxxxxxxx/lights/21/state' --data '{"alert": "all"}'

> [{
  "success": {
    "/lights/21/state/alert": "all"
  }
}]

How can I send the values 0x02 or 0x12 via REST API to the Bitron?

ChrisPl-GH, 2020-06-06:
@ebaauw
The Bitron Siren can be assigned to a group, but also in the group the sent alert does not work.

>   "12": {
    "action": {
      "bri": 128,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": true,
      "sat": 128,
      "scene": null,
      "xy": [0, 0]
    },
    "devicemembership": [],
    "etag": "ec417e3fec966a70e10c330cd39b1723",
    "id": "12",
    "lights": ["21"],
    "name": "Sirenen",
    "scenes": [],
    "state": {
      "all_on": false,
      "any_on": false
    },
    "type": "LightGroup"
  }

ChrisPl-GH, 2020-06-18:
With version 2.05.78 the siren now works.
Via an alert:select or alert:lselect, the smoke detector emits a simple or permanent acoustic and visual alarm.

closed 2020-06-18 by ChrisPl-GH

cyablo, 2020-06-29:
Sorry to bring up this issue again. As soon as the alarm is triggered, can you turn it back off? I use deconz with home assistant and I can turn on the siren from within HA, but I can not turn it off. Even not on the device itself, I need to pull the batteries.

SwoopX, 2020-06-29:
That should work with `alert:none`

kptkip, 2020-07-22:
Does this command work also via Phoscon?

SwoopX, 2020-07-22:
No, not that I'm aware of.

kptkip, 2020-07-22:
@SwoopX : After updating Phoscon to 2.05.78 (Firmware: 26580700) I can trigger the alarm of the Smoke-Sensor.

I use Phoscon from Fhem as a HueBridge. There is the sensor listed as "lamp" (like in Phoscon). There I can use:

set HueLampDeviceName alert select|lselect|none

To trigger the alarm and to switch it off.

SwoopX, 2020-07-22:
@kptkip I'm afraid you posed the wrong question 😉 

Phoscon is just another client for deconz' REST API, so the answer was for Phoscon as a client. Now, since all other software pretty much uses the REST API, the answer is of course yes.

kptkip, 2020-07-22:
Thanx for this clarification. 

Up to now I was wondering what is the difference between those two pieces of software. Now It's clear!

And no, I can't control these Features in Phoscon. ;-)