dresden-elektronik/deconz-rest-plugin #2887: Info: not only deconz that loses lights.

olemr, 2020-06-06:
Hello.

I run 3 Zigbee networks at home, mostly to stay updated with what is going on with the different vendors.
Deconz is my main network, with 40 lights, 15 switches and 1 sensor. Multiple groups.
A Philips HUE bridge caters for 2 6xGU10 fixtures in the den, and 16 E27 CT facade lights. 2 groups.
Ikea bridge just have one Ikea remote and 1 single E27 CT bulb.

All my E14, E27 and GU10 bulbs are Ikea Trådfri, 4-5 HUE strips and 3xLillies.

I must say that the Pilips HUE bridge loses Ikea bulbs just as often as deconz.
Losing HUE lights are rear, but it does happen.
Even Ikea loses its own bulb, and after the battery ran out on the remote, I had to re-pair it.

Just to let you all know that none of these systems are yet perfect, but I'm hopeful that the analysis now going on will improve matters.

Thank you all for contributing.

Smanar, 2020-06-06:
So to resume philips have better hardare than ikea ?

olemr, 2020-06-06:
Yes, or better firmware?
But you are right. Topology and other 2.4 GHz interference can also play a role.

Smanar, 2020-06-06:
It's personnal, but for me Philips is expensive but they are the better router.

closed 2020-07-11 by stale[bot]