dresden-elektronik/deconz-rest-plugin #2890: RaspBee network key corrupted to all-CC, all devices lost

tmbinc, 2020-06-06:
My hardware setup:

- Raspberry Pi 3
- RaspBee
- deCONZ + Phoscon + hass
- Mix of Xiaomi/Aquara/TI Sample/Osram devices

Summary after my investigation:

RaspBee is suddenly not using the configured link key, but an all-CC key, hence not seeing any encrypted traffic anymore. Changing the network key in the deCONZ GUI has no effect. 

Details:

Everything was working until a few weeks ago (~middle of May 2020; not sure if related to an update), where suddenly overnight all devices became unreachable. 

Network settings:

[image]

As you can see, it's a standard ZHA setup on Ch 15. Network key is set as 3fe2fbc81e2a0c017a3c58d4b653e404, TC link key as 5A6967426565416C6C69616E63653039.

Here's a capture of network traffic:

[deconz.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4740810/deconz.zip)

You can see various devices reporting ZCL attributes, all using the established network key (3fe...).  (You need to add the key in Wireshark under Edit -> Preferences -> Protocols -> Zigbee -> Pre-configured keys). 

However, all traffic from device 0000 (RaspBee) is not using this key. By joining a new devices (via TC link key) I discovered that the key that 0000 is operating on is actually cccccccccccccccccccccccccccccccc. (Also add this key to wireshark - it will automatically select which key has been used for decryption in the "Zigbee Security Header")

Pressing "Load" or "Save" in the network settings (in deCONZ GUI) does not fix anything - I can change settings, and they are reflected when reading them back, but the used key in use is always all-CC. (the all-CC key is never displayed in the UI; only the 3fe... network key)

Re-loading the zigbee network config via PhosCon doesn't change anything.

SwoopX, 2020-06-06:
Tried that already?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

tmbinc, 2020-06-07:
Yes - tried that, and it didn't work. (Does that do something fundamentally different than re-writing the network settings from deCONZ?)

Here's deCONZ writing the network key - it appears correctly in the serial port data. Yet it still uses the all-CC key. I thus suspect this to be an issue with the RaspBee firmware. 

```
$ sudo strace -f -xx -p 13011 -e write=19  -e read=19 -y 2>&1 | grep -v "poll" | grep "19<"
[pid 13151] write(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\xc0\x0b\x8a\x00\x19\x00\x12\x00\x18\x00\x3f\xe2\xfb\xc8\x1e\x2a\x0c\x01\x7a\x3c\x58\xd4\xb6\x53\xe4\x04\x1c\xf8\xc0", 29) = 29
#                                                                                                          ^^^ nework key
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\xc0", 512) = 1
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\x0b\x8a", 512) = 2
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\x00", 512) = 1
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\x08", 512) = 1
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\x00\x01", 512) = 2
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\x00", 512) = 1
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\x18\x4a", 512) = 2
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\xff", 512) = 1
[pid 13151] read(19<\x2f\x64\x65\x76\x2f\x74\x74\x79\x53\x30>, "\xc0", 512) = 1

```

tmbinc, 2020-06-07:
I was able to "solve" this by pressing "Leave" and then "Join" in deCONZ. I did not try this before.

Still, it's unclear to me what caused this to happen. I'll monitor the situation and see if this happens again.

SwoopX, 2020-06-07:
Glad you made it. In fact, your hint might prove very valuable for others experiencing the same.

I had that "pleasure" once back in the days and was lucky to have a sniffer available. After some struggle, restoring the config brought success in my case. However, I haven't tried your approach at that time.

manup, 2020-06-08:
>I was able to "solve" this by pressing "Leave" and then "Join" in deCONZ. 

Hmm interesting the Zigbee configuration page should normally do the same when the 'Load' button is pressed.

manup, 2020-06-08:
Is this a RaspBee I or II?

tmbinc, 2020-06-08:
It is a RaspBee II. (I assumed it's identical to the RaspBee I except for the RTC - is that not the case?)

tmbinc, 2020-06-08:
Regarding "Load' and "Save" in network settings, I've noticed:

- When I change something, say a byte of the key, and I press "save", then the old value re-appears.
- When I now press "load", the new value is populated in the UI.
- (But as discussed above, changing the key - even back and forth - did not change the behavior, only "leave, then join" did.)

I assume this is a UI glitch and not related to my problem.

SwoopX, 2020-06-08:
I pretty much can confirm the above.

closed 2020-08-02 by stale[bot]