dresden-elektronik/deconz-rest-plugin #2891: Implementation of the HUE Entertainment API

u20p17, 2020-06-07:
Hi,
would it be possible to implement the HUE Entertainment API in the future Deconz versions? Would be really nice!

Thank you in advance!

Best regards,

CM000n, 2020-06-09:
Absolutely. I would also find that a very interesting feature. The lack of entertainment functionality is the only reason why I am running a diyHue container besides Deconz at the moment. It would be nice to have this feature built right into Deconz.

Mimiix, 2020-06-10:
I brought this up during the dev meeting yesterday. I've received a comprehensive answer, but ended up into a no due to a few limitations. This seem only to work with hue lights as there is some design in there. Other bulbs wouldn't work with this because these are missing that design.

To implement this, it would require allot of resources to make this happen. That being said, there are more urgent matters at this moment that need attention.

You are free to investigate this and make a PR to add it.

closed 2020-07-02 by Mimiix

MBj1703, 2020-11-06:
Really sorry to read.

Even if only HUE will be supported would be fine.

oskar1999, 2021-04-05:
I'd love to see that too. Only using supported Hue bulbs would be no problem for me, I just don't wanna add another device to my setup. But I guess I have to if I wanna use Hue Entertainment.