dresden-elektronik/deconz-rest-plugin #2894: How-To compile own changes / pull request

bortim, 2020-06-07:
Hey guys,

I am trying to compile a pull-request before it goes to the official build.
My environment is a Docker-Container on a Ubuntu 20.04 LTS Raspberry-Image.

Within the container mat dependencies are missing.

That’s why I downloaded and installed the deconz and deconz-dev version, which is running within my container.

Then I downloaded the source of the pull-request and compiled it as described in the readme.

After compiling libde_rest_plugin.so was created.
Over the docker-directory I made it available in the Container and copied it to the plugins-folder.

Then I try to restart the container.

The web-Frontend starts (callable within the browser) but a gateway is not shown.

Then I started the with more logging.
The general.xml was loaded, then two plugins.
But not the Rest-Plugin.

If I switch back to the old rest-plugin within that step the loading is shown with one log-entry.

I tried to do a chmod with the original as a reference file, but it did not work.

All in all:
I have no further idea, what I can do now.

So maybe somebody can give me some hints or a manual.

Regards
bortim

SwoopX, 2020-06-07:
Looks like you compile for the wrong architecture. I also experience that when trying to compile on Rpi4 for Rpi zero, which also does't work out. I was yet too lazy to make an appropriate toolchain available.

So in all, You need to find out which architecture your docker assumes and the appropriately cross-compile it.

bortim, 2020-06-08:
Hello @SwoopX,

Ubuntu Server 20.4 LTS host:
```
$ docker version
OS/Arch:          linux/arm64
$ uname -m
aarch64
$ uname -a
Linux ubuntu 5.4.0-1011-raspi #11-Ubuntu SMP Fri May 8 07:43:33 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
$ arch
aarch64
$ dpkg --print-architecture
arm64
$ getconf LONG_BIT
64
$ file /usr/bin/ld
/usr/bin/ld: symbolic link to aarch64-linux-gnu-ld
$ file libde_rest_plugin.so
ELF 64-bit LSB shared object, ARM aarch64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=71275c4456d4d0e9e2f0a359b8e569a37de015ee, not stripped
```

Within the docker container: sudo docker exec -it deconz /bin/bash
```
$ uname -m
aarch64
$ uname -a
Linux ubuntu 5.4.0-1011-raspi #11-Ubuntu SMP Fri May 8 07:43:33 UTC 2020 aarch64 GNU/Linux
$ arch
aarch64
$ dpkg --print-architecture
arm64
$ getconf LONG_BIT
64
$ file libde_rest_plugin.so
ELF 64-bit LSB shared object, ARM aarch64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=02a7d73e49ba2db940242d76e2f4067227e05737, not stripped
```

So all in all I think that it is the same.
BUT: I don't know whty the docker container returns for uname -a the same as my host.
I would expect that there is something with debian and not with ubuntu.

Thanks and regards

SwoopX, 2020-06-08:
Hm, good question, but good to see someone knowns how to handle the tools ;)

bortim, 2020-06-08:
Now I created a Virtual Machine with Ubuntu Desktop 20.04 LTS on my AMD64-Environment.

I added the phoscon-key and repository to APT (changed from focal to bionic) and installed deconz and deconz-dev on the environment.
Afterwards I cloned the pull-request, compiled from that and copied the libde_rest_plugin.so to the plugin folder of deconz.
Restarted deconz and opened phoscon.
Then I added my remote (that's why I have to compile the pull-request) - it is not viewable/useable by Phoscon, but I can access it directly by the API - great as I want to use it within iobroker.

But: I want to let deconz run in a Docker-Container and do not have a solution for that.
Maybe I find with my new knowledge a solution on that.

bortim, 2020-06-08:
It is so confusing.
Now I did the same directly on my Raspberry Pi.
The compiled library was not working within the container.
But it worked directly on my Raspberry Pi.

Nevertheless I would prefer to run it within docker.

Mimiix, 2020-06-09:
@bortim You should ask that question here: https://github.com/marthoc/docker-deconz

Mimiix, 2020-06-10:
Closing for now. Will re-open if needed ;)

closed 2020-06-10 by Mimiix